kellerza/sunsynk #300: Three fase 12K hybrid inverter but no L1 L2 and L3 power for all Load, Inverter and Grid?

TheBazeman, 2024-06-22:
Hi Kellerza,

Thanks for the great addon, just installed my Deye inverter and communication works. Selected the three fase option but I dont have info at all on the phases itself. I am sure I am missing something, I saw the definition files which conain it.

What can I do?

Thanks in advance,
Bas

TheBazeman, 2024-06-24:
Good morning, So I have digged around a bit and it seems it doesn't recognize the inverter correct as the definitions https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/three_phase_common.py are not being loaded by the plugin while it is marked as a three fase inverter in the configuration.

Just to be clear I have the 12K-SG04LP3-EU, which is the 12K 3-phase Inverter with LV battery.

Manually adding the first sensors mentioned in the 3 phase definition file to the configuration page SENSORS:
"Inverter power", "Inverter L1 power", "Inverter L2 power", "Inverter L3 power", "Inverter voltage","Inverter frequency"
 and restart the plugin shows the values of the added sensors.

Am I expecting something different from this **awesome** plugin and should I manually add all wanted 3 phase sensors or should it do it automatically?

Hope to hear from you,

Thanks in advance, Bas

The log with plugin startup:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
08:56:49 INFO    Loading configuration: /data/options.json
08:56:49 INFO    sunsynk library version: 0.6.4
08:56:49 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  5   |   60   |     80    |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
08:56:49 INFO    Using three phase sensor definitions.
08:56:49 ERROR   Unknown sensor specified: aux_power
08:56:49 ERROR   Unknown sensor specified: essential_power
08:56:49 ERROR   Unknown sensor specified: inverter_current
08:56:49 ERROR   Unknown sensor specified: load_frequency
08:56:49 ERROR   Unknown sensor specified: non_essential_power
08:56:49 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
08:56:49 INFO    Connecting to serial:/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_B001A2DN-if00-port0
08:56:49 INFO    Reading startup sensors Prog3 Time, Rated power, Battery Low Capacity, Prog6 Time, Serial, Prog5 Time, Prog1 Time, Prog4 Time, Prog2 Time
08:56:49 INFO    ############################################################
08:56:49 INFO                 Inverter serial number '****0233'
08:56:49 INFO    ############################################################
08:56:49 INFO    MQTT: Connecting to deyeaddon@core-mosquitto:1883
08:56:49 INFO    MQTT: Connection successful
08:56:50 INFO    Read every (inverter 1)
+-----+-------------------------------------------------------------------------------+
|  s  |                                    Sensors                                    |
+-----+-------------------------------------------------------------------------------+
|  5  |       prog4_charge, prog6_charge, prog3_capacity, prog3_time, use_timer,      |
|     |     priority_load, load_limit, prog2_time, prog2_capacity, prog4_capacity,    |
|     |     pv1_power, inverter_l3_power, pv2_power, prog5_charge, battery_power,     |
|     |       prog6_capacity, prog6_time, prog3_charge, prog1_time, prog3_power,      |
|     |    inverter_l1_power, prog1_charge, prog6_power, prog1_power, prog4_power,    |
|     | grid_ct_power, inverter_l2_power, prog1_capacity, prog5_time, prog5_capacity, |
|     |   prog2_power, prog5_power, prog2_charge, grid_power, battery_low_capacity,   |
|     |                           inverter_power, prog4_time                          |
|  15 |     inverter_frequency, battery_current, grid_frequency, inverter_voltage,    |
|     |    grid_voltage, pv1_current, grid_connected, battery_soc, battery_voltage,   |
|     |                           pv1_voltage, overall_state                          |
| 300 |     day_battery_discharge, total_battery_discharge, total_battery_charge,     |
|     |   total_pv_energy, total_grid_import, day_battery_charge, total_grid_export,  |
|     |        day_pv_energy, day_grid_export, day_grid_import, day_load_energy       |
+-----+-------------------------------------------------------------------------------+
08:56:50 INFO    Report every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  60 |       grid_power, pv1_power, grid_ct_power, inverter_l3_power, pv2_power,        |
|     |       inverter_l2_power, battery_power, inverter_power, inverter_l1_power        |
[... cut ...]
|     |   prog4_power, day_battery_charge, day_pv_energy, pv1_current, grid_connected,   |
|     |   prog1_capacity, prog5_time, prog5_capacity, inverter_frequency, prog2_power,   |
|     |         overall_state, prog5_power, day_battery_discharge, prog2_charge,         |
|     |     total_battery_charge, battery_low_capacity, grid_frequency, prog4_time,      |
|     |                           battery_voltage, pv1_voltage                           |
+-----+----------------------------------------------------------------------------------+

rixxxx, 2024-06-26:
No sensors are added by default. Set of sensors are added with groups "energy_management", "power_flow_card" and "settings". All the other need to be added manually, as all users doesn't require all the sensors.

TheBazeman, 2024-06-28:
Ah ok, my bad then! Will add the sensors manually. Thanks Bas

closed 2024-06-28 by TheBazeman