kellerza/sunsynk #301: Error when tyring to update Grid_Peak_Shaving

cods4, 2024-06-23:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

I think this has started occuring since the most recent commits on the 19th of June.

Trying to change 'Grid_Peak_Shaving' no longer works. I get an error in the log:
```TypeError in read_inv1: sunsynk.rwsensors.RWSensor.reg() argument after * must be an iterable, not int```

Along with some other errors which I think are new.
See attached screenshot.
[image]

### Your environment

- Home Assistant version: 2024.6.3
- Addon:
  - Name: sunsynk-edge/dev
  - Version: dae73f8
- Inverter:
  - Make: Deye
  - Model: \_\_
  - Firmware: \_\_
- Adaptor details:
  - mbusd

### You configuration

```
DRIVER: umodbus
INVERTERS:
  - SERIAL_NR: "1234"
    HA_PREFIX: INV1
    MODBUS_ID: 1
    PORT: tcp://192.168.23.17:50001
  - SERIAL_NR: "1234"
    HA_PREFIX: INV2
    MODBUS_ID: 1
    PORT: tcp://192.168.23.17:50002
  - SERIAL_NR: "1234"
    HA_PREFIX: INV3
    MODBUS_ID: 1
    PORT: tcp://192.168.23.17:50003
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - battery_power
  - battery_current
  - inverter_power
  - inverter_voltage
  - inverter_current
  - inverter_frequency
  - grid_frequency
  - grid_power
  - grid_ld_power
  - grid_voltage
  - grid_current
  - grid_ct_power
  - load_power
  - load_frequency
  - pv1_power
  - pv1_voltage
  - pv1_current
  - pv2_power
  - pv2_voltage
  - pv2_current
  - essential_power
  - essential_1_power
  - essential_2_power
  - non_essential_power
  - total_active_energy
  - total_battery_charge
  - total_battery_discharge
  - total_grid_export
  - total_grid_import
  - total_load_energy
  - total_pv_energy
  - fault
  - overall_status
  - dc_transformer_temperature
  - radiator_temperature
  - grid_connected
  - date_time
  - control_mode
SENSORS_FIRST_INVERTER:
  - battery_temperature
  - battery_voltage
  - battery_soc
  - battery_charge_limit_current
  - battery_discharge_limit_current
  - grid_charge_enabled
  - battery_low_capacity
  - battery_shutdown_capacity
  - battery_restart_capacity
  - priority_load
  - load_limit
  - solar_export
  - use_timer
  - prog1_time
  - prog2_time
  - prog3_time
  - prog4_time
  - prog5_time
  - prog6_time
  - prog1_power
  - prog2_power
  - prog3_power
  - prog4_power
  - prog5_power
  - prog6_power
  - prog1_capacity
  - prog2_capacity
  - prog3_capacity
  - prog4_capacity
  - prog5_capacity
  - prog6_capacity
  - prog1_charge
  - prog2_charge
  - prog3_charge
  - prog4_charge
  - prog5_charge
  - prog6_charge
  - prog1_mode
  - prog2_mode
  - prog3_mode
  - prog4_mode
  - prog5_mode
  - prog6_mode
  - grid_charge_battery_current
  - battery_max_charge_current
  - battery_max_discharge_current
  - pv_curve_enable
  - limiter_function_enable
  - limit_grid_output_power
  - export_control
  - gen_peak_shaving
  - grid_peak_shaving
  - drm
MANUFACTURER: Deye
READ_ALLOW_GAP: 5
READ_SENSORS_BATCH_SIZE: 100
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 100
    CHANGE_PERCENT: 0
  - KEY: kWh
    READ_EVERY: 30
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 0
    CHANGE_PERCENT: 0
  - KEY: A
    READ_EVERY: 10
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 1
    CHANGE_PERCENT: 0
  - KEY: V
    READ_EVERY: 10
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 0
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: 15
MQTT_HOST: host
MQTT_PORT: 1883
MQTT_USERNAME: un
MQTT_PASSWORD: pw
DEBUG: 0
DEBUG_DEVICE: /dev/ttyS0

```

### Logs

See above screenshot. Unfortunately I was not in front of my computer at the time, so could only get a screenshot of the log.

kellerza, 2024-09-06:
This is likely fixed in 0.7.x

kellerza, 2025-02-15:
I added some grid peak shaving settings, see <https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/single_phase.py#L245-L251>

Would appreciate feedback on what works, what not

kellerza, 2025-03-18:
Shoul dbe fixed in the latest

closed 2025-03-18 by kellerza