kellerza/sunsynk #305: missing values in three_phase_hv.py

bsafh, 2024-06-25:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions.py)

### You Inverter

Deye SUN-10K-SG01HP3-EU-AM2

```
    SN：2312122150 
    Wechselrichtertyp：Dreiphasiger Hochspannungs-Hybrid 
    Nennleistung：10 kW 

Version Informationen

    Protokoll-Version：0104 
    NETZ：3002-1061-1E08 
    HMI：2001-C02D 
    Lithium-Akku-Versionsnummer：258 
    Lithium-Akkus2 Versionsnummer：0000 
    Deutsche Version：1004 
    Lithium-Akku-Versionsnummer：0102 
    Lithium-Akku-Hardware-Versionsnummer：0203 
```

### Describe the issue/bug and what you expect

the addon (dev branch) reports missing values for various variables

Any suggestions on proposed registers is welcome.

device_type: 6   - this is a SUN-10K-SG01HP3-EU-AM2 high voltage 3-phase hybrid inverter

for the others I can't find settings in the Solarman Business portal for this device, sorry.

### Logs (if applicable)

```logs

10:26:01 WARNING device_type: Unknown register value 6. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
10:26:01 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
10:26:01 WARNING on_grid_always_on: Unknown register value 0
10:26:01 WARNING us_version_grounding_fault: Unknown register value 0
10:26:01 WARNING allow_remote: Unknown register value 0

```

rixxxx, 2024-06-26:
Please share your config

bsafh, 2024-06-26:
```
DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "2312122150"
    HA_PREFIX: WR_Deye_1
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: serial-tcp://192.168.80.44:8234
  - SERIAL_NR: "2312126286"
    HA_PREFIX: WR_Deye_2
    MODBUS_ID: 2
    DONGLE_SERIAL_NUMBER: "0"
    PORT: serial-tcp://192.168.80.45:8234
SENSOR_DEFINITIONS: three-phase-hv
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
  - settings
  - all
  - pv1_power
  - essential_power
  - battery_power
  - battery_soc
  - non_essential_power
  - grid_ct_power
  - Inverter power
  - Inverter L1 power
  - Inverter L2 power
  - Inverter L3 power
  - Inverter voltage
  - Inverter frequency
  - Grid frequency
  - Grid power
  - Grid L1 power
  - Grid L3 power
  - Grid L2 power
  - Grid L2 voltage
  - Grid L1 voltage
  - Grid L3 voltage
  - Load power
  - Load L1 power
  - Load L2 power
  - Load L3 power
  - Load L1 voltage
  - Load L2 voltage
  - Load L3 voltage
  - PV1 power
  - PV1 voltage
  - PV1 current
  - PV2 power
  - PV2 voltage
  - PV2 current
  - Gen power
  - Gen L1 power
  - Gen L2 power
  - Gen L3 power
  - Gen Energy
  - Load Energy
  - Total Load Energy
  - Rated power
  - Device type
  - Fault
  - Overall state
  - Serial
  - DC transformer temperature
  - Radiator temperature
  - Grid Connected
  - Date Time
  - INV Relay Status
[... 217 lines cut ...]
    CHANGE_ANY: false
  - KEY: date_time
    READ_EVERY: 60
    REPORT_EVERY: 60
    CHANGE_ANY: true
  - KEY: rw
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: true
  - KEY: any_unit
    READ_EVERY: 10
    REPORT_EVERY: 120
    CHANGE_ANY: false
  - KEY: no_unit
    READ_EVERY: 10
    REPORT_EVERY: 120
    CHANGE_ANY: true
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: 15
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqttuser
MQTT_PASSWORD: secret

```

alistairts, 2024-07-01:
I also have an issue with missing/incorrect values read from my 50kW 3ph Deye inverter at my office.

When I restart your sunsynk add-on, the log shows several "unknown sensors specified" (see below log).
FYI, I have a 5kW 1ph Deye inverter at my home reading this information successfully and displaying nicely using the power flow card.
Now that I've selected three-phase sensor definitions, these sensors probably do not exist in this list...

Secondly, the decimal placement for the power reading of PV1, to 4 is incorrect, for example it reads 220.5W instead of 2205W.

Thirdly, my 50kW inverter features two battery inputs, and your add-on is only reading the data of battery 1.
Is there a register value that outputs the combined (total) battery power?

And lastly, the decimal placement for the battery voltage is also incorrect, for example it reads 58.775V instead of 587.75V

[14:48:23] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
14:48:28 INFO    Loading configuration: /data/options.json
14:48:28 INFO    sunsynk library version: 0.6.4
14:48:28 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |  *  |  15  |   60   |           |          |    True    |
|     w     |  *  |  15  |   60   |     50    |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |  *  |  30  |   60   |     50    |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
14:48:28 INFO    Using three phase sensor definitions.
14:48:28 ERROR   Unknown sensor specified: aux_power
14:48:28 ERROR   Unknown sensor specified: essential_power
14:48:28 ERROR   Unknown sensor specified: inverter_current
14:48:28 ERROR   Unknown sensor specified: load_frequency
14:48:28 ERROR   Unknown sensor specified: non_essential_power
14:48:28 ERROR   Unknown sensor specified: inverter_current
14:48:28 ERROR   Unknown sensor specified: load_frequency
14:48:28 ERROR   Unknown sensor specified: grid_ld_power
14:48:28 ERROR   Unknown sensor specified: essential_1_power
14:48:28 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
14:48:28 INFO    Connecting to tcp://172.72.2.250:8899
14:48:28 INFO    Reading startup sensors Prog2 Time, Prog6 Time, Prog4 Time, Prog5 Time, Prog1 Time, Prog3 Time, Rated power, Battery Low Capacity, Serial

rixxxx, 2024-07-01:
> 14:48:28 INFO Using three phase sensor definitions.

You are using LV definition on HV inverter and this causes the decimal to be off.

alistairts, 2024-07-01:
That makes sense, the UI configuration only had the two options single-phase and three-phase to select.
Using the YAML option I make it three-phase-hv as per your sensor definition documentation, but it fails to save.

Failed to save add-on configuration, value must be one of ['single-phase', 'three-phase'].

rixxxx, 2024-07-01:
I'm not using the HA addon, but I think HV is only present on edge(or something similar) version.

bsafh, 2024-07-01:
> I'm not using the HA addon, but I think HV is only present on edge(or something similar) version.

It's präsentiert in the dev/edge Version of the Addon. I use it for my HV inverters, works finde with a few unkown values (inverter Version etc) showing up.

rixxxx, 2024-07-01:
#309 should address those unknown values

bsafh, 2024-07-01:
> #309 should address those unknown values

#309 Talks about making the entities model-aware ... but the Model Codes back as unkown value, so that would have to be fixed first, I guess

```
10:26:01 WARNING device_type: Unknown register value 6. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
10:26:01 WARNING microinverter_export_to_grid_cutoff: Unknown register value 0
10:26:01 WARNING on_grid_always_on: Unknown register value 0
10:26:01 WARNING us_version_grounding_fault: Unknown register value 0
10:26:01 WARNING allow_remote: Unknown register value 0
```

rixxxx, 2024-07-01:
#309 addresses unknown sensors, not unknown values you have here

rixxxx, 2024-07-01:
@bsafh looking at your errors, these sensors have undocumented values. You can change them in HA once and then it doesn't complain anymore. If you don't know what they are for, I'd suggest to remove them from SENSORS.

rixxxx, 2024-07-01:
@kellerza should we add all "No work" values to all those bitmask sensors? For some we have one of the Noworks defined, for some we have only Enable and Disable. This causes the "Unknown register value". Inverters come from factory with Nowork values. Or add some flag to ignore those Nowork values?
Definition follows:
00 Nowork
01 Nowork
10 Disable
11Enable

bsafh, 2024-07-01:
`10:26:01 WARNING device_type: Unknown register value 6. Consider extending the definition with a PR. `

Well, I do know the model:
Deye SUN-10K-SG01HP3-EU-AM2

Should that not be included in the code?
I don't know how to fix that, that's why I reported it here, see opening posting of this issue.

"Or add some flag to silence those Nowork values?"
Sounds like a good idea.

rixxxx, 2024-07-01:
Device type is not part of my last messages, this is different issue.

rixxxx, 2024-07-01:
I think the documentation about register 0 is wrong. 
0X0200 组串机 inverter
0X0300 单相储能机 hybird
0X0400 微逆机 MI microinverter
0X0500 低压三相储能机 phase3 hybird
0X0600 高压三相储能机
 0600 三相高压 6-15kw
 0601 三相高压 20-50kw

My SUN-20K-SG01HP3-EU-AM2 has value 6 too. Maybe we should drop the sensor at all. Users probably know what kind of inverter they have.

bsafh, 2024-07-01:
> I think the documentation about register 0 is wrong. 0X0200 组串机 inverter 0X0300 单相储能机 hybird 0X0400 微逆机 MI microinverter 0X0500 低压三相储能机 phase3 hybird 0X0600 高压三相储能机 0600 三相高压 6-15kw 0601 三相高压 [...]

Why drop it?

`0X0600 高压三相储能机`
is in the list,  just the englisch translation is messing. Google translates it to 
`High voltage three-phase energy storage machine`, wich seem to be  correct.

So type 6 is documented, just not translated to englisch. 

> I think the documentation about register 0 is wrong.

I don't think so ...
`0X0600 高压三相储能机 three phase hybrid high voltage` 
documentation is just not totally translated to englisch.

rixxxx, 2024-07-02:
Its just wild guess, but maybe 0x0600 means 6, which means 3PH HV. 600 and 601 are wrong fore sure since your 10kW and my 20kW inverters have the same value.

rixxxx, 2024-07-02:
#310 fixes inverter type for 3ph inverters based on values seen on the field.

bsafh, 2024-07-03:
After updating to the new version I now get 

```
|    |        battery_wake_up, gen_l1_power, prog1_time, battery_float_voltage,         |
|    |       battery_1_power, grid_peak_shaving, grid_power, grid_phase_warning,        |
|    |       gen_l2_power, generator_ac_couple_frz_high, prog6_time, prog5_time,        |
|    |             grid_charge_enabled, generator_charge_start_battery_soc,             |
|    |        grid_charge_battery_current, use_timer, battery_restart_capacity,         |
|    |   battery_restart_voltage, grid_side___inside_total_apparent_power, pv1_power,   |
|    |     fan_warning, prog1_charge, min_pv_power_for_gen_start, gen_peak_shaving,     |
|    | prog2_power, prog_saturday_enabled, max_solar_power, configured_grid_frequency,  |
|    |                    grid_l1_power, battery_equalization_hours,                    |
|    |            parallel_communication_quality_warning, generator_on_soc,             |
|    |         generator_charge_enabled, prog_thursday_enabled, prog1_capacity,         |
|    |   battery_1_bms_other_flag___battery_2_force_charge, external_relay, auto_dim,   |
|    |    prog_friday_enabled, am_pm, drm, battery_shutdown_voltage, prog5_capacity,    |
|    |                  grid_ct_power, inverter_power, prog4_capacity                   |
| 10 |      battery_1_bms_hardware_version, grid_connected, battery_1_bms_voltage,      |
|    |       load_l2_voltage, battery_1_bms_discharging_voltage, grid_l3_voltage,       |
|    | battery_1_temperature, battery_1_bms_max_discharge_current_limit, gen_l3_volts,  |
|    |      overall_state, battery_1_bms_type, load_l3_voltage, battery_1_bms_soc,      |
|    |     inverter_frequency, battery_1_bms_temperature, battery_1_bms_alarm_flag,     |
|    |       battery_1_bms_charging_voltage, grid_l2_voltage, battery_1_bms_soh,        |
|    |   battery_1_soc, load_l1_voltage, inverter_voltage, pv2_current, pv1_voltage,    |
|    |            battery_1_current, dry_contact2_status, grid_relay_status,            |
|    |    battery_1_bms_fault_flag, battery_1_bms_software_version, grid_frequency,     |
|    |      dc_transformer_temperature, gen_l2_volts, undefined_load_relay_status,      |
|    |          inv_relay_status, grid_give_power_to_relay_status, sd_status,           |
|    |         battery_1_bms_discharging_current_limit, battery_1_bms_rated_ah,         |
|    |   battery_1_bms_current, battery_1_bms_max_charge_current_limit, pv2_voltage,    |
|    |   generator_relay_status, pv1_current, battery_1_voltage, fault, gen_l1_volts,   |
|    |         battery_charge_efficiency, battery_1_bms_charging_current_limit,         |
|    |     dry_contact1_status, radiator_temperature, grid_l1_voltage, grid_current     |
| 60 |        date_time, day_active_energy, total_grid_import, total_pv_energy,         |
|    |  day_gen_energy, total_active_energy, day_grid_import, total_battery_discharge,  |
|    |  day_load_energy, total_battery_charge, day_battery_discharge, day_grid_export,  |
|    |     total_grid_export, total_load_energy, day_battery_charge, day_pv_energy      |
+----+----------------------------------------------------------------------------------+
17:50:31 INFO    Report every (inverter >1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  10 |           grid_side___inside_total_apparent_power, inverter_l1_power,            |
|     |     total_active_power_from_side_to_side_of_the_grid, gen_power, pv1_power,      |
|     |     grid_l3_power, grid_l1_power, pv_power, inverter_l2_power, gen_l3_power,     |
|     |    grid_l2_power, b_phase_power_on_the_inner_side_of_the_grid, load_l1_power,    |
|     |    load_l2_power, c_phase_power_on_the_inner_side_of_the_grid, gen_l1_power,     |
|     | a_phase_power_on_the_inner_side_of_the_grid, inverter_l3_power, battery_1_power, |
|     |  load_power, load_l3_power, grid_power, gen_l2_power, pv2_power, grid_ct_power,  |
|     |                                  inverter_power                                  |
|  30 |      export_limit_power, battery_type, prog_sunday_enabled, prog2_capacity,      |
|     |   prog_tuesday_enabled, solar_export, battery_capacity_current, prog6_charge,    |
|     |             battery_max_discharge_current, system_zero_export_power,             |
|     |             loss_of_lithium_battery_report_fault, battery_1_wake_up,             |
|     | generator_charge_battery_current, prog_wednesday_enabled, generator_port_usage,  |
|     |  time_synchronization, battery_resistance, grid_peak_shaving_power, prog4_time,  |
|     |      battery_low_voltage, battery_equalization_voltage, beep, prog4_charge,      |
|     |        configured_grid_phases, prog1_power, prog6_power, prog6_capacity,         |
|     |     battery_max_charge_current, grid_signal_on, prog3_power, prog3_capacity,     |
|     |           load_limit, generator_cooling_time, lithium_hv_bms_protocol,           |
|     |            battery_absorption_voltage, prog6_voltage, grid_standard,             |
|     | grid_charge_start_battery_soc, prog_monday_enabled, prog4_voltage, prog5_charge, |
|     | battery_low_capacity, generator_off_soc, prog_time_of_use_enabled, prog2_charge, |
[... cut ...]
|     |           battery_1_bms_charging_current_limit, total_battery_charge,            |
|     |       radiator_temperature, day_grid_export, grid_l1_voltage, grid_current       |
| 300 |        battery_1_manufacturer, lithium_battery_loss_warning, device_type,        |
|     |     grid_phase_warning, parallel_communication_quality_warning, fan_warning      |
+-----+----------------------------------------------------------------------------------+
17:50:33 ERROR   NotImplementedError in read_wr_deye_1: 
17:50:33 ERROR   NotImplementedError in read_wr_deye_2: 
```

bsafh, 2024-07-03:
See as well https://github.com/kellerza/sunsynk/commit/bc65f2a796e556fcd86fb309f6c6b9fd20934ce9

bsafh, 2024-07-03:
The full register list seems to be 
[MODBUSRTU三相高压储能通信规约V104（高压）(3)(1)1111_SG01-HP3-AM2.pdf](https://github.com/user-attachments/files/16090118/MODBUSRTU.V104.3.1.1111_SG01-HP3-AM2.pdf)
in case that helps

rixxxx, 2024-07-03:
It doesn't help. I have it already.

What was the last working version?

bsafh, 2024-07-03:
The one from just before.

kellerza, 2024-09-06:
If you look at the 3ph definition file on github, you can click on “blame” and see what lines were change in which commits

you can also look at all commits with that number and see what comes just after

kellerza, 2025-01-16:
There has been several updates to the three phase definitions, so going to close this one

closed 2025-01-16 by kellerza