kellerza/sunsynk #311: Sunsynk/Deye Inverter Add-on (edge/dev) bc65f2a  does not read any values, but gives errors

bsafh, 2024-07-03:
### Issue related to

Sunsynk/Deye Inverter Add-on (edge/dev)
Current version: bc65f2a 

### Describe the issue/bug

gives error on startup

**Expected behavior**
continue to read values

### Your environment

- Home Assistant version: 2024.6.4
- Addon:
  - Name: Sunsynk/Deye Inverter Add-on (edge/dev)
  - Version: bc65f2a 
- Inverter:
  - Make: Deye
  - Model: SUN-SG01HP3-EU-AM2
  - Firmware: \_\_
- Adaptor details:
  - RS485/RS232 Waveshare

### You configuration

```yaml

DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "2312122xxx"
    HA_PREFIX: WR_Deye_1
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: serial-tcp://192.168.80.44:8234
  - SERIAL_NR: "2312126yyy"
    HA_PREFIX: WR_Deye_2
    MODBUS_ID: 2
    DONGLE_SERIAL_NUMBER: "0"
    PORT: serial-tcp://192.168.80.45:8234
SENSOR_DEFINITIONS: three-phase-hv
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
  - settings
  - all
  - pv1_power
  - essential_power
  - battery_power
  - battery_soc
  - non_essential_power
  - grid_ct_power
  - Inverter power
  - Inverter L1 power
  - Inverter L2 power
  - Inverter L3 power
  - Inverter voltage
  - Inverter frequency
  - Grid frequency
  - Grid power
  - Grid L1 power
  - Grid L3 power
  - Grid L2 power
  - Grid L2 voltage
  - Grid L1 voltage
  - Grid L3 voltage
  - Load power
  - Load L1 power
[... 318 lines cut ...]
|     |   total_battery_discharge, battery_1_bms_charging_current_limit, pv1_voltage,    |
|     |            total_load_energy, grid_l2_voltage, day_battery_discharge,            |
|     |     battery_1_bms_max_charge_current_limit, inverter_frequency, pv1_current,     |
|     | day_active_energy, gen_l2_volts, day_pv_energy, grid_frequency, load_l3_voltage, |
|     |      gen_l1_volts, battery_1_bms_max_discharge_current_limit, pv2_voltage,       |
|     |            battery_1_bms_rated_ah, day_grid_import, grid_l3_voltage,             |
|     |    dc_transformer_temperature, pv2_current, total_pv_energy, day_load_energy,    |
|     | grid_current, battery_1_voltage, battery_1_soc, battery_1_bms_soc, gen_l3_volts, |
|     |   battery_charge_efficiency, battery_1_bms_temperature, radiator_temperature,    |
|     |  battery_1_bms_soh, load_l1_voltage, battery_1_bms_voltage, total_grid_export,   |
|     |  inverter_voltage, battery_1_bms_discharging_current_limit, total_grid_import,   |
|     |     battery_1_temperature, battery_1_bms_charging_voltage, grid_l1_voltage,      |
|     |            battery_1_bms_current, battery_1_current, day_grid_export             |
| 300 |                    lithium_battery_loss_warning, fan_warning,                    |
|     |     parallel_communication_quality_warning, device_type, grid_phase_warning,     |
|     |                              battery_1_manufacturer                              |
+-----+----------------------------------------------------------------------------------+
14:20:15 ERROR   NotImplementedError in read_wr_deye_2: 
14:20:15 ERROR   NotImplementedError in read_wr_deye_1: 
14:25:00 ERROR   (28 in 5 min) NotImplementedError in read_wr_deye_2: 
14:25:00 ERROR   (28 in 5 min) NotImplementedError in read_wr_deye_1: 
14:30:00 ERROR   (30 in 5 min) NotImplementedError in read_wr_deye_2: 
14:30:00 ERROR   (30 in 5 min) NotImplementedError in read_wr_deye_1: 

```

bsafh, 2024-07-03:
After a complete reboot I now get

```
17:09:24 ERROR   Unknown sensor specified: year_grid_export
17:09:24 ERROR   Unknown sensor specified: year_load_energy
17:09:24 ERROR   Unknown sensor specified: year_pv_energy
17:09:24 ERROR   Unknown sensor specified: environment_temperature
17:09:24 ERROR   Unknown sensor specified: control_mode
17:09:24 ERROR   Unknown sensor specified: generator_input
17:09:24 ERROR   Unknown sensor specified: battery_charging_voltage
17:09:24 ERROR   Unknown sensor specified: bat1_cycle
17:09:24 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
17:09:24 INFO    Connecting to serial-tcp://192.168.80.44:8234
17:09:24 INFO    Reading startup sensors Rated power, Battery Low Capacity, Battery Shutdown Capacity, Prog4 Time, Battery Restart Capacity, Prog6 Time, Prog3 Time, Battery Float voltage, Battery Shutdown voltage, Battery Low voltage, Serial, Prog5 Time, Prog2 Time, Prog1 Time
17:10:19 ERROR   timeout reading 5 registers from 3; timeout reading 2 registers from 20; timeout reading 1 registers from 101; timeout reading 6 registers from 115; timeout reading 6 registers from 148
17:11:14 ERROR   timeout reading 5 registers from 3; timeout reading 2 registers from 20; timeout reading 1 registers from 101; timeout reading 6 registers from 115; timeout reading 6 registers from 148
17:12:09 ERROR   timeout reading 5 registers from 3; timeout reading 2 registers from 20; timeout reading 1 registers from 101; timeout reading 6 registers from 115; timeout reading 6 registers from 148
17:12:10 WARNING Retrying individual sensors: ['rated_power', 'battery_low_capacity', 'battery_shutdown_capacity', 'prog4_time', 'battery_restart_capacity', 'prog6_time', 'prog3_time', 'battery_float_voltage', 'battery_shutdown_voltage', 'battery_low_voltage', 'serial', 'prog5_time', 'prog2_time', 'prog1_time']
17:12:21 ERROR   timeout reading 2 registers from 20 Rated power
17:12:32 ERROR   timeout reading 1 registers from 117 Battery Low Capacity
17:12:43 ERROR   timeout reading 1 registers from 115 Battery Shutdown Capacity
17:12:54 ERROR   timeout reading 1 registers from 151 Prog4 Time
17:13:05 ERROR   timeout reading 1 registers from 116 Battery Restart Capacity
17:13:16 ERROR   timeout reading 1 registers from 153 Prog6 Time
17:13:27 ERROR   timeout reading 1 registers from 150 Prog3 Time
17:13:39 ERROR   timeout reading 1 registers from 101 Battery Float voltage
17:13:50 ERROR   timeout reading 1 registers from 118 Battery Shutdown voltage
17:14:01 ERROR   timeout reading 1 registers from 120 Battery Low voltage
17:14:12 ERROR   timeout reading 5 registers from 3 Serial
17:14:23 ERROR   timeout reading 1 registers from 152 Prog5 Time
17:14:34 ERROR   timeout reading 1 registers from 149 Prog2 Time
17:14:45 ERROR   timeout reading 1 registers from 148 Prog1 Time
17:14:45 CRITICAL Could not read sensors: ['Rated power', 'Battery Low Capacity', 'Battery Shutdown Capacity', 'Prog4 Time', 'Battery Restart Capacity', 'Prog6 Time', 'Prog3 Time', 'Battery Float voltage', 'Battery Shutdown voltage', 'Battery Low voltage', 'Serial', 'Prog5 Time', 'Prog2 Time', 'Prog1 Time']
17:14:45 INFO    ############################################################
17:14:45 INFO    No response on the Modbus interface serial-tcp://192.168.80.44:8234, see https://kellerza.github.io/sunsynk/guide/fault-finding
17:14:45 INFO    ############################################################
17:14:45 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.
17:15:16 ERROR   Task was destroyed but it is pending!
task: <Task pending name='ConnKeeper' coro=<PySolarmanV5Async._conn_keeper() running at /usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5_async.py:154> wait_for=<Future pending cb=[Task.task_wakeup()]>>
[17:15:16] INFO: Service restart after closing
17:15:17 INFO    Loading configuration: /data/options.json
17:15:17 INFO    sunsynk library version: 0.6.4
17:15:17 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |  *  |  60  |   60   |           |          |     1      |
|     rw    |  *  |  5   |   30   |           |          |     1      |
|    enum   |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  5   |   10   |     50    |    1     |            |
|    kwh    |  *  |  60  |  120   |           |          |            |
|  any_unit |  *  |  10  |  120   |           |          |            |
|  no_unit  |  *  |  10  |   60   |           |          |     1      |
+-----------+-----+------+--------+-----------+----------+------------+
17:15:17 INFO    Using three phase HV sensor definitions.
17:15:17 ERROR   Unknown sensor specified: aux_power
17:15:17 ERROR   Your config includes deprecated sensors. Replace battery_current with battery_1_current
17:15:17 ERROR   Your config includes deprecated sensors. Replace battery_power with battery_1_power
17:15:17 ERROR   Your config includes deprecated sensors. Replace battery_soc with battery_1_soc
17:15:17 ERROR   Unknown sensor specified: battery_voltage
17:15:17 ERROR   Unknown sensor specified: essential_power
17:15:17 ERROR   Unknown sensor specified: grid_voltage
17:15:17 ERROR   Unknown sensor specified: inverter_current
[... cut ...]
17:15:17 ERROR   Unknown sensor specified: battery_charging_voltage
17:15:17 ERROR   Unknown sensor specified: bat1_cycle
17:15:17 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
17:15:17 INFO    Connecting to serial-tcp://192.168.80.44:8234
17:15:17 INFO    Reading startup sensors Battery Float voltage, Battery Shutdown Capacity, Rated power, Prog6 Time, Battery Restart Capacity, Prog3 Time, Battery Low voltage, Serial, Battery Shutdown voltage, Prog1 Time, Prog4 Time, Battery Low Capacity, Prog5 Time, Prog2 Time
```

bsafh, 2024-07-03:
I have no idea why the communication fails. The serial-tcp parameters are still the same and the Waveshare RS485-ETH Box is working, I restarted the Waveshare Box the inverter and the HomeAssistant VM.

Still no luck.

Worked finde until the update to this dev/edge version. 

What can I do to debug?

bsafh, 2024-07-03:
See also https://github.com/kellerza/sunsynk/commit/bc65f2a796e556fcd86fb309f6c6b9fd20934ce9

kellerza, 2024-07-04:
`NotImplementedError` was a bug introduced in e353fc60cf7b21510be69258a794a8ab531c11fd. Should be fixed in 3a882cd71ecfd7d4ceec85ef2ad36c5df0fbdd5d (with some better error logs)

Update the dev/edge addon to at least version 2cec340 (latest build that include device_type for single phase inverters)

bsafh, 2024-07-04:
> Update the dev/edge addon to at least version [2cec340](https://github.com/kellerza/sunsynk/commit/2cec340c75bf30f339e8e17ea6bff74de3c22b3f) (latest build that include device_type for single phase inverters)

done

seems to start up fine

```
|    |     configured_grid_frequency, external_relay, battery_wake_up, prog6_time,      |
|    |        battery_shutdown_capacity, gen_peak_shaving, prog_friday_enabled,         |
|    |        inverter_l1_power, solar_export, prog_tuesday_enabled, load_limit,        |
|    |         load_l1_power, generator_charge_battery_current, grid_l1_power,          |
|    |      c_phase_power_on_the_inner_side_of_the_grid, prog_time_of_use_enabled,      |
|    |            prog2_time, battery_max_discharge_current, prog1_capacity,            |
|    | lithium_hv_bms_protocol, prog5_time, grid_peak_shaving, battery_1_manufacturer,  |
|    |    device_type, b_phase_power_on_the_inner_side_of_the_grid, battery_1_power,    |
|    |      battery_type, am_pm, generator_off_soc, generator_on_soc, prog5_power,      |
|    |       prog1_voltage, gen_l3_power, priority_load, grid_peak_shaving_power,       |
|    |               prog3_capacity, beep, grid_standard, load_l2_power,                |
|    | battery_1_bms_other_flag___battery_2_force_charge, generator_max_operating_time, |
|    |    prog1_power, pv1_power, grid_ct_power, grid_charge_enabled, gen_l2_power,     |
|    |    battery_low_capacity, prog4_charge, load_power, generator_charge_enabled,     |
|    |                            battery_absorption_voltage                            |
| 10 |    overall_state, inverter_frequency, dry_contact1_status, grid_relay_status,    |
|    | load_l2_voltage, pv1_current, load_l1_voltage, dry_contact2_status, pv2_current, |
|    |         battery_1_bms_charging_voltage, battery_1_bms_hardware_version,          |
|    | undefined_load_relay_status, inv_relay_status, grid_l3_voltage, grid_connected,  |
|    |       battery_1_bms_max_discharge_current_limit, inverter_voltage, fault,        |
|    |  battery_1_bms_discharging_voltage, sd_status, battery_1_bms_software_version,   |
|    |    radiator_temperature, battery_1_bms_type, gen_l1_volts, battery_1_current,    |
|    |              battery_1_bms_discharging_current_limit, pv2_voltage,               |
|    |           grid_give_power_to_relay_status, battery_charge_efficiency,            |
|    | battery_1_bms_fault_flag, load_l3_voltage, battery_1_bms_soc, battery_1_bms_soh, |
|    |    pv1_voltage, battery_1_temperature, battery_1_bms_charging_current_limit,     |
|    |           battery_1_bms_current, generator_relay_status, gen_l3_volts,           |
|    |  battery_1_bms_rated_ah, grid_current, battery_1_bms_alarm_flag, gen_l2_volts,   |
|    |     battery_1_bms_temperature, dc_transformer_temperature, grid_l2_voltage,      |
|    |    grid_l1_voltage, grid_frequency, battery_1_bms_voltage, battery_1_voltage,    |
|    |              battery_1_soc, battery_1_bms_max_charge_current_limit               |
| 60 |         total_grid_import, total_pv_energy, day_load_energy, date_time,          |
|    |      day_active_energy, day_gen_energy, day_grid_export, total_grid_export,      |
|    |            day_grid_import, total_load_energy, total_battery_charge,             |
|    |           total_battery_discharge, day_battery_charge, day_pv_energy,            |
|    |                    day_battery_discharge, total_active_energy                    |
+----+----------------------------------------------------------------------------------+
11:04:29 INFO    Report every (inverter >1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  10 |        grid_l3_power, grid_power, grid_l2_power, load_l3_power, pv_power,        |
|     |   b_phase_power_on_the_inner_side_of_the_grid, gen_l1_power, battery_1_power,    |
|     |      gen_l3_power, grid_side___inside_total_apparent_power, inverter_power,      |
|     |      load_l2_power, inverter_l1_power, pv2_power, load_l1_power, gen_power,      |
|     |                     pv1_power, grid_l1_power, grid_ct_power,                     |
|     |       total_active_power_from_side_to_side_of_the_grid, inverter_l3_power,       |
|     |            c_phase_power_on_the_inner_side_of_the_grid, gen_l2_power,            |
|     |    a_phase_power_on_the_inner_side_of_the_grid, load_power, inverter_l2_power    |
|  30 |     min_pv_power_for_gen_start, battery_1_bms_other_flag___battery_1_sleep,      |
|     |  generator_charge_start_battery_soc, battery_restart_capacity, grid_signal_on,   |
|     |          battery_float_voltage, configured_grid_phases, prog6_capacity,          |
|     |      grid_charge_start_battery_soc, battery_1_wake_up, prog_monday_enabled,      |
|     |        gen_signal_on, battery_equalization_hours, prog_wednesday_enabled,        |
|     |     export_limit_power, generator_ac_couple_frz_high, prog_saturday_enabled,     |
|     |   use_timer, prog3_time, prog1_charge, prog5_charge, parallel_battery_1_and_2,   |
|     |     prog_thursday_enabled, time_synchronization, prog3_voltage, prog2_power,     |
|     |         prog6_charge, prog5_voltage, drm, prog3_charge, prog2_capacity,          |
[... cut ...]
|     | dc_transformer_temperature, grid_l2_voltage, day_active_energy, grid_l1_voltage, |
|     |    day_grid_export, grid_frequency, battery_1_bms_voltage, battery_1_voltage,    |
|     |                       battery_1_soc, day_battery_discharge                       |
| 300 |           grid_phase_warning, parallel_communication_quality_warning,            |
|     |  battery_1_manufacturer, device_type, fan_warning, lithium_battery_loss_warning  |
+-----+----------------------------------------------------------------------------------+
```

no errors!

[image]

Thanks!

closed 2024-09-06 by kellerza