kellerza/sunsynk #312: a lot of errormasseges in the log file Register 59 and 219

concierge-at, 2024-07-03:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions.py)

### You Inverter Deye5kSg03LP1

Details about your Inverter: make/model/firmware
Deye 5ksg03lp1 
Protokoll-Version：V0.2.0.1NETZ：3384-1515HMI：0000-C367Lithium-Akku-Versionsnummer：V0.0.0.0Deutsche Version：FFFFSpanische Version：FFFFUngarische Version：FFFFItalienische Version：FFFFLCD-Typ：8100

### Describe the issue/bug and what you expect

Hello, I haven't been here long, and homeassistant is very powerful, the integration of Solarman Yellow with mqtt works very well.  I looked through the logs and there were a lot of errors

`08:11:18 ERROR   Error retry attempt(0): Connection already closed
08:15:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
08:34:48 ERROR   Error retry attempt(0): Connection already closed
08:35:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
08:41:21 ERROR   Error retry attempt(0): Connection already closed
08:45:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
08:59:27 ERROR   Error retry attempt(0): Connection already closed
09:00:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
09:01:48 ERROR   Error retry attempt(0): Connection already closed
09:05:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
09:32:13 ERROR   Error retry attempt(0): Connection already closed
09:35:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
09:38:58 ERROR   Error retry attempt(0): Connection already closed
09:40:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
09:41:18 ERROR   Error retry attempt(0): Connection already closed
09:43:31 ERROR   Error retry attempt(0): Connection already closed
09:45:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
09:45:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
09:45:58 ERROR   Error retry attempt(0): Connection already closed
09:50:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
09:52:48 ERROR   Error retry attempt(0): Connection already closed
09:55:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
09:55:08 ERROR   Error retry attempt(0): Connection already closed
09:57:31 ERROR   Error retry attempt(0): Connection already closed
10:00:00 ERROR   (2 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
10:08:48 ERROR   Error retry attempt(0): Connection already closed
10:10:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
10:11:01 ERROR   Error retry attempt(0): Connection already closed
10:13:21 ERROR   Error retry attempt(0): Connection already closed
10:15:00 ERROR   (2 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
10:15:48 ERROR   Error retry attempt(0): Connection already closed
10:20:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
10:22:33 ERROR   Error retry attempt(0): Connection already closed
10:25:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
11:06:43 ERROR   Error retry attempt(0): Connection already closed
11:10:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
11:38:17 ERROR   Error retry attempt(0): Connection already closed
11:40:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
11:40:31 ERROR   Error retry attempt(0): Connection already closed
11:45:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
12:03:02 ERROR   Error retry attempt(0): Connection already closed
12:05:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
12:09:42 ERROR   Error retry attempt(0): Connection already closed
12:10:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
12:18:58 ERROR   Error retry attempt(0): Connection already closed
12:20:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
12:45:11 ERROR   Error retry attempt(0): Connection already closed
12:47:33 ERROR   Error retry attempt(0): Connection already closed
12:50:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
12:50:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
12:54:28 ERROR   Error retry attempt(0): Connection already closed
12:55:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
12:56:48 ERROR   Error retry attempt(0): Connection already closed
12:59:08 ERROR   Error retry attempt(0): Connection already closed
13:00:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
13:00:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
13:55:41 ERROR   Error retry attempt(0): Connection already closed
14:00:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
[... 30 lines cut ...]
17:12:57 ERROR   Error retry attempt(0): Connection already closed
17:15:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
17:15:18 ERROR   Error retry attempt(0): Connection already closed
17:20:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
17:28:56 ERROR   Error retry attempt(0): Connection already closed
17:30:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
17:31:27 ERROR   Error retry attempt(0): Connection already closed
17:33:48 ERROR   Error retry attempt(0): Connection already closed
17:35:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
17:35:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 219 poll_need_to_read
17:52:03 ERROR   Error retry attempt(0): Connection already closed
17:55:00 ERROR   (1 in 5 min) OSError in read_ss: timeout reading 1 registers from 59 poll_need_to_read
18:05:47 ERROR   Error retry attempt(0): Connection already closed`

A clear and concise description of what the bug is.

Any suggestions on proposed registers is welcome.

### Logs (if applicable)

```logs

paste logs in 3x backticks...

```

rixxxx, 2024-07-03:
Both Cloud and Sunsynk are fighting for the same serial port. Solarman dongle is not doing any session management and every time both Cloud and Sunsynk are asking something they are tipping on each others toes. If you disable cloud connectivity, it gets a bit better. Ultimately I suggest to use RS485 cable.

concierge-at, 2024-07-03:
Thank you, can there something happen on this errors? Or can it run normaly,

I have try one year to get data out of the inverter with cable i wara happy fir this integration

rixxxx, 2024-07-03:
It works but sunsynk sometimes hangs and Solarman dongle also sometimes hangs. I finaly gave up and installed a cable, some others have done the same. There are probably not very many sunsynk users using the Solarman dongle.

concierge-at, 2024-07-04:
Thank you verry much, does you have please spezific information from usb to deye rs485? I tried many times to get Esp32 to rs485 geting working for me without success so i wars verry  happy after one year of trying to find this with the dongle. And in austria you dont get nothing ready to use items
Thank you for your help and information

rixxxx, 2024-07-04:
https://kellerza.github.io/sunsynk/guide/deployment-options lists multiple options. I'm using 4th without mbusd. Easyest would be to use option 1 or 4 and have USB RS485 dongle directly connected to the machine running sunsynk.

You can get USB RS485 adapter very cheap from eg AliEx..., search for "usb rs485". These ~2EUR adapters work for sure

concierge-at, 2024-07-05:
Thank you @rixxxx ,
please sorry my lot of questions.
I am not a programmer, and an Automation worker. Thank you for your Help. 
But there are 2 questions for now on you.
1. Will the entities name the same when i use usb?  I have epend a lot of time in the Dashboard to get working.
2. And the 2.nd how does i say the sunsynk integration that there
3. Will be now a usb connection and how to delete the dongle?
4. Thank you and have a nice day. Sorry for my english its not native language of me

kellerza, 2024-07-05:
> 1. Will the entities name the same when i use usb?  I have epend a lot of time in the Dashboard to get working.

Yes, as long as `HA_PREFIX` and the serial number is the same. These two form the unique ID of the MQTT sensor and HA will use the same entity name. (not add a _2 to the end).

> 2. And the 2.nd how does i say the sunsynk integration that there
> 3. Will be now a usb connection and how to delete the dongle?

Change the driver - https://kellerza.github.io/sunsynk/reference/multi-options#driver

solarman to pymodbus

concierge-at, 2024-07-05:
Thank you thank you, i will try it in the nex week i think a lot of work for me.
If i have no succsess i will come again. And if it works we take a coffee🫣 Thank you

concierge-at, 2024-08-04:
Hello everyone, Apparently I'm the only one who really can't get it right. Yesterday I connected the Waveshare rs485-usb adapter to the Deye 5ksg03lp1 and was immediately fired up because it worked straight away. But there has only been radio silence on the USB for 2 hours since then and no more data is being sent.

I have now switched back to the WiFi dongle so that I at least have data. What did I do wrong again this time?

`12:04:02 INFO    ############################################################
12:04:02 INFO                 Inverter serial number '****2469'
12:04:02 INFO    ############################################################
12:04:02 INFO    MQTT: Connecting to mqtt-user@core-mosquitto:1883
12:04:02 INFO    MQTT: Connection successful
12:04:03 INFO    Read every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  5  | prog3_capacity, load_limit, prog5_charge, prog1_time, prog4_capacity, use_timer, |
|     |     prog4_power, prog5_power, prog3_charge, prog5_capacity, prog6_capacity,      |
|     |    prog2_charge, battery_low_capacity, prog3_power, prog6_power, prog5_time,     |
|     |       prog2_capacity, prog4_time, prog2_power, prog3_time, prog1_capacity,       |
|     |      generator_input, prog4_charge, prog6_time, prog1_power, prog6_charge,       |
|     |                     prog1_charge, priority_load, prog2_time                      |
|  15 |     battery_power, grid_voltage, grid_frequency, pv2_power, essential_power,     |
|     |   grid_ct_power, battery_soc, non_essential_power, dc_transformer_temperature,   |
|     |     overall_state, load_frequency, battery_current, pv2_current, load_power,     |
|     |      grid_power, pv1_voltage, pv1_current, inverter_power, grid_connected,       |
|     |       battery_voltage, pv1_power, inverter_current, pv2_voltage, aux_power       |
| 300 |       total_pv_energy, day_load_energy, day_pv_energy, total_grid_export,        |
|     |    day_battery_charge, total_grid_import, total_load_energy, day_grid_import,    |
|     |         total_battery_discharge, day_battery_discharge, day_grid_export,         |
|     |                               total_battery_charge                               |
+-----+----------------------------------------------------------------------------------+
12:04:03 INFO    Report every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  60 | battery_power, non_essential_power, load_power, pv1_power, aux_power, pv2_power, |
|     |            essential_power, grid_ct_power, grid_power, inverter_power            |
| 300 |    total_pv_energy, prog3_capacity, grid_voltage, grid_frequency, load_limit,    |
|     |        prog5_charge, prog1_time, prog4_capacity, use_timer, prog4_power,         |
|     |       total_battery_discharge, prog5_power, prog3_charge, day_grid_export,       |
|     |           battery_soc, prog1_charge, day_pv_energy, total_grid_export,           |
|     |    dc_transformer_temperature, prog5_capacity, prog6_capacity, prog2_charge,     |
|     |        battery_low_capacity, prog3_power, total_load_energy, prog6_power,        |
|     |     load_frequency, battery_current, overall_state, pv2_current, prog5_time,     |
|     |  prog2_capacity, day_battery_charge, prog4_time, day_grid_import, pv1_current,   |
|     |    pv1_voltage, grid_connected, prog2_power, battery_voltage, prog1_capacity,    |
|     |       total_battery_charge, prog3_time, generator_input, day_load_energy,        |
|     |   prog4_charge, inverter_current, prog6_time, pv2_voltage, total_grid_import,    |
|     |   prog1_power, prog6_charge, day_battery_discharge, priority_load, prog2_time    |
+-----+----------------------------------------------------------------------------------+
12:12:51 WARNING Failed to connect [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
12:12:51 WARNING Failed to connect [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
12:12:52 WARNING Failed to connect [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
12:12:53 WARNING Failed to connect [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
12:12:54 WARNING Failed to connect [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
12:12:55 WARNING Failed to connect [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
12:12:55 WARNING Failed to connect [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
12:12:55 WARNING Failed to connect [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
12:12:55 WARNING Failed to connect [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
[... cut ...] 1 registers from 219; timeout reading 1 registers from 235; timeout reading 2 registers from 243; timeout reading 14 registers from 248; timeout reading 12 registers from 268 poll_need_to_read``

`
- SERIAL_NR: "230xxxxx69"
  HA_PREFIX: SS
  MODBUS_ID: 1
  PORT: /dev/ttyUSB1
`
dies habe ich im addon als verbindung angegeben

I'm slowly getting desperate. I wanted to get away from the dongle but it doesn't seem to be as easy as I imagined.

Thank you again for your help and greetings from Austria

kellerza, 2024-08-04:
IT does not like your port.

If you set the port as an empty string - `""`, then it should use the debug device you choose, at the botton of the options you have this:

[image]

concierge-at, 2024-08-04:
Thanks again for your help, I changed it and then values ​​come, not all of them but some come.

I don't know why it just doesn't work for me. I'll just have to live with the dongle error because it's just unstable.

In any case, it makes me cry, anything that is faster than a donkey is no longer for me

jeybee74, 2025-04-03:
Did you get it to work?

concierge-at, 2025-04-22:
Good morning, did I get it to work? Not sure, more like a limp.

I tried a different approach, which worked fine until two days ago. Since I can't access the inverter (it's a few hundred kilometers away from me), I reactivated the Solarman integration via the Deye dongle.

it works but i have to ask Mr. Keller what this mean in der LOG

`[16:12:49] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
16:12:54 INFO    Loading config: /data/options.json
16:12:54 INFO    sunsynk library version: 0.8.2
16:12:54 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|    enum   |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  15  |   60   |    80.0   |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
16:12:54 INFO    Using Single phase sensor definitions.
16:12:54 ERROR   Unknown sensor specified: radiator_temperatur
16:12:54 INFO    Added hidden sensors as other sensors depend on it: Device type, Protocol, Serial, Rated power
16:12:54 INFO    Connecting to tcp://192.168.0.134:8899
16:12:54 INFO    Reading startup sensors Device type, Protocol, Serial
16:12:54 INFO    ############################################################
16:12:54 INFO                 Inverter serial number '****2469'
16:12:54 INFO    ############################################################
16:12:54 INFO    Device type: Single phase hybrid, using the single-phase sensor definitions
16:12:54 INFO    Protocol version: 2.1
16:12:54 INFO    Reading all sensors Device type, Protocol, Serial, PV2 power, PV2 current, PV2 voltage, Generator input, DC transformer temperature, Load power, Total Load Energy, AUX power, Battery current, Battery power, Battery SOC, Battery voltage, Day Battery Charge, Day Battery Discharge, Day Grid Export, Day Grid Import, Day Load Energy, Day PV Energy, Essential power, Grid Connected, Grid CT power, Grid frequency, Grid L2 power, Grid power, Grid voltage, Inverter current, Inverter power, Load frequency, Load L1 power, Load L2 power, Non-Essential power, Overall state, Priority Load, PV1 current, PV1 power, PV1 voltage, Use Timer, Total Battery Charge, Total Battery Discharge, Total Grid Export, Total Grid Import, Total PV Energy, Load Limit, Prog1 Capacity, Prog1 charge, Prog1 power, Prog1 Time, Prog2 Capacity, Prog2 charge, Prog2 power, Prog2 Time, Prog3 Capacity, Prog3 charge, Prog3 power, Prog3 Time, Prog4 Capacity, Prog4 charge, Prog4 power, Prog4 Time, Prog5 Capacity, Prog5 charge, Prog5 power, Prog5 Time, Prog6 Capacity, Prog6 charge, Prog6 power, Prog6 Time, Date Time, Grid Charge Battery current, Grid Charge enabled, Solar Export, Battery Max Charge current, Battery Max Discharge current, Battery Shutdown Capacity, Battery Restart Capacity, Battery Low Capacity, Rated power
16:12:55 INFO    MQTT: Connecting to mqtt-user@core-mosquitto:1883
16:12:55 INFO    MQTT: Connection successful
16:12:56 INFO    Read every (inverter 1)
+-----+---------------------------------------------------------------------------------+
|  s  |                                     Sensors                                     |
+-----+---------------------------------------------------------------------------------+
|  5  |   prog5_power, grid_charge_enabled, prog1_time, prog2_capacity, prog5_charge,   |
|     |       prog1_capacity, use_timer, load_limit, prog4_capacity, prog2_charge,      |
|     |  solar_export, battery_low_capacity, prog2_power, grid_charge_battery_current,  |
|     |    prog4_power, prog4_time, prog5_time, prog3_charge, prog3_time, prog2_time,   |
|     |     prog6_time, prog5_capacity, prog3_capacity, prog4_charge, priority_load,    |
|     |            battery_max_discharge_current, prog3_power, prog1_charge,            |
|     |        battery_restart_capacity, prog6_charge, battery_shutdown_capacity,       |
|     |    prog6_capacity, generator_input, battery_max_charge_current, prog1_power,    |
|     |                             device_type, prog6_power                            |
[... cut ...]
the lines at the beginning look like a successful attempt, but the lines at the end seem frightening to me. 
I am just a user, not a programmer, and I would certainly have hundreds of questions for Mr. Keller.

Maybe someone here who knows something about this can take a look. I actually wanted to get rid of the Wi-Fi dongle, but so far I haven't found a permanent, stable solution that works for me.

Thank you for reading; perhaps there's a solution for this.

Thanks and best regards from Austria

concierge-at, 2025-04-29:
> Did you get it to work?

i think he has not or is not reading messages about that

kellerza, 2026-08-15:
If you get many errors, try using less sensors, maybe just 1/22 sensors and see if this makes any difference. 

It might alos be that some of these registers never respond (59 / 219).

The only way to fault find this would be to start with a very small set of sensors and build it up from there

closed 2026-08-15 by kellerza