kellerza/sunsynk #313: Optimal Read Batch Size & Allowable Gap for Solarman Dongle

alistairts, 2024-07-08:
What is the optimal batch size and allowable gap for the Solaman Dongle interface?
I am hitting read/write limitations within my dongle which causes it to reboot and ponder life for up to 10min, so I want to know what values are best for these settings?
For example, when moving the slider to set the battery SoC values, I land up selecting several incorrect values until I reach my desired value. But the interface has actually been writing each value in succession and I believe this is overloading the dongle.
Here are my latest settings FYI...

MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 30
SCHEDULES:
  - KEY: W
    READ_EVERY: 15
    REPORT_EVERY: 60
  - KEY: RW
    READ_EVERY: 15
    REPORT_EVERY: 60
    CHANGE_ANY: true
  - KEY: any_unit
    READ_EVERY: 30
    REPORT_EVERY: 60
    CHANGE_BY: 80
  - KEY: no_unit
    READ_EVERY: 15
    REPORT_EVERY: 60
    CHANGE_ANY: true

kellerza, 2024-07-08:
Do you have known SOC values? What if you set these via scripts, does it work any better?

I've never played around too much with the Solarman dongle... it's no fun waiting for something that ponders life for 10minutes at a time.

alistairts, 2024-07-08:
Yes the SoC value is know, my work around is to use the "box" number entity mode so I can type in the desired value. But I am still interested in the actual limitations of the dongle...
For my understanding as per my settings, the addon will read 30 registers per a cycle, then it will not read any registers for 2 cycles, then it will read the next 30 registers and repeat?
If so, how long does a read cycle take?
Despite having not used the dongle very much, have you any tips for improving the update time of the data read. I can't get less than 60s per update even though my setting states 15s.

man55, 2024-08-26:
I would also like to know the optimal settings when using the native Dongle. This is the most common method, since the key is already included with the most Deye Invertors and you do not need to puzzle over complex connection solutions. For example, I can’t even imagine how I can connect an inverter to a home server with HA via USB, they are located in different buildings and the length of the USB cable through several floors and the street would be at least 100 meters, which is incredible.

Today I put `READ_SENSORS_BATCH_SIZE = 8` because I see constant reading errors in the log with the message "connection closed".

Here is my Shedules:
```
- KEY: W
  READ_EVERY: 5
  REPORT_EVERY: 60
  CHANGE_ANY: true
- KEY: rw
  READ_EVERY: 15
  REPORT_EVERY: 60
  CHANGE_ANY: true
- KEY: any_unit
  READ_EVERY: 30
  REPORT_EVERY: 60
  CHANGE_ANY: true
- KEY: no_unit
  READ_EVERY: 5
  REPORT_EVERY: 60
  CHANGE_ANY: true
- KEY: "%"
  READ_EVERY: 30
  REPORT_EVERY: 300
  CHANGE_ANY: true
- KEY: date_time
  READ_EVERY: 60
  REPORT_EVERY: 3600
  CHANGE_ANY: false
```
In general, I don't see any particular problems, judging by the graph, the main sensors are updated every 5 seconds, sometimes a little more. And this is normal.

I also use a HA Ping integration with configured notifications to control the connection.

kellerza, 2024-09-06:
I don’t have any suggestion beyond what is documented

the callback stats can show improvement/not if you do play around with different schedules

all reads all sensors (batch-by-batch) before waiting for the next schedule. If this read for too long the the long callback stat should increase

mamontuka, 2025-04-06:
Thank you. Im check your proposed configuration, all works with much better response

closed 2026-08-16 by kellerza