kellerza/sunsynk #314: Signed Values

alistairts, 2024-07-09:
I have the following custom sensors for a 50kW HV system which I believe are signed integers. What operands should be added to the code to make it handle a signed integer?
Sensor(630, "Inverter current L1", AMPS, 0.01),
Sensor(631, "Inverter current L2", AMPS, 0.01),
Sensor(632, "Inverter current L3", AMPS, 0.01),
Sensor((633, 691), "Inverter power L1", WATT, -1),
Sensor((634, 692), "Inverter power L2", WATT, -1),
Sensor((635, 693), "Inverter power L3", WATT, -1),
Sensor((636, 694), "Inverter power", WATT, -1), 
When power flows from the battery to the load, the current and power display correctly; but when the power flows from grid to battery the current goes from ~6Amps to ~600Amps, and the power goes from ~10,000W to ~10,000,000,000W.

kellerza, 2024-07-09:
Negative factors indicate a signed value
I’m not sure if signed values with more that 1 register was ever really tested/added

What does the current sensor do if you add a negative? -0.01

alistairts, 2024-07-09:
Ahhah, I have changed the current sensor and it is now correct. But for the inverter power, it appears to have no effect.

kellerza, 2024-07-10:
> Ahhah, I have changed the current sensor and it is now correct. But for the inverter power, it appears to have no effect.

As mentioned, I believe it only works for single register values today. If you post some examples value and how it should look decoded it might help

kellerza, 2024-07-10:
> On the topic of custom sensors, how can I get the following MathSensor to work? I'm reading the total for the combined loads minus the total of smart loads, to get the total for essential loads. MathSensor((527, 528), (537, 538)), "Total Essential Loads Energy", KWH, factors=(0.1, -0.1)),

Mathsensors only do single register values, so this wn't work. You can always write your own class and implement register to value.

alistairts, 2024-07-15:
I recorded both the high word and low word values of my inverters internal power when my generator was running early hours of the morning, all as unsigned sensors. There is no sun at this point therefore my PV power was zero, but since the generator was charging the battery the inverter internal power will be negative
Generator (reg625) = 28,966.0W
Inverter output power (reg653) = 7,389.3W
Inverter internal power high (reg 694) = 65,535.0W
Inverter internal power low (reg636) = 43,959.3W
Will you be able to create the signed double word sensor?
I will be joining two 50kW inverters in parallel in the future, so my power measurements will definitely be exceeding the limitations of a 16 bit signed integer.

alistairts, 2024-07-19:
Are you able to create the new definition to handle 32bit signed integers?

heinrich321, 2024-08-19:
Hi @alistairts I think im sitting with the same issue now ... have 4 50kW in parallel. I think im getting overflow on some of my phases.

alistairts, 2024-08-19:
My work around this issue is to read both the high and low registers as a 32bit unsigned integer and then I created a template to convert it to signed. Here is my template for the total inverter power, and I did the same for each phases as well.

{% if states('sensor.inverter_inverter_power') | float < 2147483648 %}
{{ states('sensor.inverter_inverter_power') | float }}
{% else %}
{{ states('sensor.inverter_inverter_power') | float - 4294967296 }}
{% endif %}

 Note I customised the labelling of my sensors.
    Sensor((633, 691), "Inverter power L1", WATT, 1),
    Sensor((634, 692), "Inverter power L2", WATT, 1),
    Sensor((635, 693), "Inverter power L3", WATT, 1),
    Sensor((636, 694), "Inverter power", WATT, 1),

kellerza, 2024-09-05:
Hi, can try the latest edge version?
Hopefully this solved it - https://github.com/kellerza/sunsynk/commit/7cbb11c19201cbedd7a35eebbb39f7c9488f85e8

alistairts, 2024-10-15:
Hi Kellerza. Only today had I a chance to test and it has not worked.
[image]
I'm running your latest edge version.
[image]
My custom sensors are as follows:
    Sensor((633, 691), "Inverter power L1", WATT, -1),
    Sensor((634, 692), "Inverter power L2", WATT, -1),
    Sensor((635, 693), "Inverter power L3", WATT, -1),
    Sensor((636, 694), "Inverter power", WATT, -1),

kellerza, 2024-10-16:
Have a look at this commit https://github.com/kellerza/sunsynk/commit/7cbb11c19201cbedd7a35eebbb39f7c9488f85e8 & especially the test value. I'm sure these used your values. 

Looking at this now, it might not execute `signed` with 2 registers because of the if statement in sensors.py

kellerza, 2024-10-16:
Can you try the latest dev version? https://github.com/kellerza/sunsynk/commit/408482d5d0a0da1bc9c624a4876cd627b0bfa041

alistairts, 2024-10-18:
Thanks for your efforts! Your latest version has fixed the signed 32bit issue.

closed 2024-10-18 by kellerza