kellerza/sunsynk #316: Unable to connect:  Cannot enter letter in serial number

jasonkaisersmith, 2024-07-10:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

When trying to configure, the serial number field only accepts numeric values.  My serial number ends with a 'B', which is truncated.

As such, it fails to connect to the device and I am unable to retrieve any details.

**Expected behavior**

Serial number input must accept valid product serial numbers which can include alpha characters.

### Your environment

- Home Assistant version: 2024.7.2
- Addon:
  - Name: sunsynk-multi
  - Version: 2023-08-17
- Inverter:
  - Make: Sunsynk/Deye/Sol-Ark/...
  - Model: \_\_
  - Firmware: \_\_
- Adaptor details:
  - RS485/RS232/solarman
  - RS485/RS232 brand

### You configuration

```yaml

paste config in the block surrounded by 3x backticks

```

### Logs

```logs

paste logs in 3x backticks...

```

rixxxx, 2024-07-26:
Which inverter model?

jasonkaisersmith, 2024-07-26:
> Which inverter model?

Deye SUN-M80G4-EU-Q0

kellerza, 2024-08-04:
Can you be more specific, the serial number field seems to take a string

https://github.com/kellerza/sunsynk/blob/main/hass-addon-sunsynk-multi/config.yaml#L56

jasonkaisersmith, 2024-08-04:
[image]

During the initial configuration, I am only able to enter numeric values.  When I try to enter anything else then nothing gets entered, and the field turns red (or if I enter a mix, only numerics are accepted).

In the image, you can see on the right side of the "Device serial Number" field that there are "up/down" arrows which are a sign that it only accepts a numeric input.

Inspecting the HTML code in my browser, I can also see that the "inputmode" is set to "numeric"

`<input class="mdc-text-field__input" aria-labelledby="label" aria-controls="helper-text" aria-describedby="helper-text" type="number" placeholder="" required="" inputmode="numeric">`

kellerza, 2024-08-04:
It’s probably a different addon

see install instructions here: <https://kellerza.github.io/sunsynk/>

closed 2024-08-04 by kellerza

mikey0097, 2024-09-02:
I'm having this issue also. (so glad I'm not the only one)

I've tried installing this as a docker standalone and direct onto a pi running HA addon to try different computers.

The issue listed above is what I'm running into. My sunsynk wifi logger has letters in. But the DONGLE_SERIAL_NUMBER only accepts numbers.

Any input would be greatly appreciated.

kellerza, 2024-09-03:
This limit was removed

But it might very well be that you are running a newer dongle that will not work with this addon

mikey0097, 2024-09-03:
If I set the dongle serial number to a random set of numbers "1234" for example it pulls some data to start with.

As soon as I include any letters in the dongle serial number this addon won't let me save the config and when I do this via standalone docker it won't connect at all.

Really keen to try and get this working. Looks a fantastic integration that is actually what I'm looking for my sunsynk 5kw inverter.

kellerza, 2024-09-03:
Did you update to the latest version?

mikey0097, 2024-09-03:
Boom. Just updated to today's release. It appears to be working on a PI now.

I'll do some testing on it over the next few days.

I'll try to transition this back to my linux server running this within docker as well.

The PI is only a temporary setup.

Thanks very much for the quick fix :-D

kellerza, 2024-09-03:
If it's working, you're probably not using the **solarman** driver and do not need to specify **DONGLE_SERIAL_NUMBER**. 
It's optional and only used for **solarman**