kellerza/sunsynk #325: Sunsynk HA Add-on not adding the sensors

starrateit, 2024-08-13:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

I am sure this is something which I am doing wrong more than a bug but I cant understand what is causing it. 
I have installed the HA addon and required dependencies and I all appears to be working. 
But for some reason the sensor's are not being created.
So it is reading the registers seems to be connected to MQTT.

I have another setup running with the same settings which seem s to work fine and I had no issues. please can you assist? 

**Expected behavior**
Should be creating sensors with sensor.ss_*Sensor*
for example: sensor.ss_battery_current
### Your environment

- Home Assistant version: 2024.3.3
- Addon:
  - Name: sunsynk-multi
  - Version: 2023-08-17  -  0.6.5
- Inverter:
  - Make: Sunsyn
  - Model: 16kW
  - Firmware: \_\_
- Adaptor details:
  - RS485
/dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0

### You configuration

```- SERIAL_NR: "2211262915"
  HA_PREFIX: SS
  MODBUS_ID: 1
  PORT: tcp://192.168.250.151:502'''

paste config in the block surrounded by 3x backticks

```- SERIAL_NR: "2211262915"
  HA_PREFIX: SS
  MODBUS_ID: 1
  PORT: tcp://192.168.250.151:502'''

### Logs

```See log below: 
[10:24:30] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
10:24:36 INFO    Loading configuration: /data/options.json
10:24:36 INFO    sunsynk library version: 0.6.4
10:24:36 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  5   |   60   |     80    |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
10:24:36 INFO    Using Single phase sensor definitions.
10:24:36 INFO    Importing /share/hass-addon-sunsynk/mysensors.py...
10:24:36 INFO      custom sensors: pv3_power, pv3_voltage, pv3_current, load_frequency, inverter_output_current, priority_mode
10:24:36 ERROR   Unknown sensor specified: day_load_energy_grid_frequency
10:24:36 ERROR   Unknown sensor specified: totalsolar
10:24:36 ERROR   Unknown sensor specified: grid_connection_status
10:24:36 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
10:24:36 INFO    Connecting to tcp://192.168.250.151:502
10:24:36 INFO    PyModbus 3.5.2 tcp: 192.168.250.151:502
10:24:36 INFO    Reading startup sensors Battery Low Capacity, Prog4 Time, Battery Shutdown Capacity, Prog1 Time, Prog3 Time, Prog2 Time, Prog6 Time, Rated power, Serial, Prog5 Time
10:24:36 INFO    ############################################################
10:24:36 INFO                 Inverter serial number '****2915'
10:24:36 INFO    ############################################################
10:24:36 INFO    MQTT: Connecting to sunsynk@192.168.250.151:1883
10:24:36 INFO    MQTT: Connection successful
10:24:37 INFO    Read every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  5  |  prog2_charge, pv3_power, battery_max_charge_current, aux_power, grid_ct_power,  |
|     |    battery_shutdown_capacity, prog3_capacity, prog1_charge, essential_power,     |
|     |        prog6_time, prog3_time, prog1_capacity, grid_ld_power, load_limit,        |
|     |    battery_low_capacity, prog1_time, prog2_time, prog6_charge, battery_power,    |
|     |   prog1_mode, pv1_power, prog4_time, prog6_capacity, prog4_charge, pv2_power,    |
|     | non_essential_power, prog2_capacity, prog4_capacity, prog5_capacity, grid_power, |
|     |      priority_load, prog5_charge, prog3_charge, inverter_power, prog5_time       |
|  15 |     overall_state, dc_transformer_temperature, battery_voltage, pv2_current,     |
|     |       battery_soc, pv3_current, pv2_voltage, load_frequency, pv1_current,        |
|     |      inverter_output_current, grid_current, pv3_voltage, inverter_voltage,       |
|     |                         pv1_voltage, battery_temperature                         |
| 300 |    total_active_energy, day_grid_import, total_pv_energy, total_grid_export,     |
|     |    day_grid_export, day_battery_discharge, total_grid_import, day_pv_energy,     |
|     |         total_battery_charge, total_load_energy, total_battery_discharge         |
+-----+----------------------------------------------------------------------------------+
10:24:37 INFO    Report every (inverter 1)
+-----+---------------------------------------------------------------------------------+
|  s  |                                     Sensors                                     |
+-----+---------------------------------------------------------------------------------+
|  60 |    pv3_power, grid_ld_power, grid_power, aux_power, pv1_power, grid_ct_power,   |
|     |  inverter_power, essential_power, pv2_power, battery_power, non_essential_power |
| 300 |      dc_transformer_temperature, prog2_charge, battery_max_charge_current,      |
|     |      battery_voltage, pv3_current, battery_soc, battery_shutdown_capacity,      |
|     |     grid_current, day_battery_discharge, prog3_capacity, total_grid_import,     |
|     |   prog1_charge, prog6_time, prog3_time, prog1_capacity, total_battery_charge,   |
|     |   pv1_voltage, total_pv_energy, load_limit, pv2_voltage, battery_low_capacity,  |
|     |      pv1_current, prog1_time, prog2_time, prog6_charge, total_load_energy,      |
|     |     prog1_mode, total_battery_discharge, overall_state, total_active_energy,    |
|     |    pv2_current, load_frequency, prog4_time, day_grid_export, prog6_capacity,    |
|     |  prog4_charge, pv3_voltage, day_pv_energy, prog2_capacity, battery_temperature, |
|     |    prog4_capacity, inverter_output_current, prog5_capacity, day_grid_import,    |
|     | priority_load, prog5_charge, total_grid_export, prog3_charge, inverter_voltage, |
|     |                                    prog5_time                                   |
+-----+---------------------------------------------------------------------------------+
'''

paste logs in 3x backticks...

```

kellerza, 2024-08-13:
Everything seems ok from the log.
Are you sure that you connect to the same MQTT server & have MQTT discovery enabled? (prefix=homeassistant) - reconfigure the MQTT integration

starrateit, 2024-08-13:
I know it is so confusing
yes I am sure it is the right server. 
It shows the connections on the MQTT log, I am not using it for anything else but your add on so it is quite strange. 

What do you mean about the below? 
(prefix=homeassistant) - reconfigure the MQTT integration

starrateit, 2024-08-13:
I have sorted it out. Thanks was the discovery was not enabled.
Thanks for the help

closed 2024-08-14 by kellerza