kellerza/sunsynk #331: Sunsynk Load Limit, Time of Use settings not taking.

HarrisHome, 2024-08-29:
Hi KellerZA,

Been using your Sunsynk/Deye Inverter Add-on (multi) latest version 0.6.5 for a while now.
I have the Sunsynk 8Kw inverter connected with the Wave USB to RS485.
All sensors and everything pulls through with no errors, HOWEVER...
I recently installed a PTC element in my geyser (which is on the non-essentials side) and configured at the inverter to Zero-Export.
This updated in my Home Assistant inverter interface from "Essentials' to "Zero Export" - perfect.
Power values all show correctly for essentials, non-essential and CT - No Problem.
Problem is when I click on the Load Limit Selector in Home Assistant and change it from Essentials to Zero Export, I can see it change for a brief second on the inverters screen and then reverts back. Same thing happens when you change a SOC value in the Time of Use settings in Home Assistant. It pushes the write request to the inverter, inverter sees this but keeps its current value.
If you make any changes on the inverter itself it changes on Home Assistant but not vice versa.
Almost like the inverter says I see your valid request BUT denied.
The Log shows this:
```16:45:20 INFO    Queue update prog1_capacity=59
   16:45:20 INFO    Writing sensor prog1_capacity=59 Registers:{268=0x003b}
   16:54:35 INFO    Queue update load_limit=Zero Export
   16:54:35 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002}```

If I can get it to accept the Load Limit changes from Home Assistant, then I can write an Automation to switch this Automatically for when there is excess Solar during the day and not waste energy from the batteries at night to heat the water.

My YAML Config:```
DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "1234567890"
    HA_PREFIX: sunsynk
    MODBUS_ID: 1
    PORT: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A10LK6XO-if00-port0
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - battery_temperature
  - inverter_voltage
  - inverter_frequency
  - grid_current
  - pv2_power
  - pv2_voltage
  - pv2_current
  - day_active_energy
  - day_reactive_energy
  - month_grid_energy
  - month_load_energy
  - month_pv_energy
  - dc_transformer_temperature
  - radiator_temperature
  - settings
  - power_flow_card
SENSORS_FIRST_INVERTER:
  - settings
  - load_limit
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 5
READ_SENSORS_BATCH_SIZE: 60
SCHEDULES:
  - KEY: any_unit
    READ_EVERY: 5
    REPORT_EVERY: 30
    CHANGE_ANY: true
    CHANGE_BY: 0
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: ****
MQTT_USERNAME: SomeUser
MQTT_PASSWORD: SomePassword
DEBUG: 0
DEBUG_DEVICE: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A10LK6XO-if00-port0```

Hope you can enlighten me.

Many Thanks
Clynton

kellerza, 2024-09-01:
Can you write to any register?

There are some of the physical ports on the inverter that are read-only

You can also try the dev version of the addon

HarrisHome, 2024-09-01:
No, even if you toggle Priority to Load - it just changes back after a few seconds. You can see the command go through to the inverter from Home Assistant in the Add-On logs, but it just ignores it an reverts to it's state as selected in the inverter itself.
I tried changing the state from the Solarman App - it actually changes from there and updates a few seconds later in Home Assistant with the new state. So it's not read only. I can change all variables from the solarman app and after clicking upload it changes all states in HomeAssitant. So must be a write issue from the Add-On.
I bit of a pain using 1 app to switch states and another to monitor etc... BUT will have to do for now until I try the Dev version.
Will update here once I try the Dev version and let you know.

kellerza, 2024-09-02:
Some of the RS485 ports are **read-only** - see the first paragraph here: https://solar-assistant.io/help/deye/configuration

HarrisHome, 2024-09-02:
Update - Installed the Edge/Dev version.
Stopped the Multi Version and configured the Dev Version.
Started it and tried changing the Limit Load and Time of Use Settings - They just reverted back to their old settings.
I got this response in the Log:
17:02:27 INFO    Queue update load_limit=Zero Export
/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_sensor.py:161: RuntimeWarning: coroutine 'ASensor.publish' was never awaited
  self.publish(val)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
17:02:28 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
17:09:38 INFO    Queue update prog1_capacity=50
17:09:39 INFO    Writing sensor prog1_capacity=50 Registers:{268=0x0032} 

So seems my Sunsynk 8kw Single Phase is Read Only on the RS485 port.
Just bugs me I can change Load Limit through the SolarMan App and not on HA via your Add-On...
I wonder if settings via the RS485 port are read only when it detects a Solarman Dongle in the RS232 port at the same time?
Will unplug the Solarman Dongle and try again.

Will post here again my findings.

kellerza, 2024-09-05:
Can you try 0.7.1 / latest edge?

HarrisHome, 2024-09-06:
Hi KellerZA,
I see you've been busy, so I clicked through your updates.
On your latest now 0.7.4 and the latest edge.
Tried both versions (1 at a time by disabling the other).
Still the same issue - Load Limit (or any other switch for that matter) will not write to the inverter.
It selects in Home Assistant and you can see the write request in the Add-On log but reverts a few seconds later...
15:20:27 INFO    Queue update load_limit=Zero Export
15:20:29 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
But works the other way, IE. Make the change on the inverter itself and updates on Home Assistant accordingly.
I checked through the Solar-Assistant link you referred to above - My Sunsynk 8kw Single Phase Inverter can read and write all settings on ALL ports.
Do you know if the SolarMan Wifi Dongle takes priority or "locks" the writeable registers whilst it's plugged in?
If "YES", then that could be the issue...
I can change Load Limit and other writeable settings via the SolarMan App, just not through Home Assistant via your Add-On.

kellerza, 2024-09-06:
It seems like it’s read-only. From the logs it does seem to be writing, but the write has no effect :-(

you could try to unplug the solarman dongle (2 small screws), but I’m not sure it will make a difference.

you can try enabling debug mode, but the log will be very busy

HarrisHome, 2024-09-08:
Another update.
I unplugged the Solarman Dongle, leaving the RS232 port open - No Change as you predicted...
Ticked the Limit to Load (Essentials) on the Actual Inverter through settings and watched it update in Home Assistant.
Then selected Zero Export on Home Assistant and a few seconds later it switched back on it's own to Essentials.
I also include the Debug Logs in this process - just in case you "spot" anything.
2024-09-08 17:49:57,048 INFO    ha_addon_sunsynk_multi.a_sensor Queue update load_limit=Zero Export
2024-09-08 17:49:57,049 DEBUG   mqtt_entity.client MQTT: Publish 1R SUNSYNK/status/19****4061/load_limit, Zero Export
2024-09-08 17:49:57,153 INFO    sunsynk.sunsynk Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
2024-09-08 17:49:57,154 DEBUG   pymodbus.logging Adding transaction 0
2024-09-08 17:49:57,154 DEBUG   pymodbus.logging Resetting frame - Current Frame in buffer - 
2024-09-08 17:49:57,155 DEBUG   pymodbus.logging send: 0x1 0x10 0x0 0xf4 0x0 0x1 0x2 0x0 0x2 0x32 0xe5
2024-09-08 17:49:57,205 DEBUG   pymodbus.logging recv: 0x1 0x10 0x0 0xf4 0x0 0x1 0x40 0x3b old_data:  addr=None
2024-09-08 17:49:57,205 DEBUG   pymodbus.logging Processing: 0x1 0x10 0x0 0xf4 0x0 0x1 0x40 0x3b
2024-09-08 17:49:57,206 DEBUG   pymodbus.logging Getting Frame - 0x10 0x0 0xf4 0x0 0x1
2024-09-08 17:49:57,206 DEBUG   pymodbus.logging Factory Response[WriteMultipleRegistersResponse': 16]
2024-09-08 17:49:57,206 DEBUG   pymodbus.logging Frame advanced, resetting header!!
2024-09-08 17:49:57,207 DEBUG   pymodbus.logging Getting transaction 0
2024-09-08 17:49:57,269 DEBUG   pymodbus.logging Adding transaction 0
2024-09-08 17:49:57,270 DEBUG   pymodbus.logging Resetting frame - Current Frame in buffer - 
2024-09-08 17:49:57,270 DEBUG   pymodbus.logging send: 0x1 0x3 0x0 0xf4 0x0 0x1 0xc5 0xf8
2024-09-08 17:49:57,317 DEBUG   pymodbus.logging recv: 0x1 0x3 0x2 0x0 0x2 0x39 0x85 old_data:  addr=None
2024-09-08 17:49:57,317 DEBUG   pymodbus.logging Processing: 0x1 0x3 0x2 0x0 0x2 0x39 0x85
2024-09-08 17:49:57,317 DEBUG   pymodbus.logging Getting Frame - 0x3 0x2 0x0 0x2
2024-09-08 17:49:57,317 DEBUG   pymodbus.logging Factory Response[ReadHoldingRegistersResponse': 3]
2024-09-08 17:49:57,318 DEBUG   pymodbus.logging Frame advanced, resetting header!!
2024-09-08 17:49:57,318 DEBUG   pymodbus.logging Getting transaction 0
2024-09-08 17:49:57,320 DEBUG   sunsynk.sunsynk Time taken to fetch 1 registers starting at 244 : 0.05s
2024-09-08 17:49:57,320 DEBUG   sunsynk.sunsynk Request registers: [244] glen=1. Response [2] len=1. regs={244: 2}
2024-09-08 17:49:57,320 DEBUG   sunsynk.state register {3: 12601, 4: 12594, 5: 12856, 6: 13360, 7: 13873, 16: 14464, 17: 1, 219: 20, 250: 100, 251: 500, 252: 900, 253: 1300, 254: 1700, 255: 2100, 59: 2, 60: 245, 61: 0, 62: 0, 63: 5429, 64: 0, 65: 2546, 66: 2156, 67: 2155, 68: 5956, 69: 0, 70: 147, 71: 107, 72: 3468, 73: 0, 74: 3145, 75: 0, 76: 2, 77: 0, 78: 557, 79: 4996, 80: 0, 81: 33, 82: 0, 83: 0, 84: 246, 90: 1587, 91: 1449, 92: 0, 93: 0, 94: 2000, 95: 0, 96: 6043, 97: 0, 108: 296, 109: 77, 110: 0, 111: 71, 112: 0, 150: 2335, 151: 0, 152: 2335, 153: 2336, 154: 2327, 160: 247, 161: 0, 162: 37, 163: 0, 164: 621, 165: 0, 166: 0, 167: 0, 168: 0, 169: 0, 170: 26, 171: 0, 172: 26, 173: 1489, 174: 0, 175: 1489, 182: 1150, 183: 5259, 184: 88, 185: 0, 186: 0, 187: 0, 188: 0, 189: 0, 190: 1570, 191: 2987, 192: 4996, 193: 4996, 194: 1, 243: 1, 244: 2, 245: 8000, 246: 0, 247: 0, 248: 255, 249: 0, 256: 8000, 257: 8000, 258: 8000, 259: 8000, 260: 8000, 261: 8000, 268: 30, 269: 30, 270: 30, 271: 30, 272: 70, 273: 50, 274: 5, 275: 5, 276: 5, 277: 5, 278: 5, 279: 5} = Essentials (old=Zero Export)

You see anything out of the ordinary or IS my Sunsynk Single Phase 8Kw Inverter RS485 port Read Only?

HarrisHome, 2025-04-09:
> Did you get it to work?

Unfortunately not.
It reads and pulls all the data in to Home Assistant and populates all the dashboards, HOWEVER the only thing I can't do is write back to the inverter (to make changes to the Time of Use, Priority Load / Battery changes etc).
I got 1 of the very first 8kw sunsynk inverters in SA, which came with a Solarman Dongle.
I had Software and Firmware updates done via the Solarman Dongle back then to bring it up to the new "Flow Chart" UI.
Somehow the inverter only has read rights and can not be written to either by the RS485 port or Serial / WiFi.
I've tried all options.
I emailed Sunsynk and they told me I have to get the New Sunsynk Dongle and plug it in instead of the SolarMan Dongle (They apparently don't have access via the old solarman dongle any more), so they can log in and see if they can update the firmware to a newer version which may enable the write back function through the RS485 port.
I'm just worried they end up bricking my inverter (which I have heard so many horror stories about), so leaving it for now.

jeybee74, 2025-04-09:
Oh, what a bummer.Maybe you could bring it to Sunsynk so they could try to update it or something, but maybe it's risky like you said.I have a Sunsynk 3phase 12kW, the solar assistant guys told me it would not work with rs485, only read they said..But it works like a charm with this integration and USB-RS485 device😊BR9. apr.

kellerza, 2026-08-15:
You can try to trace the sensor that does not want to work and this will show if the port is read-only or not

```yaml
SENSOR_OVERRIDES:
- prog4_power.trace=1
```