kellerza/sunsynk #333: Deye 6k - PyModBus Configuration issue

unwiredtech, 2024-09-01:
I am trying to use pymodbus but I am getting sensor reading issues. I also tried to use Solarman, I was able to connect but after the next readout, I am getting timeout, also I tried to re-adjust the schedules but no avail.

**CONFIG:**
`DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "****6285"
    HA_PREFIX: DY
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "29654xxxxx"
    PORT: tcp://192.168.0.117:502
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: hass
MQTT_PASSWORD: "*****"
`

`s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
14:45:12 INFO    Loading configuration: /data/options.json
14:45:12 INFO    sunsynk library version: 0.6.4
14:45:12 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  5   |   60   |     80    |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
14:45:12 INFO    Using Single phase sensor definitions.
14:45:12 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
14:45:12 INFO    Connecting to tcp://192.168.0.117:502
14:45:12 INFO    PyModbus 3.5.2 tcp: 192.168.0.117:502
14:45:12 INFO    Reading startup sensors Prog2 Time, Battery Low Capacity, Prog5 Time, Rated power, Prog1 Time, Prog3 Time, Prog6 Time, Prog4 Time, Serial
14:45:21 ERROR   OSError reading 5 registers from 3: failed to read register 3 - function code: 131; OSError reading 2 registers from 16: failed to read register 16 - function code: 131; OSError reading 1 registers from 219: failed to read register 219 - function code: 131; OSError reading 6 registers from 250: failed to read register 250 - function code: 131
14:45:30 ERROR   OSError reading 5 registers from 3: failed to read register 3 - function code: 131; OSError reading 2 registers from 16: failed to read register 16 - function code: 131; OSError reading 1 registers from 219: failed to read register 219 - function code: 131; OSError reading 6 registers from 250: failed to read register 250 - function code: 131`

kellerza, 2024-09-01:
What hardware do you use?
Have you tried the Fault Finding section in the docs?

unwiredtech, 2024-09-02:
Hello @kellerza 

Yes, I did try the fault finding. 
For the hardware I am using Waveshare, the one you recommended.
I can live with the current setup driver - Solarman.

Question, I can use the split RS485/CAN? Because I wanted to use my Eastron as my CT and right now I am using the Meter Port for the HAS integration.

kellerza, 2024-09-02:
Some of the RS485 ports are **read-only** - see the first paragraph here: https://solar-assistant.io/help/deye/configuration

You should be able to split it - the Wiring page should have details

unwiredtech, 2024-09-02:
Thanks!

We can close this issue for now. I am settled with Solarman as Driver.

Earlier this afternoon the solarman data got choked, I increased the read
time and gap for home assistant. I am observing the traffic. I really need
this integration as I will be remotely monitoring and configuring this
inverter.

Cheers!

closed 2024-09-06 by kellerza