kellerza/sunsynk #334: solar export

cc120689, 2024-09-04:
Hello again Kellerza, it seems that with the latest update the Solar export sensor has stopped responding, when deactivating it it cannot be reactivated and in the solarman app it appears as "select" or neither activated nor deactivated, if I remove it from the configuration and add this in mysensors.py it does work::
SENSORS += SelectRWSensor(247, "Solar Sell", options={0: "No", 1: "Yes"})

kellerza, 2024-09-05:
Can you try adding these two as well and see what they do?

```python
SENSORS += SwitchRWSensor(247, "Solar Export_1", on=1)
SENSORS += SwitchRWSensor(247, "Solar Export_1", bitmask=1)
```

kellerza, 2024-09-05:
Can you try 0.7.1 / latest edge?

cc120689, 2024-09-05:
20:22:52 INFO                 Inverter serial number '****6414'
20:22:52 INFO    ############################################################
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 79, in <module>
    main()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 74, in main
    loop.run_until_complete(main_loop(loop))
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 40, in main_loop
    await ist.hass_discover_sensors()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 165, in hass_discover_sensors
    ents = [
           ^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 166, in <listcomp>
    s.create_entity(dev, ist=self) for s in self.ss.values() if not s.hidden
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_sensor.py", line 217, in create_entity
    self.entity = RWEntity(**ent, on_change=on_change)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: RWEntity.__init__() missing 1 required positional argument: 'device'
[20:22:52] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
20:25:33 INFO    Loading configuration: /data/options.json
20:25:33 INFO    sunsynk library version: 0.6.4
20:25:33 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|    enum   |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  5   |   5    |           |          |     1      |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
|     a     |  +  |  5   |   5    |           |          |     1      |
|     hz    |  +  |  5   |   5    |           |          |            |
|     v     |  +  |  5   |   5    |           |          |            |
|     %     |  +  |  60  |   60   |           |          |     1      |
+-----------+-----+------+--------+-----------+----------+------------+
20:25:33 INFO    Using Single phase sensor definitions.
20:25:33 INFO    Importing /share/hass-addon-sunsynk/mysensors.py...
20:25:33 INFO      custom sensors: grid_peak_shaving_power, max_export_power, pv_total_power, solar_sell
20:25:33 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
20:25:33 INFO    Connecting to tcp://192.168.1.170:502
20:25:33 INFO    PyModbus 3.7.2 tcp: 192.168.1.170:502
20:25:33 INFO    Reading startup sensors Serial, Rated power, Prog1 Time, Prog6 Time, Prog5 Time, Battery Shutdown Capacity, Prog4 Time, Battery Low Capacity, Prog2 Time, Prog3 Time
20:25:33 INFO    ############################################################
20:25:33 INFO                 Inverter serial number '****6414'
20:25:33 INFO    ############################################################
[... 107 lines cut ...]
    loop.run_until_complete(main_loop(loop))
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 40, in main_loop
    await ist.hass_discover_sensors()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 165, in hass_discover_sensors
    ents = [
           ^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 166, in <listcomp>
    s.create_entity(dev, ist=self) for s in self.ss.values() if not s.hidden
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_sensor.py", line 217, in create_entity
    self.entity = RWEntity(**ent, on_change=on_change)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...] ed positional argument: 'device'
[20:25:33] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped"

kellerza, 2024-09-05:
Please try 0.7.3 🤦‍♂️

cc120689, 2024-09-05:
With 0.7.3
20:30:07 INFO                 Inverter serial number '****6414'
20:30:07 INFO    ############################################################
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 79, in <module>
    main()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 74, in main
    loop.run_until_complete(main_loop(loop))
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 40, in main_loop
    await ist.hass_discover_sensors()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 165, in hass_discover_sensors
    ents = [
           ^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 166, in <listcomp>
    s.create_entity(dev, ist=self) for s in self.ss.values() if not s.hidden
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_sensor.py", line 217, in create_entity
    self.entity = RWEntity(**ent, on_change=on_change)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: RWEntity.__init__() missing 1 required positional argument: 'device'
[20:30:07] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
20:33:49 INFO    Loading configuration: /data/options.json
20:33:49 INFO    sunsynk library version: 0.6.4
20:33:49 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|    enum   |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  5   |   5    |           |          |     1      |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
|     a     |  +  |  5   |   5    |           |          |     1      |
|     hz    |  +  |  5   |   5    |           |          |            |
|     v     |  +  |  5   |   5    |           |          |            |
|     %     |  +  |  60  |   60   |           |          |     1      |
+-----------+-----+------+--------+-----------+----------+------------+
20:33:49 INFO    Using Single phase sensor definitions.
20:33:49 INFO    Importing /share/hass-addon-sunsynk/mysensors.py...
20:33:49 INFO      custom sensors: grid_peak_shaving_power, max_export_power, pv_total_power, solar_sell
20:33:49 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
20:33:49 INFO    Connecting to tcp://192.168.1.170:502
20:33:49 INFO    PyModbus 3.7.2 tcp: 192.168.1.170:502
20:33:49 INFO    Reading startup sensors Prog6 Time, Battery Shutdown Capacity, Prog3 Time, Battery Low Capacity, Prog2 Time, Prog5 Time, Serial, Prog1 Time, Rated power, Prog4 Time
20:33:49 INFO    ############################################################
20:33:49 INFO                 Inverter serial number '****6414'
[... 7 lines cut ...]
    loop.run_until_complete(main_loop(loop))
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/__main__.py", line 40, in main_loop
    await ist.hass_discover_sensors()
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 165, in hass_discover_sensors
    ents = [
           ^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 166, in <listcomp>
    s.create_entity(dev, ist=self) for s in self.ss.values() if not s.hidden
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_sensor.py", line 212, in create_entity
    self.entity = RWEntity(
                  ^^^^^^^^^
TypeError: mqtt_entity.entities.RWEntity() got multiple values for keyword argument 'entity_category'
[20:33:49] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

kellerza, 2024-09-06:
0.7.4 will hopefully do the magic

cc120689, 2024-09-07:
It works perfectly again! Thanks!

closed 2024-09-07 by cc120689

f1wade, 2025-04-04:
I'm on 0.8.2, and i get this issue.
Do I need to downgrade to 0.7.4?

f1wade, 2025-04-04:
Never mind, I had the same sensor written in my sensors which was overwriting the default, which works.