kellerza/sunsynk #335: HA add-on stops updating, MQTT shows SS RS485 time-out

apuntigam, 2024-09-04:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

Since yesterday evening around 19h30 (based on my HA graphs) the Sunsynk add-on periodically stops updating. The dashboard looks normal, until you realise nothing has change on it for some time. The add-on log shows that data is no longer being read from the inverter's registers.

In the MQTT integration, the sensors show outdated values (from when the updating stopped), but in the MQTT Logbook it shows "SS RS485 timeout changed to XX", where XX is a number that seems to increment by 11 every few 2 minutes. For example:
"SS RS485 timeout changed to 24"
"SS RS485 timeout changed to 35"
"SS RS485 timeout changed to 46"

I looked at RS485 module admin screen and it is still sending data.
 
If I restart the Sunsynk add-on, it works normally for a while but then reverts to the above again.

I don't know if it is the Sunsynk add-on, the MQTT integration or the Mosquito broker causing the issue - don't have the technical know-how - but if you could give me some pointers on finding out what it is that's causing the issue, I'm happy to do the digging to figure it out!

**Expected behavior**
The add-on should just keep updating, as it has been doing flawlessly for months.

### Your environment

- Home Assistant version: 2024.8.3
- Addon:
  - Name: sunsynk-multi
  - Version: 0.7.0
- Inverter:
  - Make: Sunsynk
  - Model: 12kW 3-phase
  - Firmware: M 2005-1144 / S  18-07 / C 1001-E043
- Adaptor details:
  - RS485/RS232 (USR-TCP232-410S), same adapter that's been working successfully for months, never made any changes to the firmware, etc.
  
### Your configuration
I have no idea where to find this... 

```yaml
paste config in the block surrounded by 3x backticks
```

### Logs

This is what the Sunsynk log looks like once it stops updating...

```
2024-09-04 17:02:27,185 DEBUG   pymodbus.logging Processing: 0x13 0x5 0x0 0x0 0x0 0xf 0x1 0x3 0xc 0x0 0x1 0x0 0x2 0x0 0x0 0x0 0xb 0x0 0x0 0x0 0x1
2024-09-04 17:02:27,185 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-04 17:02:29,122 DEBUG   pymodbus.logging Adding transaction 4870
2024-09-04 17:02:29,122 DEBUG   pymodbus.logging send: 0x13 0x6 0x0 0x0 0x0 0x6 0x1 0x3 0x1 0xf4 0x0 0x1
2024-09-04 17:02:29,171 DEBUG   pymodbus.logging recv: 0x13 0x6 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2 old_data:  addr=None
2024-09-04 17:02:29,171 DEBUG   pymodbus.logging Processing: 0x13 0x6 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2024-09-04 17:02:29,171 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-04 17:02:32,123 DEBUG   pymodbus.logging Adding transaction 4870
2024-09-04 17:02:32,123 DEBUG   pymodbus.logging send: 0x13 0x6 0x0 0x0 0x0 0x6 0x1 0x3 0x1 0xf4 0x0 0x1
2024-09-04 17:02:35,126 DEBUG   pymodbus.logging Adding transaction 4870
2024-09-04 17:02:35,126 DEBUG   pymodbus.logging send: 0x13 0x6 0x0 0x0 0x0 0x6 0x1 0x3 0x1 0xf4 0x0 0x1
2024-09-04 17:02:35,179 DEBUG   pymodbus.logging recv: 0x13 0x6 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2 old_data:  addr=None
2024-09-04 17:02:35,179 DEBUG   pymodbus.logging Processing: 0x13 0x6 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2024-09-04 17:02:35,179 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-04 17:02:38,127 DEBUG   pymodbus.logging Adding transaction 4870
2024-09-04 17:02:38,127 DEBUG   pymodbus.logging send: 0x13 0x6 0x0 0x0 0x0 0x6 0x1 0x3 0x1 0xf4 0x0 0x1
2024-09-04 17:02:38,183 DEBUG   pymodbus.logging recv: 0x13 0x6 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2 old_data:  addr=None
2024-09-04 17:02:38,183 DEBUG   pymodbus.logging Processing: 0x13 0x6 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2024-09-04 17:02:38,183 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-04 17:02:40,124 DEBUG   pymodbus.logging Adding transaction 4871
2024-09-04 17:02:40,124 DEBUG   pymodbus.logging send: 0x13 0x7 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0x1c 0x0 0x1
2024-09-04 17:02:40,179 DEBUG   pymodbus.logging recv: 0x13 0x7 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x4 0xe2 old_data:  addr=None
2024-09-04 17:02:40,179 DEBUG   pymodbus.logging Processing: 0x13 0x7 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x4 0xe2
[... 47 lines cut ...]
2024-09-04 17:03:08,190 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-04 17:03:11,133 DEBUG   pymodbus.logging Adding transaction 4873
2024-09-04 17:03:11,133 DEBUG   pymodbus.logging send: 0x13 0x9 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0x2b 0x0 0x4
2024-09-04 17:03:11,194 DEBUG   pymodbus.logging recv: 0x13 0x9 0x0 0x0 0x0 0xb 0x1 0x3 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
2024-09-04 17:03:11,194 DEBUG   pymodbus.logging Processing: 0x13 0x9 0x0 0x0 0x0 0xb 0x1 0x3 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2024-09-04 17:03:11,194 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-04 17:03:13,131 DEBUG   pymodbus.logging Adding transaction 4874
2024-09-04 17:03:13,131 DEBUG   pymodbus.logging send: 0x13 0xa 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0x4a 0x0 0x5
2024-09-04 17:03:13,193 DEBUG   pymodbus.logging recv: 0x13 0xa 0x0 0x0 0x0 0xd 0x1 0x3 0xa 0x4 0xe5 0x14 0x9b 0x0 0x57 0x0 0x0 0x4 0xd9 old_data:  addr=None
2024-09-04 17:03:13,193 DEBUG   pymodbus.logging Processing: 0x13 0xa 0x0 0x0 0x0 0xd 0x1 0x3 0xa 0x4 0xe5 0x14 0x9b 0x0 0x57 0x0 0x0 0x4 0xd9
2024-09-04 17:03:13,193 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-04 17:03:16,132 DEBUG   pymodbus.logging Adding transaction 4874
2024-09-04 17:03:16,132 DEBUG   pymodbus.logging send: 0x13 0xa 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0x4a 0x0 0x5
2024-09-04 17:03:19,134 DEBUG   pymodbus.logging Adding transaction 4874
2024-09-04 17:03:19,134 DEBUG   pymodbus.logging send: 0x13 0xa 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0x4a 0x0 0x5
2024-09-04 17:03:19,191 DEBUG   pymodbus.logging recv: 0x13 0xa 0x0 0x0 0x0 0xd 0x1 0x3 0xa 0x4 0xe5 0x14 0x9b 0x0 0x57 0x0 0x0 0x4 0xd3 old_data:  addr=None
2024-09-04 17:03:19,191 DEBUG   pymodbus.logging Processing: 0x13 0xa 0x0 0x0 0x0 0xd 0x1 0x3 0xa 0x4 0xe5 0x14 0x9b 0x0 0x57 0x0 0x0 0x4 0xd3
2024-09-04 17:03:19,191 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-04 17:03:22,135 DEBUG   pymodbus.logging Adding transaction 4874
2024-09-04 17:03:22,135 DEBUG   pymodbus.logging send: 0x13 0xa 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0x4a 0x0 0x5
2024-09-04 17:03:22,195 DEBUG   pymodbus.logging recv: 0x13 0xa 0x0 0x0 0x0 0xd 0x1 0x3 0xa 0x4 0xe5 0x14 0x9b 0x0 0x57 0x0 0x0 0x4 0xcf old_data:  addr=None
2024-09-04 17:03:22,196 DEBUG   pymodbus.logging Processing: 0x13 0xa 0x0 0x0 0x0 0xd 0x1 0x3 0xa 0x4 0xe5 0x14 0x9b 0x0 0x57 0x0 0x0 0x4 0xcf
2024-09-04 17:03:22,196 DEBUG   pymodbus.logging Frame check failed, ignoring!!

```

apuntigam, 2024-09-05:
Just some more log info:

"""
2024-09-05 05:29:54,120 DEBUG   pymodbus.logging Processing: 0x39 0x55 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x15 0x90 0x1 0x2c
2024-09-05 05:29:54,120 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-05 05:29:56,061 DEBUG   pymodbus.logging Adding transaction 14678
2024-09-05 05:29:56,061 DEBUG   pymodbus.logging send: 0x39 0x56 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x6c 0x0 0x2
2024-09-05 05:29:56,114 DEBUG   pymodbus.logging recv: 0x39 0x56 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x96 0x0 0xb4 old_data:  addr=None
2024-09-05 05:29:56,114 DEBUG   pymodbus.logging Processing: 0x39 0x56 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x96 0x0 0xb4
2024-09-05 05:29:56,114 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-05 05:29:59,062 DEBUG   pymodbus.logging Adding transaction 14678
2024-09-05 05:29:59,062 DEBUG   pymodbus.logging send: 0x39 0x56 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x6c 0x0 0x2
2024-09-05 05:29:59,116 DEBUG   pymodbus.logging recv: 0x39 0x56 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x96 0x0 0xb4 old_data:  addr=None
2024-09-05 05:29:59,116 DEBUG   pymodbus.logging Processing: 0x39 0x56 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x96 0x0 0xb4
2024-09-05 05:29:59,116 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-05 05:30:00,052 ERROR   ha_addon_sunsynk_multi.errors (1 in 5 min) OSError in callback read_ss: timeout reading 2 registers from 101; timeout reading 2 registers from 108; timeout reading 6 registers from 115; timeout reading 3 registers from 128; timeout reading 6 registers from 141; timeout reading 3 registers from 500; timeout reading 2 registers from 506; timeout reading 8 registers from 514; timeout reading 8 registers from 522; timeout reading 2 registers from 534; timeout reading 1 registers from 540; timeout reading 1 registers from 552; timeout reading 4 registers from 555; timeout reading 5 registers from 586; timeout reading 4 registers from 609; timeout reading 1 registers from 619; timeout reading 6 registers from 622; timeout reading 6 registers from 633; timeout reading 4 registers from 650; timeout reading 8 registers from 672 poll_need_to_read

2024-09-05 05:30:00,052 ERROR   ha_addon_sunsynk_multi.errors (1 in 5 min) OSError in callback read_ss: timeout reading 2 registers from 101; timeout reading 2 registers from 108; timeout reading 6 registers from 115; timeout reading 3 registers from 128; timeout reading 6 registers from 141; timeout reading 1 registers from 500; timeout reading 1 registers from 540; timeout reading 1 registers from 552; timeout reading 4 registers from 555; timeout reading 5 registers from 586; timeout reading 4 registers from 609; timeout reading 1 registers from 619; timeout reading 6 registers from 622; timeout reading 6 registers from 633; timeout reading 4 registers from 650; timeout reading 8 registers from 672 poll_need_to_read

2024-09-05 05:30:00,052 DEBUG   mqtt_entity.client MQTT: Publish 0 SUNSYNK/status/2209237054/timeout, 3847
2024-09-05 05:30:00,053 DEBUG   mqtt_entity.client MQTT: Publish 0 SUNSYNK/status/2209237054/cb_stats, 0
2024-09-05 05:30:00,054 DEBUG   mqtt_entity.client MQTT: Publish 0 SUNSYNK/status/2209237054/cb_stats_attr, {"period": "120s", "long_calls": "0/0", "long_calls_avg": "0s", "slips": "119/120", "slips_avg": "147.28s", "busy": 119}
2024-09-05 05:30:02,064 DEBUG   pymodbus.logging Adding transaction 14678
2024-09-05 05:30:02,064 DEBUG   pymodbus.logging send: 0x39 0x56 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x6c 0x0 0x2
2024-09-05 05:30:02,118 DEBUG   pymodbus.logging recv: 0x39 0x56 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x96 0x0 0xb4 old_data:  addr=None
2024-09-05 05:30:02,118 DEBUG   pymodbus.logging Processing: 0x39 0x56 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x96 0x0 0xb4
2024-09-05 05:30:02,118 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-05 05:30:05,065 DEBUG   pymodbus.logging Adding transaction 14678
2024-09-05 05:30:05,065 DEBUG   pymodbus.logging send: 0x39 0x56 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x6c 0x0 0x2
2024-09-05 05:30:05,120 DEBUG   pymodbus.logging recv: 0x39 0x56 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x96 0x0 0xb4 old_data:  addr=None
2024-09-05 05:30:05,120 DEBUG   pymodbus.logging Processing: 0x39 0x56 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x96 0x0 0xb4
2024-09-05 05:30:05,120 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-05 05:30:07,064 DEBUG   pymodbus.logging Adding transaction 14679
2024-09-05 05:30:07,064 DEBUG   pymodbus.logging send: 0x39 0x57 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x73 0x0 0x6
2024-09-05 05:30:07,122 DEBUG   pymodbus.logging recv: 0x39 0x57 0x0 0x0 0x0 0xf 0x1 0x3 0xc 0x0 0xa 0x0 0x32 0x0 0x14 0x11 0xf8 0x14 0x50 0x12 0x8e old_data:  addr=None
2024-09-05 05:30:07,122 DEBUG   pymodbus.logging Processing: 0x39 0x57 0x0 0x0 0x0 0xf 0x1 0x3 0xc 0x0 0xa 0x0 0x32 0x0 0x14 0x11 0xf8 0x14 0x50 0x12 0x8e
2024-09-05 05:30:07,122 DEBUG   pymodbus.logging Frame check failed, ignoring!!
2024-09-05 05:30:10,066 DEBUG   pymodbus.logging Adding transaction 14679
2024-09-05 05:30:10,066 DEBUG   pymodbus.logging send: 0x39 0x57 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x73 0x0 0x6
2024-09-05 05:30:10,124 DEBUG   pymodbus.logging recv: 0x39 0x57 0x0 0x0 0x0 0xf 0x1 0x3 0xc 0x0 0xa 0x0 0x32 0x0 0x14 0x11 0xf8 0x14 0x50 0x12 0x8e old_data:  addr=None
2024-09-05 05:30:10,124 DEBUG   pymodbus.logging Processing: 0x39 0x57 0x0 0x0 0x0 0xf 0x1 0x3 0xc 0x0 0xa 0x0 0x32 0x0 0x14 0x11 0xf8 0x14 0x50 0x12 0x8e
2024-09-05 05:30:10,124 DEBUG   pymodbus.logging Frame check failed, ignoring!!
[... cut ...]
2024-09-05 05:30:38,074 DEBUG   pymodbus.logging Adding transaction 14681
2024-09-05 05:30:38,074 DEBUG   pymodbus.logging send: 0x39 0x59 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x8d 0x0 0x6
2024-09-05 05:30:38,137 DEBUG   pymodbus.logging recv: 0x39 0x59 0x0 0x0 0x0 0xf 0x1 0x3 0xc 0x0 0x1 0x0 0x2 0x0 0x0 0x0 0xb 0x0 0x0 0x0 0x1 old_data:  addr=None
2024-09-05 05:30:38,137 DEBUG   pymodbus.logging Processing: 0x39 0x59 0x0 0x0 0x0 0xf 0x1 0x3 0xc 0x0 0x1 0x0 0x2 0x0 0x0 0x0 0xb 0x0 0x0 0x0 0x1
2024-09-05 05:30:38,137 DEBUG   pymodbus.logging Frame check failed, ignoring!!
"""

apuntigam, 2024-09-06:
I updated to 0.7.3 last night and since then things seem to be working properly again.

kellerza, 2024-09-06:
Great! Glad to hear 0.7.3 did the trick

0.7.4 has another fix (for sensors you don't seem to be using)

closed 2024-09-06 by kellerza