kellerza/sunsynk #343: Grid Power showing incorrectly after Firmware upgrade.

Marzlio, 2024-09-10:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

After the firmware upgrade from version: 9.0.2.2 to 9.0.2.6 grid power is showing incorrectly.
I believe it is expecting WATTS and getting KW. But i could be wrong
when connecting to mbusd and checking the inverter values for 169, 168 and 168 i get the following 

```bash
Register 167 = 94
Register 168 = 0
Register 169 = 94
```
When checking on the inverter is was 940W 
Apologies in advance if i missed something obvious.

**Expected behaviour**

Not 100% sure but i think it needs to be multiplied by 10?

when i use the below it shows correctly

```yaml

  - platform: template
    sensors:
      sunsynk_grid_power_scaled:
        friendly_name: "Grid Power (Scaled)"
        unit_of_measurement: 'W'
        value_template: "{{ states('sensor.sunsynk_grid_power') | float * 10 }}"
```

### Your environment

- Home Assistant version: 2023.8.x
- Addon:
  - Name: sunsynk-multi
  - Version: 0.7.4
- Inverter:
  - Make: Sunsynk
  - Model: 16kW 1P
  - Firmware: 9.0.2.6
- Adaptor details:
  - RS485/RS232

kellerza, 2024-09-11:
If the definition of the sensor was changed by the firmware upgrade you can redefine the sensor (as a custom sensor)

Currently 169 is defined as:
```python
Sensor(169, "Grid power", WATT, -1)
```

Adding the following custom sensor should override the standard single phase sensor
```python
Sensor(169, "Grid power", WATT, -10)
```

If there are other sensors that also changed, we might need a new definition.
Existing ones are defined here: https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/

Marzlio, 2024-09-11:
This changed worked, thanks!
I will work through the definitions to check what else they changed.

Marzlio, 2024-09-11:
Everything else looks to be working.
Thank you!

Happy if we close this one.

closed 2024-09-12 by Marzlio