kellerza/sunsynk #346: HA Addon: no communication with inverters since firmware update on the inverter

bsafh, 2024-09-18:
### Issue related to

Sunsynk / Home Assistant Add-On
dev/edge and stable  version: d48d3f3 
stable version: 0.7.4 

Deye high voltage three phase hybrid inverter 10k
new firmware 
    Protokoll-Version：0104 
    MAIN：3002-1077-1E08 
    HMI：2001-C032 

RS485-ETH adapter is reachable via IP from Addon, 

MQTT broker is accepting messages. 

### Describe the issue/bug

Since update of the inverter firmware to the above mentioned version, no MQTT messages are seen any more in the MQTT broker (and therefore not in Homeassistant as well). Nothing. Total silence, not a single message. No error messages in Addon log.

looks like the inverters do not talk to the RS485-ETH box any more - is there any setting in the Deye inverter setting to disable or enable the RS485 communication?

The Solarman App and the Solarman HA integration do continue to work without issues. But nothing over RS485 any more.

**Expected behavior**

messages ar passewd to MQTT broker

### Your environment

- Home Assistant version: 2023.9.2
- Addon:
  - Name: sunsynk-multi / dev-edge
  - Version: 0.7.4   / d48d3f3 
- Inverter:
  - Make: Deye high voltage three phase hybrid inverter 10k
  - Model: Deye high voltage three phase hybrid inverter 10k EU version
  - Firmware: MAIN：3002-1077-1E08 
    HMI：2001-C032 
- Adaptor details:
  - RS485
  - RS485-ETH

### You configuration

```

DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "23121221XX"
    HA_PREFIX: WR_Deye_1
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: serial-tcp://192.168.80.44:8234
  - SERIAL_NR: "23121262YY"
    HA_PREFIX: WR_Deye_2
    MODBUS_ID: 2
    DONGLE_SERIAL_NUMBER: "0"
    PORT: serial-tcp://192.168.80.45:8234
SENSOR_DEFINITIONS: three-phase-hv
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
  - settings
  - all
  - pv1_power
  - grid_ct_power
  - Inverter power
  - Inverter L1 power
[... 323 lines cut ...]
15:17:20 ERROR   Unknown sensor specified: grid_voltage
15:17:20 ERROR   Unknown sensor specified: load_frequency
15:17:20 ERROR   Unknown sensor specified: aux_power
15:17:20 ERROR   Unknown sensor specified: essential_power
15:17:20 ERROR   Unknown sensor specified: essential_1_power
15:17:20 ERROR   Unknown sensor specified: essential_2_power
15:17:20 ERROR   Unknown sensor specified: non_essential_power
15:17:20 ERROR   Unknown sensor specified: day_reactive_energy
15:17:20 ERROR   Unknown sensor specified: month_grid_energy
15:17:20 ERROR   Unknown sensor specified: month_load_energy
15:17:20 ERROR   Unknown sensor specified: month_pv_energy
15:17:20 ERROR   Unknown sensor specified: year_grid_export
15:17:20 ERROR   Unknown sensor specified: year_load_energy
15:17:20 ERROR   Unknown sensor specified: year_pv_energy
15:17:20 ERROR   Unknown sensor specified: environment_temperature
15:17:20 ERROR   Unknown sensor specified: control_mode
15:17:20 ERROR   Unknown sensor specified: generator_input
15:17:20 ERROR   Unknown sensor specified: battery_charging_voltage
15:17:20 ERROR   Unknown sensor specified: bat1_cycle
15:17:20 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
15:17:20 INFO    Connecting to serial-tcp://192.168.80.44:8234
15:17:20 INFO    PyModbus 3.7.2 serial-tcp: 192.168.80.44:8234
15:17:20 INFO    Reading startup sensors Prog3 Time, Rated power, Prog5 Time, Battery Shutdown Capacity, Prog1 Time, Battery Low voltage, Battery Restart Capacity, Battery Float voltage, Battery Low Capacity, Battery Shutdown voltage, Serial, Prog4 Time, Prog2 Time, Prog6 Time

```

kellerza, 2024-09-18:
Most likely the modbus id went back to 0. Otherwise, treat it like a new install & follow the fault-finding section in the docs

or downgrade :wink:

bsafh, 2024-09-18:
well, the modbus IDs of the 2 inverters have been 0 and 1, so falling back to 0 should at least result in one inverter sending data.

I do have one RS485-ETH bridge per inverter, so there should be no interference, even if both are on 0

I do not have a copy of the previous firmware ...

kellerza, 2024-09-18:
Don't use 0

bsafh, 2024-09-18:
One inverter is on 0, the other on 1, just as before the update. Neither of them sends any data (they both have there own RS485-ETH adapter).

bsafh, 2024-09-19:
I just received an answer from the Deye support:

"Can i update HMI to solve problem. Because the firmware is special for using RS485."

Whatever that means?

bsafh, 2024-09-20:
well, the Deye support installed another update of the HMI and now RS485 works again (without any changes to the wires or the adapter).

This combination works: 
MAIN：3002-1077-1E08
HMI：2001-C0F4

HMI 2001-C032 does NOT support RS485  on the SUN-10K-SG01HP3-EU-AM2 !!!

closed 2024-09-20 by bsafh