kellerza/sunsynk #352: 0.7.5 Update error

OwlBawl, 2024-10-13:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

get error notification in HA when trying update add-on from 0.7.4 to 0.7.5 via GUI

Can't install ghcr.io/kellerza/hass-addon-sunsynk-multi/amd64:0.7.5: 500 Server Error for http+docker://localhost/v1.45/images/create?tag=0.7.5&fromImage=ghcr.io%2Fkellerza%2Fhass-addon-sunsynk-multi%2Famd64&platform=linux%2Famd64: Internal Server Error ("manifest unknown")

ErnestoBrandao, 2024-10-13:
Same problem for me. and no 0.7.5 version for download

czgif, 2024-10-14:
Hello,

I have same problem : get error notification in HA when trying update add-on from 0.7.4 to 0.7.5 via GUI

drevil54, 2024-10-14:
Hello,
I have same error:
Can't install ghcr.io/kellerza/hass-addon-sunsynk-multi/amd64:0.7.5: 500 Server Error for http+docker://localhost/v1.45/images/create?tag=0.7.5&fromImage=ghcr.io%2Fkellerza%2Fhass-addon-sunsynk-multi%2Famd64&platform=linux%2Famd64: Internal Server Error ("manifest unknown")

kellerza, 2024-10-14:
Please try again

deondti, 2024-10-14:
I believe kellerza has fixed the issue, mine is working now

drevil54, 2024-10-14:
yes, now update is working, thank you!

closed 2024-10-14 by kellerza