kellerza/sunsynk #353: Wrong offset of Battery temperature sensor in single_phase.py for Deye

OwlBawl, 2024-10-13:
### Your environment
- Home Assistant version: last
- Addon:
  - Name: multi
  - Version: 0.7.4
  - Inverter:
  - Make: Deye
  - Model: SUN 6K SG03 LP1 EU
  - Firmware: 9.0.2.6
- Adaptor details:
  - RS485 
  
### Describe the issue/bug and what you expect

Wrong offset of sensor I think

##########
# Battery
##########
SENSORS += (
    TempSensor(182, "**Battery temperature**", CELSIUS, 0.1),

in /single_phase.py
(https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions)

Add-on shows 75*C 
while Solarman shows 25*C (correct temp) using the same register [0x00B6] with offset: 1000

**From Deye ModBus description file:**
182 - battery temperature - R [0,3000] - 0.1 - ℃
**Real value of offset + 1000**
(example 1200 is 20.0 ℃)

OwlBawl, 2024-10-14:
```
from sunsynk import AMPS, CELSIUS, KWH, VOLT, WATT
from sunsynk.rwsensors import NumberRWSensor, SelectRWSensor, TimeRWSensor
from sunsynk.sensors import (
    MathSensor,
    Sensor,
    SensorDefinitions,
    TempSensor,
)

SENSORS = SensorDefinitions()

from sunsynk.sensors import Sensor

class BatteryTempSensor(Sensor):
    def __init__(self, register, name, unit):
        super().__init__(register, name, unit, 0.1)

    def decode(self, value):
        # Adjust the value by subtracting 1000 and apply the scaling
        return (value - 1000) * 0.1

SENSORS += BatteryTempSensor(182, "Battery Temp", CELSIUS)
```

kellerza, 2024-10-14:
Do all the temperature sensors work like this?
There is no generic offset handler in the sensors today, so will have to use a class like this

OwlBawl, 2024-10-14:
No, DC and Radiator temperatures readings are correct, only battery has this offset...

kellerza, 2024-10-16:
Can you try the latest dev? https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/single_phase.py#L30

OwlBawl, 2024-10-16:
If I understood correctly - installed Sunsynk/Deye Inverter Add-on (edge/dev) for test
That's what i've got:
[image]
real temp was 23*C

OwlBawl, 2024-10-16:
I think this offset doubled 
TempSensor(182, "Battery temperature", CELSIUS, 0.1, offset=1000),

with
class TempSensor(Sensor):
    """Offset by 100 for temperature."""

    offset: int = 100

    def reg_to_value(self, regs: RegType) -> ValType:
        """Decode the temperature (offset)."""
        try:
            val = regs[0]
            return int_round((float(val) * abs(self.factor)) - self.offset)  # type: ignore
        except (TypeError, ValueError) as err:
            _LOGGER.error("Could not decode temperature: %s", err)
        return None
        
**I'm not a code specialist, but chatGPT says:**

class TempSensor(Sensor):
    """Offset by a default of 100 for temperature, unless overridden."""

    def __init__(self, reg_address, name, unit, factor, offset=100):
        """Initialize the TempSensor with a specific offset if provided."""
        super().__init__(reg_address, name, unit, factor)
        self.offset = offset

    def reg_to_value(self, regs: RegType) -> ValType:
        """Decode the temperature (offset)."""
        try:
            val = regs[0]
            # Use self.offset for the offset value
            return int_round((float(val) * abs(self.factor)) - self.offset)  # type: ignore
        except (TypeError, ValueError) as err:
            _LOGGER.error("Could not decode temperature: %s", err)
        return None

This will set the offset to 1000 for this particular sensor while others would still use the default offset of 100.

OwlBawl, 2024-10-17:
seems the value offset doubled with sensors.py main file:

class TempSensor(Sensor):
    """Offset by 100 for temperature."""

    offset: int = 100

    def reg_to_value(self, regs: RegType) -> ValType:
        """Decode the temperature (offset)."""
        try:
            val = regs[0]
            return int_round((float(val) * abs(self.factor)) - self.offset)  # type: ignore
        except (TypeError, ValueError) as err:
            _LOGGER.error("Could not decode temperature: %s", err)
        return None

OwlBawl, 2024-10-18:
Didn't get what you mean... My custom sensor I wrote few messages earlier is working correct. My last message was about your "dev" add-on test with new temp sensor... that showed wrong temp... and code was from addon sensors.py
But I'm not an expert and do not code, so maybe I was wrong. 
Anyway.. the battery temp sensor still incorrect (as on last screen) in main and dev addon

OwlBawl, 2024-10-18:
Battery temp sensor shows
Main branch -78,0 °C
Dev branch - 978 °C
While 22 °C in real

OwlBawl, 2024-10-18:
made custom 
SENSORS += Sensor(182, "Battery Test", CELSIUS, 0.1)
and.... it shows 22 °C

kellerza, 2024-10-19:
> SENSORS += Sensor(182, "Battery Test", CELSIUS, 0.1)
> and.... it shows 22 °C

This shows the offset is 0 😉

OwlBawl, 2024-10-19:
So the integrated sensor has extra offset?
Could fix the add-on now?

OwlBawl, 2024-11-07:
any plans to fix Battery temp sensor for Deye? Showing wrong data
[image]

While custom sensor
SENSORS += Sensor(182, "Battery Temp", CELSIUS, 0.1)
shows correct.

maslyankov, 2024-12-28:
> 182

What he means is that by having this definition:

SENSORS += Sensor(182, "Battery Temp", CELSIUS, 0.1)

You are effectively not setting any offset to the value. In deye modbus docs, it states that the "battery temperature" register is offset by 1000, so therefore the definition of "TempSensor(182, "Battery temperature", CELSIUS, 0.1, offset=1000)" in single-phase definition file.  Is the reading you get same on the inverter screen as in Home Assistant/Sunsynk? Also is your inverter recently updated?

OwlBawl, 2024-12-28:
> What he means is that by having this definition:
>
> SENSORS += Sensor(182, "Battery Temp", CELSIUS, 0.1)
>
> You are effectively not setting any offset to the value. In deye modbus docs, it states that [...]

That behavior is just from the beginning of multi add-on usage - 3 firmwares of Deye SUN 6k-SG03/05 inverter incl. stock one.
The default Temp shows -75C
While custom made - correctly - 25C 
as on screenshot i've provided.

maslyankov, 2024-12-29:
> That behavior is just from the beginning of multi add-on usage - 3 firmwares of Deye SUN 6k-SG03/05 inverter incl. stock one. The default Temp shows -75C While custom made - correctly - 25C as on [...]

I have fixed it. Please try the following docker image: ghcr.io/maslyankov/hass-addon-sunsynk-multi:edge
Please let me know if it works or not. Thanks!

maslyankov, 2025-01-03:
@OwlBawl did you manage to test the above version?

OwlBawl, 2025-01-03:
No, it's disconnected for service now, so can't check changes.

kellerza, 2025-01-06:
@OwlBawl I realixed what I did wrong here, when you omit the offset it becomes 100, not 0.

This aligns with the docs (1000 before the 0.1 factor) and works ok for my Sunsynk

Can you try the edge version when you have things back up and running?

maslyankov, 2025-01-15:
close issue (as it seems fixed with above commit)?

kellerza, 2025-01-15:
@OwlBawl feel free to re-open is not ok. This work for my single-phase inverter

closed 2025-01-15 by kellerza

slipx06, 2025-03-07:
@kellerza my battery temp is still incorrect on a Deye 8kW. Using the latest definitions

[image]

```

##########
# Battery
##########
SENSORS += (
    TempSensor(182, "Battery temperature", CELSIUS, 0.1),
    Sensor(183, "Battery voltage", VOLT, 0.01),
    Sensor(184, "Battery SOC", "%"),
    Sensor(190, "Battery power", WATT, -1),
    Sensor(191, "Battery current", AMPS, -0.01),
    # Charge and Discharge limit vary based on temperature and SoC
    Sensor(314, "Battery Charge Limit current", AMPS, -1),  # #327
    Sensor(315, "Battery Discharge Limit current", AMPS, -1),  # #327
)
```