kellerza/sunsynk #359: Modbus to Sunsynk inverter communication stops working after a while

heinrichpieterse, 2024-11-04:
### Issue related to

Sunsynk / pymodbus with Home Assistant Add-On, modbus communication stops working after a while.

### Describe the issue/bug

My modbus to Sunsynk Inverter with HA addon was working perfectly for 6 months, until recently I started to get errors (see logs: nonzero urb status: -71) and the communication freezes (readings stay constant). Plugging the USB to RS485 adapter out and back in fixes the comms again. Happens again within 12 to 24 hours after restoring comms.

### Your environment

- HA running in a VM (VM Workstation Player) on Windows host. 
- Home Assistant Core 2024.9.1, Home Assistant OS 13.1
- Addon:
  - Name: sunsynk-multi
  - Version: 0.7.0 (Was on 0.7.4, reverted back to 0.7.0 to see if it was breaking change, but was not the cause)
- Inverter:
  - Make: Sunsynk
  - Model: 8kW Inverter 1-phase
  - Firmware: TBD
- Adaptor details:
  - Wave USB to RS485 W17286 (Plugged into Windows host pc, usb routed to HA VM)

### You configuration

```DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "2xxxxxxx5"
    HA_PREFIX: SS
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: /dev/ttyUSB0
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
  - pv2_voltage
  - pv2_current
  - settings
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 5
    CHANGE_ANY: false
    CHANGE_BY: 50
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: box
PROG_TIME_INTERVAL: 15
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mosquitto_user
MQTT_PASSWORD: xxxx
DEBUG_DEVICE: /dev/ttyS0

```

### Logs

```logs
From HA CLI:

[25228.851791] ftdi_sio ttyUSB0: error from flowcontrol urb
[25256.895267] ftdi_sio ttyUSB0: usb_serial_generic_write_bukl_callback - nonzero urb status: -71
[25256.949869] ftdi_sio ttyUSB0: error from flowcontrol urb
[63756.152281] ftdi_sio ttyUSB0: usb_serial_generic_write_bulk_callback - nonzero urb status: -71

```

kellerza, 2024-11-04:
Since you have direct serial, try adding the mbusd addon. 

Then the sunsynk addon uses IP to mbusd adn mbusd connects to the serial interface. Mine is rock solid, and I run mbusd on an old Rpi 1B

HeinPi, 2024-11-04:
Thanks, will try it. PS. You're an absolute legend for creating and supporting this add-on so well! Thank you!

HeinPi, 2024-11-04:
I have it running with the mbusd now. So still using pymodbus with port tcp://homeassistant.local:502, is that the best way? Lets see if it resolves my issue.

HeinPi, 2024-11-26:
Issue still persists with mbusd, I tried to eliminate a couple of environmental variables like different usb port, possible emf interference from other hardware, redoing wire connections. Any suggestions would be welcome.

closed 2025-03-18 by kellerza