kellerza/sunsynk #360: MQTT often fails

gurmukhp, 2024-11-04:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

Sometimes, the addon stops sending any definitions via MQTT to Home Assistant.

**Expected behavior**
MQTT should be stable and not stop updating.

To help mitigate this issue, I've set the Raspberry Pi to restart every morning. This has definitely helped, but it still occurs. A restart *always* fixes the issue.

Is there a way I can find out why this is happening? Or even better just maybe restart the MQTT server in the AddOn every 30 mins or so?

[image]

Thanks!

### Your environment

- Home Assistant version: 2024.7.0
- Addon:
  - Name: sunsynk-multi
  - Version: 2023-08-17
- Inverter: Sunsynk 3.6kW ECCO 
- Adaptor details: Waveshare Industrial USB to RS485 Converter

```
---
DRIVER: pymodbus
INVERTERS:
- SERIAL_NR: '2212138137'
  HA_PREFIX: Sunsynk (Keller ZA)
  MODBUS_ID: 1
  PORT: /dev/ttyUSB0
SENSOR_DEFINITIONS: single-phase
SENSORS:
- energy_management
- power_flow_card
- settings
- solar_export
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Sunsynk
READ_SENSORS_BATCH_SIZE: 60
NUMBER_ENTITY_MODE: auto
MQTT_HOST: 192.168.x.x
MQTT_PORT: 1883
MQTT_USERNAME: ''
MQTT_PASSWORD: ''
DEBUG: 0
DEBUG_DEVICE: "/dev/ttyAMA0"
```

kellerza, 2024-11-04:
Look at the logs of both addons

kellerza, 2025-01-15:
Feel free to reopen with logs of both addons

closed 2025-01-15 by kellerza

gurmukhp, 2025-01-15:
Apologies for not getting the logs to you. I plan to get to this at some point. For now, I just set Home Assistant to restart the server once the MQTT stuff is offline and that seems to work well. It goes offline once every 2 weeks or so and never recovers unless restarted.