kellerza/sunsynk #361: Unknown register value 512

deondti, 2024-11-04:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

When i run Solarman it runs perfect.
Pymodbus not working on my 16Kw DEYE
""WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions""
and
""ERROR   (12 in 5 min) OSError in callback read_ss: timeout reading 1 registers from 603; timeout reading 1 registers from 611 poll_need_to_read""
are the errors I receive in the log, no posting to MQTT, MQTT is connected, the only change is the solarman/pymodbus connection.

**Expected behavior**
No Errors, posting to MQTT

### Your environment

- Home Assistant version: 2023.8.x
- Addon:
  - Name: sunsynk-multi
  - Version:  0.7.5 , latest
- Inverter:
  - Make: DEYE
  - Model: 16Kw
  - Firmware: main; 9025-1726 , protocol V0.2.0.1
- Adaptor details:
  - RS485/RS232/solarman
  - RS485/RS232 ftdi

### You configuration

```yaml
DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "2404180592"
    HA_PREFIX: SS
    MODBUS_ID: 1
    DONGLE SERIAL NUMBER: "0"
    PORT: /dev/ttyUSB0
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
  - pv3_power
  - all
  - battery_soc
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Deye
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: 15
MQTT_HOST: 192.168.20.22
MQTT_PORT: 1884
MQTT_USERNAME: ICM
MQTT_PASSWORD: ICM123456
DEBUG_DEVICE: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

```

[... 113 lines cut ...]
|     |      battery_shutdown_voltage, prog3_capacity, prog4_power, prog4_voltage,       |
|     |    prog6_capacity, battery_charge_limit_current, prog4_mode, prog1_capacity,     |
|     |     battery_current, battery_max_discharge_current, battery_restart_voltage,     |
|     |              prog4_capacity, battery_float_voltage, total_pv_energy              |
+-----+----------------------------------------------------------------------------------+
12:50:43 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
12:50:43 ERROR   OSError in callback read_ss: timeout reading 1 registers from 603; timeout reading 1 registers from 611 poll_need_to_read

Traceback (most recent call last):

  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/timer_callback.py", line 81, in wrap_callback
    await cb_call

  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/sensor_callback.py", line 110, in callback_sensor
    await ist.read_sensors(

  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 54, in read_sensors
    await self.inv.read_sensors(sensors)

  File "/usr/local/lib/python3.11/site-packages/sunsynk/sunsynk.py", line 134, in read_sensors
    raise IOError("; ".join(errs))

OSError: timeout reading 1 registers from 603; timeout reading 1 registers from 611 poll_need_to_read

```

kellerza, 2024-11-04:
Remove sensor all. Use less sensors

deondti, 2024-11-04:
It was exactly that, no more errors and it's pushing to mqtt.

I will add my sensors.

Thank you

closed 2024-11-05 by kellerza