kellerza/sunsynk #366: Some definitions in three_phase_hv.py have the wrong multiplier and/or are “signed 16-bit values”

JimRockford1, 2024-11-08:
### Your environment
- Inverter:
  - Make: DEYE 
  - Model: SUN 20K-SG01 HP3-EU-AM2

### Describe the issue/bug and what you expect

Some definitions have the wrong multiplier and/or are “signed 16-bit values”:

In three_phase_hv.py (https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/three_phase_hv.py):

Sensor(215, “Battery 1 BMS voltage”, VOLT, 0.01)  ->  the correct factor is “0.1”.
Sensor(216, “Battery 1 BMS current”, AMPS)  ->  the correct factor is “0.1”. It is also a “signed 16 bit value” (I don't know how to define this in the sensor description).

In three_phase_common.py (https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/three_phase_common.py):

Sensor(607, “Total Active Power from Side to Side of the Grid”, “W”)  ->  This is also a “signed 16 bit value”.

maslyankov, 2024-12-28:
> ### Your environment
> * Inverter:
>
>   * Make: DEYE
>   * Model: SUN 20K-SG01 HP3-EU-AM2
>
> ### Describe the issue/bug and what you expect
> Some definitions have the wrong multiplier and/or are “signed [...]

Hi! For Sensor(607, “Total Active Power from Side to Side of the Grid”, “W”) -> This is also a “signed 16 bit value”. , I see it mentioned as Signed 16 in modbus docs. For 215, 216, I do not see it mentioned to be a Signed 16 bit. How did you come to this conclusion?

maslyankov, 2024-12-28:
Verified issues (with modbus docs) have been fixed in https://github.com/maslyankov/sunsynk/pull/7

kellerza, 2025-01-06:
Thanks @maslyankov - this helped, feel free to create a PR in the future as well :wink:

I also added the Total active power from side to side.
BMS Amps could very well be signed, since you can charge/discharge the battery, so I added these

@JimRockford1 - can you check if these also need a 0.1 factor?

```python
    Sensor(212, "Battery 1 BMS charging current limit", AMPS),
    Sensor(213, "Battery 1 BMS discharging current limit", AMPS),
```