kellerza/sunsynk #367: Update issue

Saurman, 2024-11-09:
About a week ago I've started to receive update failed messages. Restart did not help

> Failed to perform the action update/install. Error updating Sunsynk/Deye Inverter Add-on (edge/dev): Can't install ghcr.io/kellerza/hass-addon-sunsynk-multi/aarch64:92dd837: 500 Server Error for http+docker://localhost/v1.47/images/create?tag=92dd837&fromImage=ghcr.io%2Fkellerza%2Fhass-addon-sunsynk-multi%2Faarch64&platform=linux%2Farm64: Internal Server Error ("manifest unknown")

deondti, 2024-11-14:
I'm getting the same but with the DEV version.
There was a update available , it failed to update, I eventually uninstalled it , now it won't re install  ( manifest unknown)

"""Can't install ghcr.io/kellerza/hass-addon-sunsynk-multi/aarch64:92dd837: 500 Server Error for http+docker://localhost/v1.47/images/create?tag=92dd837&fromImage=ghcr.io%2Fkellerza%2Fhass-addon-sunsynk-multi%2Faarch64&platform=linux%2Farm64: Internal Server Error ("manifest unknown")
"""

I have patiently been waiting for it to get resolved

kellerza, 2024-11-14:
@maslyankov - should we build for more platforms, or can we point the addon config to the multi-platform build?

Today, in config.yml, we have 

```
version: "92dd837"
image: ghcr.io/kellerza/hass-addon-sunsynk-multi/{arch}
arch:
  - aarch64
  - amd64
  - armhf
  - armv7
  - i386
```

and [hass-addon-sunsynk-multi/aarch64](https://github.com/users/kellerza/packages/container/package/hass-addon-sunsynk-multi%2Faarch64) is not being updated

maslyankov, 2024-11-14:
In the context of Docker images, aarch64 and arm64 are often used interchangeably because they both refer to the 64-bit ARM architecture. Technically, aarch64 is the architecture name defined by the ARM company, while arm64 is commonly used by many platforms, including Docker and Linux distributions, as an alias.

So, while aarch64 and arm64 may appear in different contexts, they refer to the same 64-bit ARM architecture. Docker will typically recognize both as equivalent, but when building multi-platform Docker images, it’s standard to use arm64 as the platform designation.

I think it's safe to point the addon config to the multiplatform image.

kellerza, 2024-11-14:
Thanks @maslyankov - I removed `/{arch}` from the edge addon

@Saurman @deondti - make sure you update the store info - "check for updates" should hopefully do it & then try again

deondti, 2024-11-14:
@kellerza Mine has now download successfully , thank you for this project and all your efforts, I don't think you always get the gratitude for this, I'm sure I speak for many that we really appreciate this.

Saurman, 2024-11-14:
@kellerza fix confirmed, updated to 92dd837

closed 2024-11-14 by Saurman