kellerza/sunsynk #368: ERROR: timeount

mdeakin99, 2024-11-13:
### Issue related to

Sunsynk Home Assistant Add-On

### Describe the issue/bug

Addon unable to read from the sensors/inverter/registers.  I have tried

1) Two different waveshare dongles
2) Tried using modbus addon and without modbus addon
3) Connecting green wire to ground and connecting green/white wire to ground

### Your environment

- Home Assistant version: 2024.11.1
- Addon:
  - Name: sunsynk-multi
  - Version: 2023-08-17
- Inverter:
  - Make: Sunsynk
  - Model: ECO 3.6kw (I believe)
  - Firmware: 2.3.8.6
- Adaptor details:
   - RS485/RS232 brand WaveShare

### You configuration

```yaml

DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "SERIANNUMBERHERE"
    HA_PREFIX: SS
    MODBUS_ID: 1
    PORT: /dev/ttyACM0
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: sunsynkdirect
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 2
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: "60"
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqtt_user
MQTT_PASSWORD: mpqttPasswordHere

```

### Logs

From Sunsynk addon

s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
14:56:27 INFO    Loading configuration: /data/options.json
14:56:27 INFO    sunsynk library version: 0.6.5
14:56:27 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|    enum   |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  5   |   60   |     80    |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
14:56:27 INFO    Using Single phase sensor definitions.
14:56:27 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
14:56:27 INFO    Connecting to /dev/ttyACM0
14:56:27 INFO    PyModbus 3.7.2 Serial: /dev/ttyACM0
14:56:27 INFO    Reading startup sensors Battery Low Capacity, Rated power, Prog1 Time, Serial, Prog4 Time, Prog2 Time, Prog6 Time, Prog5 Time, Prog3 Time
14:57:55 ERROR   timeout reading 2 registers from 3; timeout reading 2 registers from 5; timeout reading 1 registers from 7; timeout reading 2 registers from 16; timeout reading 1 registers from 219; timeout reading 2 registers from 250; timeout reading 2 registers from 252; timeout reading 2 registers from 254

```logs

paste logs in 3x backticks...

```

Jacquedebie, 2024-12-12:
Getting the same error.. Started about 3 weeks back

JoshyZA, 2024-12-18:
I am having the same issue. I am hoping someone has an easy fix to this as it seems to be something on the inverter side from the tests I have done.

### Issue related to:

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

The add-on does not seem to be able to read/receive the values from the inverter.
It seems to me that there's an issue on the inverter side (maybe firmware)

Things I have tried:
1. Checked that my Sunsynk inverter is up to date on firmware.
2. Checked the serial number of the inverter is correct.
3. Changed the inverters Modbus SN to "2" and "1" while adjusting it on the add-on as well.
4. Did a full reboot on the inverter
5. Moved my HAOS server to another premises which has an 8kw Sunsync in and this add-on worked perfectly fine with all the same hardware. The only difference was the inverter and internal network.

**Expected behaviour**
I expected it to work on the 5kw as it does on the 8kw

### Your environment

- Home Assistant version: (Core) 2024.12.4
- Addon:
  - Name: sunsynk-multi
  - Version: [v0.7.5](https://github.com/kellerza/sunsynk/releases/tag/v0.7.5)
- Inverter:
  - Make: Sunsynk
  - Model: 5kw
  - Firmware: M 3.3.7.4 / S 1.5.1.5 / C E.4.2.4
- Adaptor details:
  - RS485
  - RS485 to USB with earth

### You configuration

```yaml

DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "Inverter Serial Number"
    HA_PREFIX: SS
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: ""
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
  - pv1_power
  - essential_power
  - battery_power
  - battery_soc
  - non_essential_power
  - grid_ct_power
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 8
SCHEDULES:
  - KEY: W
    READ_EVERY: 1
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
[... 71 lines cut ...]
19:49:06 INFO    Using Single phase sensor definitions.
19:49:06 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
19:49:06 INFO    Connecting to /dev/ttyUSB0
19:49:06 INFO    PyModbus 3.7.2 Serial: /dev/ttyUSB0
19:49:06 INFO    Reading startup sensors Prog3 Time, Prog6 Time, Prog4 Time, Prog5 Time, Prog2 Time, Battery Low Capacity, Prog1 Time, Serial, Rated power
19:49:50 ERROR   timeout reading 5 registers from 3; timeout reading 2 registers from 16; timeout reading 1 registers from 219; timeout reading 6 registers from 250
19:50:34 ERROR   timeout reading 5 registers from 3; timeout reading 2 registers from 16; timeout reading 1 registers from 219; timeout reading 6 registers from 250
19:51:18 ERROR   timeout reading 5 registers from 3; timeout reading 2 registers from 16; timeout reading 1 registers from 219; timeout reading 6 registers from 250
19:51:18 WARNING Retrying individual sensors: ['prog3_time', 'prog6_time', 'prog4_time', 'prog5_time', 'prog2_time', 'battery_low_capacity', 'prog1_time', 'serial', 'rated_power']
19:51:29 ERROR   timeout reading 1 registers from 252 Prog3 Time
19:51:41 ERROR   timeout reading 1 registers from 255 Prog6 Time
19:51:52 ERROR   timeout reading 1 registers from 253 Prog4 Time
19:52:03 ERROR   timeout reading 1 registers from 254 Prog5 Time
19:52:14 ERROR   timeout reading 1 registers from 251 Prog2 Time
19:52:25 ERROR   timeout reading 1 registers from 219 Battery Low Capacity
19:52:36 ERROR   timeout reading 1 registers from 250 Prog1 Time
19:52:47 ERROR   timeout reading 5 registers from 3 Serial
19:52:58 ERROR   timeout reading 2 registers from 16 Rated power
19:52:59 CRITICAL Could not read sensors: ['Prog3 Time', 'Prog6 Time', 'Prog4 Time', 'Prog5 Time', 'Prog2 Time', 'Battery Low Capacity', 'Prog1 Time', 'Serial', 'Rated power']
19:52:59 INFO    ############################################################
19:52:59 INFO    No response on the Modbus interface /dev/ttyUSB0, see https://kellerza.github.io/sunsynk/guide/fault-finding
19:52:59 INFO    ############################################################
19:52:59 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.

```

SyphonxZA, 2025-01-06:
Anyone managed to fix this? Mine started today after I updated HA and a few other containers to their latest versions, however my pi running the sunsynk multi container is separate and thus beyond network connectivity interruptions I can't see how there updates would break stat collections from my inverters.

SyphonxZA, 2025-01-06:
So got mine working again. I have 2x inverters in parallel and pass the USB0 and USB1 into the container. However I cannot control which inverter is assigned USB0 and which is USB1 on a reboot.

I believe the slave inverter was being assigned as USB0 on boot each time. After rebooting with the slave disconnected and only connecting it after boot my stats were working as expected.

So if you have multiple USB devices connected, make sure that the inverter is assigned the correct USB interface.

kellerza, 2025-01-06:
> However I cannot control which inverter is assigned USB0 and which is USB1 on a reboot.

If you look at debug_device in the settings, you might see the USB by ID as well - this should not change?

I see the following: `/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A9007CPR-if00-port0`

SyphonxZA, 2025-01-07:
> > However I cannot control which inverter is assigned USB0 and which is USB1 on a reboot.
> 
> If you look at debug_device in the settings, you might see the USB by ID as well - this should not change?
> 
> I see the following: `/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A9007CPR-if00-port0`

Thanks, had a look now and switched to using the device IDs. When I first set this up I only had one inverter so using USB0 was sufficient.

For reference the devices section in docker compose now looks like:

`    devices:
    - /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AG0KH9AW-if00-port0:/dev/ttyUSB1
    - /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AQ04AFK8-if00-port0:/dev/ttyUSB0
 `

kellerza, 2025-01-16:
The latest version reads the minimum amount of data (serial, device type, protocol version) as startup sensor, before it tries to read any more sensors.

If this step fails, there is unfortunately nothing that can be done until you fixed either the physical cabling, HW, port you plug in/splitter, connection settings, modbus id).

I always advise to go via mbusd and not use direct usb from the addon. My guess is that the timing for mbusd (written in C) is a lot better than the timing on the Python modbus driver (maybe due to gil/something else 🤷 ).

closed 2025-01-16 by kellerza

mdeakin99, 2025-01-16:
In the end I used a different piece of hardware and integration and it’s all working for me now.  Happy to share more if appropriate ie make direct conversation