kellerza/sunsynk #371: Bettery Low Capacity write fail

maslyankov, 2024-11-24:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/). You can test your own definitions using [custom sensors](https://kellerza.github.io/sunsynk/reference/mysensors)

### Your environment
- Home Assistant version: __
- Addon:
  - Name: **edge**
  - Version: latest
- Inverter:
  - Make: Deye
  - Model: SUN-30K
  - Firmware: 3102-1077-1D08
  - Protocol Version: 0104
- Adaptor details:
  - RS485 via EthernetTCP
  - 
### Describe the issue/bug and what you expect

When you try to set Battery Low Capacity value, it gets set to 0. I have noticed this behavior on other Deye models as well. It can be fixed/set to proper value from Solarman app.

[image]

Any suggestions on proposed registers is welcome.

### Logs (if applicable)

```logs

paste logs in 3x backticks...

```

gacopl, 2024-12-08:
Noticed the same here sg04lp3

maslyankov, 2024-12-26:
Same issue is with "Grid Charge Enabled". When we change setting from Sunsynk mqtt, it goes to an undefined state in Solarman, which results in "False" setting state (Not charging from grid).

maslyankov, 2024-12-26:
I found what the issue is... So it turns out sunsynk is writing wrong value into modbus registers for some  SwitchRWSensors. They should be 0x0001 (1) for True, but it writes 0x00FF (255) as seen in the logs below: 

```
// Setting to "ON"

sunsynk-mult-amd64-1  | 2024-12-26 21:56:33,152 INFO    sunsynk.sunsynk Writing sensor grid_charge_enabled=ON Registers:{130=0x00ff} 
sunsynk-mult-amd64-1  | 2024-12-26 21:56:33,304 DEBUG   sunsynk.sunsynk Time taken to fetch 1 registers starting at 130 : 0.04s
sunsynk-mult-amd64-1  | 2024-12-26 21:56:33,305 DEBUG   sunsynk.sunsynk Request registers: [130] glen=1. Response [255] len=1. regs={130: 255}
sunsynk-mult-amd64-1  | 2024-12-26 21:56:33,305 DEBUG   sunsynk.state register {3: 12851, 4: 12593, 5: 12339, 6: 13360, 7: 12596, 20: 54464, 21: 1, 117: 55, 148: 100, 149: 500, 150: 900, 151: 1300, 152: 1700, 153: 2100, 62: 6156, 63: 6677, 64: 13339, 102: 600, 108: 100, 109: 100, 127: 45, 128: 40, 129: 0, 
    130: 255, 
    141: 1, 142: 2, 146: 255, 147: 1, 154: 12000, 155: 12000, 156: 12000, 157: 12000, 158: 12000, 159: 12000, 166: 55, 167: 55, 168: 90, 169: 90, 170: 55, 171: 55, 172: 1, 173: 0, 174: 0, 175: 0, 176: 0, 177: 0, 500: 2, 514: 84, 515: 84, 516: 6996, 517: 0, 518: 6998, 519: 0, 520: 798, 521: 0, 522: 26247, 523: 0, 524: 5, 525: 0, 526: 764, 529: 3, 534: 11268, 535: 0, 552: 5, 555: 0, 556: 0, 557: 0, 558: 0, 586: 1076, 587: 5023, 588: 62, 589: 0, 590: 96, 591: 193, 609: 5000, 619: 5270, 622: 1769, 623: 1753, 624: 1748, 625: 5270, 636: 65453, 644: 2306, 645: 2307, 646: 2387, 650: 3369, 651: 184, 652: 1634, 653: 5187, 672: 0, 673: 0, 674: 0, 675: 0, 676: 1594, 677: 0, 678: 257, 679: 0} = False (old=True)

// Setting to "ON", properly (modbus register 130), config changed via Solarman app

sunsynk-mult-amd64-1  | 2024-12-26 21:59:05,359 DEBUG   sunsynk.state register {3: 12851, 4: 12593, 5: 12339, 6: 13360, 7: 12596, 20: 54464, 21: 1, 117: 55, 148: 100, 149: 500, 150: 900, 151: 1300, 152: 1700, 153: 2100, 62: 6156, 63: 6677, 64: 14106, 102: 600, 108: 100, 109: 100, 127: 45, 128: 40, 129: 0, 
    130: 1, 
    141: 1, 142: 2, 146: 255, 147: 1, 154: 12000, 155: 12000, 156: 12000, 157: 12000, 158: 12000, 159: 12000, 166: 55, 167: 55, 168: 90, 169: 90, 170: 55, 171: 55, 172: 1, 173: 0, 174: 0, 175: 0, 176: 0, 177: 0, 500: 2, 514: 84, 515: 84, 516: 6996, 517: 0, 518: 6998, 519: 0, 520: 798, 521: 0, 522: 26247, 523: 0, 524: 5, 525: 0, 526: 764, 529: 3, 534: 11268, 535: 0, 552: 5, 555: 0, 556: 0, 557: 0, 558: 0, 586: 1076, 587: 5023, 588: 62, 589: 0, 590: 94, 591: 189, 609: 5000, 619: 5242, 622: 1754, 623: 1746, 624: 1742, 625: 5242, 636: 65453, 644: 2303, 645: 2320, 646: 2384, 650: 3340, 651: 209, 652: 1610, 653: 5159, 672: 0, 673: 0, 674: 0, 675: 0, 676: 1169, 677: 0, 678: 259, 679: 0} = -83 (old=-82)

```

maslyankov, 2024-12-26:
The RWSwitches I think I can fix. For the battery low capacity, I am not yet sure what happens. This is when I try to set it to 50% in via sunsynk:

```
sunsynk-mult-amd64-1  | 2024-12-26 23:00:20,153 INFO    sunsynk.sunsynk Writing sensor battery_low_capacity=50 Registers:{117=0x0000} 
sunsynk-mult-amd64-1  | 2024-12-26 23:00:20,304 DEBUG   sunsynk.sunsynk Time taken to fetch 1 registers starting at 117 : 0.04s
sunsynk-mult-amd64-1  | 2024-12-26 23:00:20,305 DEBUG   sunsynk.sunsynk Request registers: [117] glen=1. Response [0] len=1. regs={117: 0}
sunsynk-mult-amd64-1  | 2024-12-26 23:00:20,305 DEBUG   sunsynk.state register {3: 12851, 4: 12593, 5: 12339, 6: 13360, 7: 12596, 20: 54464, 21: 1, 117: 0, 148: 100, 149: 500, 150: 900, 151: 1300, 152: 1700, 153: 2100, 62: 6156, 63: 6678, 64: 14360, 102: 600, 108: 100, 109: 100, 127: 45, 128: 40, 129: 0, 130: 1, 141: 1, 142: 2, 146: 255, 147: 1, 154: 12000, 155: 12000, 156: 12000, 157: 12000, 158: 12000, 159: 12000, 166: 55, 167: 55, 168: 90, 169: 90, 170: 55, 171: 55, 172: 1, 173: 0, 174: 0, 175: 0, 176: 0, 177: 0, 500: 2, 514: 84, 515: 86, 516: 6996, 517: 0, 518: 7000, 519: 0, 520: 863, 521: 0, 522: 26313, 523: 0, 524: 5, 525: 0, 526: 828, 529: 3, 534: 11268, 535: 0, 552: 5, 555: 0, 556: 0, 557: 0, 558: 0, 586: 1077, 587: 4975, 588: 61, 589: 0, 590: 166, 591: 339, 609: 4995, 619: 6486, 622: 2172, 623: 2156, 624: 2158, 625: 6486, 636: 65452, 644: 2305, 645: 2324, 646: 2383, 650: 6050, 651: 176, 652: 176, 653: 6402, 672: 0, 673: 0, 674: 0, 675: 0, 676: 1914, 677: 0, 678: 246, 679: 0} = 55 (old=0)
```

Basically goes down to zero (Register 117), as in the interface and Solarman app.

maslyankov, 2024-12-28:
> Noticed the same here sg04lp3

Pls let me know if you find any other switches/configs that are not working properly. Would be very helpful if you could also send along debug logs of values when setting from sunsynk (broken values) and when setting with Solarman app (working values). Thanks!

closed 2024-12-28 by maslyankov