kellerza/sunsynk #372: SunSynk Parallel single phase over Direct serial

RenierBoshoff, 2024-11-29:
Hallo good people.

With Direct Serial interface, how can I get two inverters to detect?
I have two SunSynk 8kW inverters in parallel as single phase, and I can get read outs from each separately (the MQTT detects two devices) (ie. comms working) but not together.

Any ideas what I can do?

I assumed that I need to set up the Addon like this, but no luck.
I have tried setting it to 3phase also.

[image]

I get this errors:
[image]

_Originally posted by @RenierBoshoff in https://github.com/kellerza/sunsynk/issues/131#issuecomment-2507595281_

Waz-Cpt, 2024-11-29:
Hello @RenierBoshoff ,

If you are running the units in Parallel for single phase, you would likely need to use helpers to sum up elements that you want as total values across both inverters.

Kind Regards,

RenierBoshoff, 2024-11-29:
Thanks @Waz-Cpt 
I see something like that in some code a friend used.

looks like this:
```
        # TOTAL CALCULATIONS
        load_power_total:
          friendly_name: "Load Power - Total"
          unit_of_measurement: 'W'
          value_template: "{{ states('sensor.ss_inv_loadpwr') | float + states('sensor.ss2_inv_loadpwr') | float }}"
```

Waz-Cpt, 2024-11-29:
It is really easy to use the helpers. You can do templates or just basic summation. Check the helpers in the devices sections, the is a tab for helpers.

kellerza, 2025-01-06:
@RenierBoshoff - can you post your config again? (Ek sien net "skermskoot" :wink:)

Are you sure you have 2 inverster configured? With different PREFIXES in the config?

```yaml
INVERTERS:
  - SERIAL_NR: "111xx"
    HA_PREFIX: SS
    MODBUS_ID: 1
    PORT: /dev/serial1
  - SERIAL_NR: "222xx"
    HA_PREFIX: SS
    MODBUS_ID: 2
    PORT: /dev/serial2
```

If you change the port/physically change the cable to the other inverter it will create 2 devices through MQTT because of different serial numbers

RenierBoshoff, 2025-01-07:
Hallo @kellerza 
Thanks for reaching out!

Here is the post again:

First screenshot was of the Options set to pymodbus:

```yaml
INVERTERS:
   - SERIAL_NR: "111xx"
     HA_PREFIX: SS
     MODBUS_ID: 1
     PORT: /dev/ttyUSB0
   - SERIAL_NR: "222xx"
     HA_PREFIX: SSlave
     MODBUS_ID: 2
     PORT: /dev/ttyUSB0
```

I see mine differs from yours in that your code have the same HA_PREFIX, mine different.
And yours have different PORT while mine is the same.

Do I need to use two different RS4855 to USB converters so that I can have two PORTs?

Here is the error screenshot: 
[image]
(I hope the attachment works)

Thanks again for assisting.

In the mean time I have configured the Modbus directly inside the Configuration.yaml, but sometimes I get weird readings like 66kW load while the actual load is pretty low. Then after a couple of minutes it shows the correct value again.
And I can not change any settings at the moment via HA

Kind regards

kellerza, 2025-01-07:
Can you paste your full config?

kellerza, 2025-03-18:
Partial config you pasted is missing from your main config - error: `Failed to save add-on configuration. Missing options 'INVERTERS' in root`

For direct serial I always suggest going via mbusd. It is not possible to have two direct connections to the same USB port without mbusd. 

See the docs for mbusd guidance

closed 2025-03-18 by kellerza