kellerza/sunsynk #374: SENSORS_FIRST_INVERTER creating same sef of entities on all inverters

gacopl, 2024-12-10:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

SENSORS_FRIST_INVERTER, impacts just reporting. 
So if I want for instance settings to be available only on Master inverter. 
Addon will produce All inverter devices with same set of entitties, just >1 inverter will show read only entities as unavailable in case of reading sensors. But in case of RW sensors they still work on the >1 inverter. 
I am able to change settings on Slave inverter even though i shouldn't

**Expected behavior**
If i populate SENSORS_FIRST_INVERTER, entities chosen in that section are present only on first inverter and addon produces devices with different entities list

### Your environment

- Home Assistant version: 2024.11.4
- Addon:
  - Name: sunsynk-multi
  - Version: 0.7.5
- Inverter:
  - Make: Deye
  - Model: 12K-SG04LP3
  - Firmware: does not matter
- Adaptor details:
  - RS485 waveshare
  -

### You configuration

```yaml

DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "XXX"
    HA_PREFIX: DeyeM
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: tcp://X:502
  - SERIAL_NR: "X"
    HA_PREFIX: DeyeS
    MODBUS_ID: 2
    DONGLE_SERIAL_NUMBER: "0"
    PORT: tcp://X:502
SENSOR_DEFINITIONS: three-phase
SENSORS:
  - energy_management
  - pv2_power
  - voltage
  - load_power
  - power_flow_card
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Deye
READ_ALLOW_GAP: 6
READ_SENSORS_BATCH_SIZE: 50
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 10
    CHANGE_ANY: true
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: 15

```

### Logs

```logs

paste logs in 3x backticks...

```

kellerza, 2025-01-09:
Can you try the latest edge and let me know if this is better?

gacopl, 2025-01-09:
hmm i dont see any change
```

gacopl, 2025-01-10:
I just updated to latest edge and send we are getting somewhere i finally saw mqtt deregisters of unneeded sensors. But also these errors

00:49:58 ERROR   ValueError in callback discovery_info: This entity should only be on the 1st inverter: ASensor(opt=SensorOption(sensor=NumberRWSensor(address=(170,), name='Prog5 Capacity', unit='%', factor=1, bitmask=0, min=NumberRWSensor(address=(117,), name='Battery Low Capacity', unit='%', factor=1, bitmask=0, min=NumberRWSensor(address=(115,), name='Battery Shutdown Capacity', unit='%', factor=1, bitmask=0, min=0, max=...), max=NumberRWSensor(address=(116,), name='Battery Restart Capacity', unit='%', factor=1, bitmask=0, min=..., max=100)), max=100), schedule=Schedule(key='rw', read_every=5, report_every=300, change_by=0, change_percent=0, change_any=True), visible=True, affects=set(), first=True), entity=None, _last=None, retain=True)
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/timer_callback.py", line 81, in wrap_callback
    await cb_call
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/driver.py", line 26, in callback_discovery_info
    ents = [s.create_entity(s.entity, ist=ist) for s in states if not s.hidden]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/driver.py", line 26, in <listcomp>
    ents = [s.create_entity(s.entity, ist=ist) for s in states if not s.hidden]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ha_addon_sunsynk_multi/a_sensor.py", line 123, in create_entity
    raise ValueError(f"This entity should only be on the 1st inverter: {self}")
ValueError: This entity should only be on the 1st inverter: ASensor(opt=SensorOption(sensor=NumberRWSensor(address=(170,), name='Prog5 Capacity', unit='%', factor=1, bitmask=0, min=NumberRWSensor(address=(117,), name='Battery Low Capacity', unit='%', factor=1, bitmask=0, min=NumberRWSensor(address=(115,), name='Battery Shutdown Capacity', unit='%', factor=1, bitmask=0, min=0, max=...), max=NumberRWSensor(address=(116,), name='Battery Restart Capacity', unit='%', factor=1, bitmask=0, min=..., max=100)), max=100), schedule=Schedule(key='rw', read_every=5, report_every=300, change_by=0, change_percent=0, change_any=True), visible=True, affects=set(), first=True), entity=None, _last=None, retain=True)
00:50:00 ERROR   (1 in 5 min) ValueError in callback discovery_info: This entity should only be on the 1st inverter: ASensor(opt=SensorOption(sensor=NumberRWSensor(address=(170,), name='Prog5 Capacity', unit='%', factor=1, bitmask=0, min=NumberRWSensor(address=(117,), name='Battery Low Capacity', unit='%', factor=1, bitmask=0, min=NumberRWSensor(address=(115,), name='Battery Shutdown Capacity', unit='%', factor=1, bitmask=0, min=0, max=...), max=NumberRWSensor(address=(116,), name='Battery Restart Capacity', unit='%', factor=1, bitmask=0, min=..., max=100)), max=100), schedule=Schedule(key='rw', read_every=5, report_every=300, change_by=0, change_percent=0, change_any=True), visible=True, affects=set(), first=True), entity=None, _last=None, retain=True)

kellerza, 2025-01-10:
Hopefully this is fixed now

gacopl, 2025-01-10:
yup works like a charm now THANK YOU VERY MUCH SIR

closed 2025-01-10 by gacopl