kellerza/sunsynk #378: Sensors ProgX Capacity missing for Deye SUN 25K HV

klacol, 2024-12-28:
### My environment
- Home Assistant version: 2024.12.5
- Addon:
  - Name: **multi** 
  - Version: 0.7.5
- Inverter:
  - Make: Deye SUN 25K Highvolt
  - Model: SUN-25K-SG01HP3-EU-AM2
  - Firmware: 3002-1077-1E08
- Adaptor details:
  - Waveshare USB-to-RS485 @ /dev/ttyAMA0 
  - 
### Describe the issue/bug and what you expect

I am seeking the following sensors for the HV-Battery:

```
    NumberRWSensor(166, "Prog1 Capacity", "%", min=BATTERY_LOW_CAP),
    NumberRWSensor(167, "Prog2 Capacity", "%", min=BATTERY_LOW_CAP),
    NumberRWSensor(168, "Prog3 Capacity", "%", min=BATTERY_LOW_CAP),
    NumberRWSensor(169, "Prog4 Capacity", "%", min=BATTERY_LOW_CAP),
    NumberRWSensor(170, "Prog5 Capacity", "%", min=BATTERY_LOW_CAP),
    NumberRWSensor(171, "Prog6 Capacity", "%", min=BATTERY_LOW_CAP),
```

The settings card looks like this:

[image]

They seem to be included in the [three_phase_common.py](https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/three_phase_common.py).  Is the three_phase_hv.py derived from the three_phase_common.py or are the sensors really missing in the three_phase_hv.py?

klacol, 2024-12-28:
Great integration here. I will check the custom sensors and will then give feedback here.

klacol, 2024-12-28:
I am wrong. The sensor definition was registered. Together with my prefix:

```yaml
Entity: number.deye_prog1_capacity
  Deye Prog1 Capacity

State: unknown	

Attributes:
  min: 20
  max: 100
  step: 1
  mode: auto
  unit_of_measurement: %
  icon: mdi:battery-lock
  friendly_name: Deye Prog1 Capacity
```

The state is missing, and therefore the card is complaining.

klacol, 2024-12-28:
This are the return results I get  when reading the ``System Work Mode-2`` in the [Deye Cloud](https://www.deyecloud.com/). Not sure, if this helps:

```yaml
98 Register:1
111 Register:1
146 Register:255
Time of Use:1
Spain Mode Enable:0
SOC2:20%
SOC2:170V
Time 1 Grid Charge:0
Time 1 Gen:0
Time 1 GM:0
TIme 1 BU:0
TIme 1 CH:0
Time 2 Grid Charge:0
Time 2 Gen:0
Time 2 GM:0
TIme 2 BU:0
TIme 2 CH:0
Time 3 Grid Charge:0
Time 3 Gen:0
Time 3 GM:0
TIme 3 BU:0
TIme 3 CH:0
Time 4 Grid Charge:0
Time 4 Gen:0
Time 4 GM:0
TIme 4 BU:0
TIme 4 CH:0
Time 5 Grid Charge:0
Time 5 Gen:0
Time 5 GM:0
TIme 5 BU:0
TIme 5 CH:0
Time 6 Grid Charge:0
Time 6 Gen:0
Time 6 GM:0
TIme 6 BU:0
TIme 6 CH:0
Selling Mode Time 1:0100
Selling Mode Time 2:0500
Selling Mode Time 3:0900
Selling Mode Time 4:1300
Selling Mode Time 5:1700
Selling Mode Time 6:2100
Time1 GPS:25000W
Time2 GPS:25000W
Time3 GPS:25000W
Time4 GPS:25000W
Time5 GPS:25000W
Time6 GPS:25000W
Time 1 Voltage1:290V
Time 2 Voltage1:290V
Time 3 Voltage1:290V
Time 4 Voltage1:290V
Time 5 Voltage1:290V
Time 6 Voltage1:290V
Time 1 SOC1:15%
Time 2 SOC1:15%
Time 3 SOC1:15%
Time 4 SOC1:15%
Time 5 SOC1:15%
Time 6 SOC1:15%
OK.
```

kellerza, 2024-12-31:
Can you please check if they appear if you explicitly include prog1_capacity, prog2_capacity in your config? (might not be part of the group)

HV extends the common sensors, see https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/three_phase_hv.py#L4

klacol, 2025-01-01:
Thanks for your hint (BTW Happy New Year). 

I have looked in HA, and now the cards are complete, even without any change. Not sure what has happed.

[image]

klacol, 2025-01-01:
Now, I can start to play around with it :-)

closed 2025-01-01 by klacol