kellerza/sunsynk #381: Wrong "ON" value for "Solar Export"

rrveex, 2025-01-02:
### Your environment
- Inverter:
  - Make: Deye
  - Model: SUN-15K-SG05LP3-EU-SM2
  - Firmware: "Ver 2021-1147-1807"

### Describe the issue/bug and what you expect

[image]

If I toggle this to "on", value `255` is written to modbus address `145`. The inverter *doesn't* turn on export.

If I (manually) write the value `1` to address `145`, the inverter starts exporting.

Value `0` (either manually or via GUI toggle) correctly stops export.

maslyankov, 2025-01-03:
I have fixed it. Please try the following docker image: ghcr.io/maslyankov/hass-addon-sunsynk-multi:edge
Please let me know if it works or not. Thanks!

rrveex, 2025-01-03:
Thank you for fixing!

Sorry, I'm running HA in a VM, it'd take probably a day or so for me to learn how to run and (re-)configure it in docker for testing. A quick search on how I could get to the .py file directly didn't help.

BTW, `112` (activate battery) - the doc is wrong, 0=disabled, 1=enabled, tested by checking/unchecking in the HMI and querying modbus.

maslyankov, 2025-01-03:
> Thank you for fixing!
>
> Sorry, I'm running HA in a VM, it'd take probably a day or so for me to learn how to run and (re-)configure it in docker for testing. A quick search on how I could get to the [...]

Thank you for verifying this! Will fix this as well. Meanwhile, you do not need to change your usage approach. You can use this as addon: https://sunsynk.wectrl.net/guide/getting-started

rrveex, 2025-01-03:
[image]

Works!

kellerza, 2025-01-06:
@rrveex this is not yet fixed if in a different fork

Two solutions could be on=1 or mask=1 (@rixxxx do you use this setting?)

@maslyankov - is the intention to create a PR/maintain your own fork?

maslyankov, 2025-01-06:
> @rrveex this is not yet fixed if in a different fork
> 
> Two solutions could be on=1 or mask=1 (@rixxxx do you use this setting?)
> 
> @maslyankov - is the intention to create a PR/maintain your own fork?

It was not my intention to maintain my own fork at first, but I guess I was expecting more timely responses to the PRs. I have made several improvements in my fork so far. Do you think you can make me a maintainer in this repo, so I can be more helpful?

kellerza, 2025-01-07:
I understand frustration with long reviews, so open to discuss. But usually, it's not more than 2/3 days, is it? 

PRs over my summer holidays is an exception, just back from ~month of no laptop, beach & grinding through Wind and Truth

closed 2025-01-10 by kellerza