kellerza/sunsynk #382: Entity cb_stats cannot be added as the entity category is set to config

SyphonxZA, 2025-01-06:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

I have 2x inverters in parallel and the stats are collected by a RPI 3 (running the latest docker container) connected to the RS485 port of each inverter. Stats come through fine but I get errors in HA logs for **cb_stats** and **timeout** sensors.

**Expected behavior**

Get rid of errors, either by removing sensors or fix underlying code. I have reviewed the sensors documentation (https://kellerza.github.io/sunsynk/reference/definitions) but it does not mention which predefined sensor group includes the timeout and CB stats sensors.

### Your environment

- Home Assistant version: 2025.1.0
- Addon:
  - Name: sunsynk-multi
  - Version: 0.7.5
- Inverter:
  - Make: Sunsynk
  - Model: 5KW Hybrid
- Adaptor details:
  - RS485
 
### You configuration

---
DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "REMOVED"
    HA_PREFIX: SS1
    MODBUS_ID: 01
    PORT: /dev/ttyUSB0
  - SERIAL_NR: "REMOVED"
    HA_PREFIX: SS2
    MODBUS_ID: 02
    PORT: /dev/ttyUSB1
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - power_flow_card
  - inverter_voltage
  - load_frequency
  - pv2_power
  - pv2_voltage
  - pv2_current
  - total_load_energy
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 8
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 30
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
MQTT_HOST: 192.168.10.50
MQTT_PORT: 1883
MQTT_USERNAME: REMOVED
MQTT_PASSWORD: REMOVED
DEBUG: 0
DEBUG_DEVICE: "/dev/ttyAMA0"

```yaml

paste config in the block surrounded by 3x backticks

```

### Logs

```logs

2025-01-06 17:28:43.842 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.ss1_cb_stats for domain sensor with platform mqtt

Traceback (most recent call last):

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity_platform.py", line 608, in _async_add_entities

    await coro

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity_platform.py", line 927, in _async_add_entity

    await entity.add_to_platform_finish()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/mqtt/entity.py", line 1101, in add_to_platform_finish

    await super().add_to_platform_finish()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity.py", line 1382, in add_to_platform_finish

    await self.async_internal_added_to_hass()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/sensor/__init__.py", line 258, in async_internal_added_to_hass

    raise HomeAssistantError(

        f"Entity {self.entity_id} cannot be added as the entity category is set to config"

    )

homeassistant.exceptions.HomeAssistantError: Entity sensor.ss1_cb_stats cannot be added as the entity category is set to config

2025-01-06 17:28:43.957 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.ss1_timeout for domain sensor with platform mqtt

Traceback (most recent call last):

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity_platform.py", line 608, in _async_add_entities

    await coro

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity_platform.py", line 927, in _async_add_entity

    await entity.add_to_platform_finish()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/mqtt/entity.py", line 1101, in add_to_platform_finish

    await super().add_to_platform_finish()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity.py", line 1382, in add_to_platform_finish

    await self.async_internal_added_to_hass()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/sensor/__init__.py", line 258, in async_internal_added_to_hass

    raise HomeAssistantError(

        f"Entity {self.entity_id} cannot be added as the entity category is set to config"

    )

homeassistant.exceptions.HomeAssistantError: Entity sensor.ss1_timeout cannot be added as the entity category is set to config

2025-01-06 17:28:45.354 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.ss2_cb_stats for domain sensor with platform mqtt

Traceback (most recent call last):

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity_platform.py", line 608, in _async_add_entities

    await coro

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity_platform.py", line 927, in _async_add_entity

    await entity.add_to_platform_finish()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/mqtt/entity.py", line 1101, in add_to_platform_finish

    await super().add_to_platform_finish()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity.py", line 1382, in add_to_platform_finish

    await self.async_internal_added_to_hass()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/sensor/__init__.py", line 258, in async_internal_added_to_hass

    raise HomeAssistantError(

        f"Entity {self.entity_id} cannot be added as the entity category is set to config"

    )

homeassistant.exceptions.HomeAssistantError: Entity sensor.ss2_cb_stats cannot be added as the entity category is set to config

2025-01-06 17:28:45.416 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.ss2_timeout for domain sensor with platform mqtt

Traceback (most recent call last):

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity_platform.py", line 608, in _async_add_entities

    await coro

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity_platform.py", line 927, in _async_add_entity

    await entity.add_to_platform_finish()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/mqtt/entity.py", line 1101, in add_to_platform_finish

    await super().add_to_platform_finish()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/helpers/entity.py", line 1382, in add_to_platform_finish

    await self.async_internal_added_to_hass()

  File "/usr/local/lib/python3.13/site-packages/homeassistant/components/sensor/__init__.py", line 258, in async_internal_added_to_hass

    raise HomeAssistantError(

        f"Entity {self.entity_id} cannot be added as the entity category is set to config"

    )

homeassistant.exceptions.HomeAssistantError: Entity sensor.ss2_timeout cannot be added as the entity category is set to config

```

SyphonxZA, 2025-01-06:
Fixed this by deleting the items from Home Assistant and rebooting. They were added back but seem to be behaving correctly now.

closed 2025-01-06 by SyphonxZA

kellerza, 2025-01-06:
You can also try the "edge" verison, as this already has these defined as diagnostic

I'll try to create a new multi from edge soon - there has been a couple of enhancements in the edge version