kellerza/sunsynk #387: WARNING device_type: Unknown register value 512

Wayne-nulitics, 2025-01-10:
### Your environment
- Home Assistant version: Core 2025.1.2 / Supervisor 2024.12.3 / OS 14.1 / Frontend 20250109.0
- Addon:
  - Name: Sunsynk/Deye Inverter Add-on (edge/dev)
  - Version: bd28f7f 
- Inverter:
  - Make: Sunsynk
  - Model: 16K MAX
  - Firmware: 
- Adaptor details:
  - RS485/RS232
  - 
### Describe the issue/bug and what you expect

Log gets filled with the following warning:

WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions

15:01:16 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:01:17 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:01:23 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:01:31 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:01:32 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:01:38 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:01:46 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:01:47 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:01:53 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:01 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:02 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:08 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:16 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:17 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:23 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:31 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:35 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:43 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:44 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:45 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:52 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:02:55 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
15:03:03 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions

kellerza, 2025-01-10:
Can you share your exact inverter model?

The new edge version should give less warnings

Wayne-nulitics, 2025-01-10:
Sure, the Sunsync 16.0kW, Single Phase LV Hybrid (3x MTTP)
Software: M 9.0.2.7 / S 1.7.2.6  / C E.4.3.D

Wayne-nulitics, 2025-01-10:
Some more logs:

14:57:39 INFO    Using Single phase sensor definitions.
14:57:39 INFO    Added hidden sensors as other sensors depend on it: Device type, Serial, Rated power
14:57:39 INFO    Connecting to /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A10P75QI-if00-port0
14:57:39 INFO    PyModbus 3.8.3 Serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A10P75QI-if00-port0
14:57:39 INFO    Reading startup sensors Device type, Serial
14:57:45 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
14:57:45 INFO    ############################################################
14:57:45 INFO                 Inverter serial number '****xxxxx'
14:57:45 INFO    ############################################################
14:57:45 INFO    Device type: None, using the single-phase sensor definitions
14:57:45 INFO    Reading all sensors Device type, Serial, PV1 power, PV2 power, PV3 power, Inverter power, Battery power, Load power, Grid power, Essential power, Grid Connected, Non-Essential power, Grid CT power, Max Solar power, Day Battery Discharge, PV1 current, PV2 current, PV3 current, PV1 voltage, PV2 voltage, PV3 voltage, Inverter voltage, Overall state, Day Battery Charge, Battery voltage, Battery SOC, Battery current, Day Load Energy, AUX power, Total PV Energy, Inverter current, Inverter frequency, Day Grid Export, Day Grid Import, Day PV Energy, Grid frequency, Grid L2 power, Grid voltage, Load frequency, Load L1 power, Load L2 power, Priority Load, Use Timer, Rated power, Load Limit, Prog1 Capacity, Prog1 charge, Prog1 power, Prog1 Time, Prog2 Capacity, Prog2 charge, Prog2 power, Prog2 Time, Prog3 Capacity, Prog3 charge, Prog3 power, Prog3 Time, Prog4 Capacity, Prog4 charge, Prog4 power, Prog4 Time, Prog5 Capacity, Prog5 charge, Prog5 power, Prog5 Time, Prog6 Capacity, Prog6 charge, Prog6 power, Prog6 Time, Date Time, Grid Charge Battery current, Grid Charge enabled, Solar Export, Battery Max Charge current, Battery Max Discharge current, Battery Shutdown Capacity, Battery Restart Capacity, Battery Low Capacity
14:57:46 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
14:57:46 INFO    MQTT: Connecting to wventer@core-mosquitto:1883
14:57:47 INFO    MQTT: Connection successful
14:57:48 INFO    Read every (inverter 1)
+-----+---------------------------------------------------------------------------------+
|  s  |                                     Sensors                                     |
+-----+---------------------------------------------------------------------------------+
|  5  |         prog4_capacity, battery_max_discharge_current, essential_power,         |
|     |     grid_charge_battery_current, grid_power, prog1_capacity, load_l2_power,     |
|     |            solar_export, battery_max_charge_current, prog3_capacity,            |
|     |  battery_restart_capacity, load_limit, prog6_capacity, prog5_charge, pv1_power, |
|     | aux_power, prog1_charge, prog6_charge, prog3_power, battery_power, device_type, |
|     |       battery_low_capacity, prog2_charge, non_essential_power, prog3_time,      |
|     |  inverter_power, prog4_time, use_timer, prog6_power, prog2_time, prog4_charge,  |
|     | prog2_capacity, prog6_time, prog3_charge, prog5_power, prog1_power, prog1_time, |
|     |       prog5_time, grid_l2_power, grid_ct_power, pv2_power, load_l1_power,       |
|     | max_solar_power, prog5_capacity, rated_power, prog2_power, grid_charge_enabled, |
|     |   pv3_power, prog4_power, battery_shutdown_capacity, priority_load, load_power  |
|  15 | overall_state, inverter_current, serial, battery_soc, pv3_current, pv1_current, |
|     | grid_connected, pv1_voltage, grid_frequency, inverter_voltage, battery_current, |
|     |   inverter_frequency, load_frequency, pv2_voltage, grid_voltage, pv2_current,   |
|     |                           battery_voltage, pv3_voltage                          |
|  60 |                                    date_time                                    |
| 300 |    day_pv_energy, day_battery_discharge, day_grid_export, day_battery_charge,   |
|     |                day_grid_import, day_load_energy, total_pv_energy                |
+-----+---------------------------------------------------------------------------------+
14:57:48 INFO    Report every (inverter 1)
+-----+---------------------------------------------------------------------------------+
|  s  |                                     Sensors                                     |
+-----+---------------------------------------------------------------------------------+
|  60 |    load_l1_power, grid_ct_power, pv1_power, pv2_power, aux_power, date_time,    |
|     |      battery_power, essential_power, grid_power, pv3_power, load_l2_power,      |
|     |          load_power, non_essential_power, grid_l2_power, inverter_power         |
| 300 |           prog4_capacity, battery_soc, battery_max_discharge_current,           |
|     |   grid_charge_battery_current, day_grid_import, prog1_capacity, solar_export,   |
|     |      battery_max_charge_current, battery_restart_capacity, prog3_capacity,      |
|     |    load_limit, total_pv_energy, day_pv_energy, prog6_capacity, prog5_charge,    |
|     |    prog1_charge, prog6_charge, grid_connected, prog3_power, inverter_voltage,   |
[... cut ...]
14:58:17 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
14:58:23 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
14:58:34 WARNING device_type: Unknown register value 512. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions

Wayne-nulitics, 2025-01-10:
And the config file:

DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "xxx"
    HA_PREFIX: SS
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A10P75QI-if00-port0
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - pv1_power
  - pv2_power
  - pv3_power
  - inverter_power
  - battery_power
  - load_power
  - grid_power
  - essential_power
  - grid_connected
  - non_essential_power
  - grid_ct_power
  - max_solar_power
  - day_battery_discharge
  - pv1_current
  - pv2_current
  - pv3_current
  - pv1_voltage
  - pv2_voltage
  - pv3_voltage
  - inverter_voltage
  - overall_state
  - day_battery_charge
  - battery_voltage
  - battery_soc
  - battery_current
  - day_load_energy
  - aux_power
  - total_pv_energy
  - inverter_current
  - inverter_frequency
  - power_flow_card
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 60
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: 15
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: xxx
MQTT_PASSWORD: xxx

kellerza, 2025-01-11:
Please try the latest version.

Would be interested to see your Protocol version, which should be printed in the latest version

```
14:57:45 INFO ############################################################
14:57:45 INFO Inverter serial number '****xxxxx'
14:57:45 INFO ############################################################
14:57:45 INFO Device type: None, using the single-phase sensor definitions
        ?????????????????????
14:57:45 INFO Reading all sensors 
````

Wayne-nulitics, 2025-01-11:
Stable now, you solve it.
...
06:06:38 INFO    Device type: Single phase LV hybrid (3x MTTP), using the single-phase sensor definitions
06:06:38 INFO    Protocol version: 1.0
06:06:38 INFO    Reading all sensors Device type, Protocol, Serial, PV1 power, PV2 power, PV3 power, Inverter power, Battery power, Load power, Grid power, Essential power, Grid Connected, Non-Essential power, Grid CT power, Max Solar power, Day Battery Discharge, PV1 current, PV2 current, PV3 current, PV1 voltage, PV2 voltage, PV3 voltage, Inverter voltage, Overall state, Day Battery Charge, Battery voltage, Battery SOC, Battery current, Day Load Energy, AUX power, Total PV Energy, Inverter current, Inverter frequency, Day Grid Export, Day Grid Import, Day PV Energy, Grid frequency, Grid L2 power, Grid voltage, Load frequency, Load L1 power, Load L2 power, Priority Load, Use Timer, Rated power, Load Limit, Prog1 Capacity, Prog1 charge, Prog1 power, Prog1 Time, Prog2 Capacity, Prog2 charge, Prog2 power, Prog2 Time, Prog3 Capacity, Prog3 charge, Prog3 power, Prog3 Time, Prog4 Capacity, Prog4 charge, Prog4 power, Prog4 Time, Prog5 Capacity, Prog5 charge, Prog5 power, Prog5 Time, Prog6 Capacity, Prog6 charge, Prog6 power, Prog6 Time, Date Time, Grid Charge Battery current, Grid Charge enabled, Solar Export, Battery Max Charge current, Battery Max Discharge current, Battery Shutdown Capacity, Battery Restart Capacity, Battery Low Capacity
06:06:39 INFO    MQTT: Connecting to x@core-mosquitto:1883
06:06:40 INFO    MQTT: Connection successful

closed 2025-01-15 by kellerza