kellerza/sunsynk #388: v0.8.1 hasn't built for aarch64

patch0, 2025-01-10:
### Issue related to

hass-addon-sunsynk-multi/aarch64 docker image

### Describe the issue/bug

v0.8.1 of package hasn't been [built/published for aarch64](https://github.com/kellerza/sunsynk/pkgs/container/hass-addon-sunsynk-multi%2Faarch64)

### Expected behavior

It is automatically built on publish :)

### Help..

I took a look through your github CI file, and it *should* be setting `architectures` correctly, as

```json
["aarch64","amd64","armhf","armv7","i386"]
```
and then the availability is set correctly too

```
aarch64=true
amd64=true
armhf=true
armv7=true
i386=true
```

I'm not sure why the aarch64 architecture is missing from the matrix of builds on the [v0.8.1 tag build](https://github.com/kellerza/sunsynk/actions/runs/12711966660)

kellerza, 2025-01-10:
Do you use the multi-arch version?

kellerza, 2025-01-10:
Likely related to the config.yaml of the multi version

patch0, 2025-01-10:
Yeah, so I'm using the multi version

```
➜  ~ docker pull ghcr.io/kellerza/hass-addon-sunsynk-multi/aarch64:0.8.1
Error response from daemon: manifest unknown
```

(compare to the amd64 version)

```
docker pull ghcr.io/kellerza/hass-addon-sunsynk-multi/amd64:0.8.1
0.8.1: Pulling from kellerza/hass-addon-sunsynk-multi/amd64
cb8611c9fe51: Downloading [===========>                                       ]    932kB/3.993MB
```

The [config.yaml](https://github.com/kellerza/sunsynk/blob/main/hass-addon-sunsynk-multi/config.yaml#L13) does have `aarch64` listed, so I'm not clear why the github action didn't build it.

kellerza, 2025-01-10:
This should be the platform independent one:

`ghcr.io/kellerza/hass-addon-sunsynk-multi:0.8.1`

kellerza, 2025-01-10:
In the recent commit I removed `/{arch}` from config.yaml for the multi version

This was already the case in the edge version

maslyankov, 2025-01-10:
> Yeah, so I'm using the multi version
>
>
>
> ```
>
> ➜  ~ docker pull ghcr.io/kellerza/hass-addon-sunsynk-multi/aarch64:0.8.1
>
> Error response from daemon: manifest unknown
>
> ```
>
>
>
> (compare to the amd64 [...]

Aarch64 and arm64 are both 64-bit arm builds, literally mean the same thing. You can use the multi platform build or the linux/arm64 build which is basically just an alias of the multi platform one.

patch0, 2025-01-10:
Yeah I think I've missed something here!  (Just realised that `aarch64` and `arm64` are the same..)

I'm using Home Assistant OS, and I don't *think* I manually set which architecture of the addon.  I'll dig around.

patch0, 2025-01-10:
OK, so.. which config.yaml should I be looking at?  Or how do I install the arch-independent version?

kellerza, 2025-01-10:
It's already fixed in the repo, but HA OS won't reload config.yaml until such time as the version changes (which needs a new release)

Or you can uninstall/install

patch0, 2025-01-10:
ok, I'll uninstall/reinstall.  Easy enough!

kellerza, 2025-01-10:
Uninstall, update the store, wait some, reinstall

kellerza, 2025-01-10:
I just released 0.8.2 with this fix, hopefully all the other changes are ok :-/

closed 2025-01-10 by kellerza

patch0, 2025-01-10:
Yes, that's worked.  It didn't work the first time, but then I updated the store and it was all good.

Thanks for your work on this.  I was hoping to contribute at some point!

commit 98f6ae20 (2025-01-10): fix #388 #387 add protocol version