kellerza/sunsynk #390: Deye/Sunsync 3 phase check "Total Active Power from Side to Side of the Grid"

proggaras, 2025-01-15:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/). You can test your own definitions using [custom sensors](https://kellerza.github.io/sunsynk/reference/mysensors)

### Your environment

- Home Assistant version: 2025.1.0
- Addon:
  - Name: edge/dev
  - Version: d950663
- Inverter:
  - Make: Deye
  - Model: SUN-25K-SG01HP3-EU-AM2
  - Firmware: HMI: Ver 2001-C030 MAIN: Ver 3002-1066-1E08
- Adaptor details:
  - RS485
  - Waveshare RS485_TO_POE_ETH

### Describe the issue/bug and what you expect

With this [commit](https://github.com/kellerza/sunsynk/commit/e6a5c0b6bb343ec01d69be11eac33a62fa75f117) the value 607, "Total Active Power from Side to Side of the Grid" was changed from unsigned to signed. 

According to the [Documentation](https://forum.iobroker.net/assets/uploads/files/1712644425833-c3ca14a0-178c-4f68-be8c-fa68a414b97b-deye-sun-inverter-modbus-en.pdf) 607 has no S16 flag in the last column.
Instead 604, 605, 606 has the S16 flag. 

Is that correct? Please let me know if I should create a pull request. 

[image]

maslyankov, 2025-01-15:
Probably you are looking at an old version of the documentation. Here you can see (the most up to date one I could find) that this register also has S16 notation: https://github.com/maslyankov/sunsynk/blob/main/inverter-docs/deye/three-phase/MODBUS%20RTU%E4%B8%89%E7%9B%B8%E5%82%A8%E8%83%BD%E9%80%9A%E4%BF%A1%E8%A7%84%E7%BA%A6V105.3-20240710-Lee_Dobson.docx

kellerza, 2025-01-15:
If the individual phases are signed, it makes sense that the total is signed.  PR welcome to update 604-606 to signed.

The names seem to be a direct translation, but I don't have a suggestion to improve them at the moment. (If you do change them the old/new need to be added to the deprecated dict as well)

maslyankov, 2025-01-15:
TBH, the registers in the project regarding Grid and Load are partially wrong. I will do a PR updating them with the missing high word registers, as currently they overflow over 32767.

kellerza, 2025-01-15:
Which inverters do you have? @maslyankov  / @proggaras

maslyankov, 2025-01-15:
> Which inverters do you have? @maslyankov / @proggaras

I am testing the things here with SUN-30K-SG01HP3-1 and SUN-12K-SG04LP3-EU (on different sites).

proggaras, 2025-01-15:
@kellerza I have a Deye SUN-25K-SG01HP3-EU-AM2.
@maslyankov Thx for the necessary PR. As soon as its merged I can provide feedback.

proggaras, 2025-01-15:
Now I installed it. All sensors that I'm using seem to work. Except "Inverter Power"
@maslyankov I think it's caused by the name that's used twice (case-sensitive is not working). 

src/sunsynk/definitions/three_phase_common.py
`Sensor((636, 694), "Inverter power", WATT, -1), `
and 
`SwitchRWSensor(80, "Inverter Power", on=1),  # 0=off, 1=on`

maslyankov, 2025-01-15:
> Now I installed it. All sensors that I'm using seem to work. Except "Inverter Power"
>
> @maslyankov I think it's caused by the name that's used twice (case-sensitive is not working). [...]

Basically I have tested it on my install and had no issues. The name should not be an issue as they are different domains.

proggaras, 2025-01-15:
> Basically I have tested it on my install and had no issues. The name should not be an issue as they are different domains.

If you check my config and logs below. You will see just there is just lower case. Also in Home Assistant I will just find the switch: 
[image]

### You configuration

```yaml
DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "xxxxxxx"
    HA_PREFIX: ge2so1
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: tcp://192.168.159.53:8899
SENSOR_DEFINITIONS: three-phase-hv
SENSORS:
  - energy_management
  - power_flow_card
  - grid_ct_power
  - Inverter power
  - Inverter Power
  - ...
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Deye
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 15
    REPORT_EVERY: 60
    CHANGE_BY: 80
  - KEY: RW
    READ_EVERY: 15
    REPORT_EVERY: 60
    CHANGE_ANY: true
  - KEY: ENUM
    READ_EVERY: 15
    REPORT_EVERY: 60
    CHANGE_ANY: true
  - KEY: any_unit
    READ_EVERY: 30
    REPORT_EVERY: 60
    CHANGE_BY: 80
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: 15
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqtt
MQTT_PASSWORD: xxxxx

```

### Logs

``` 
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  15 |             pv1_power, inverter_power, priority_load, grid_l3_power,             |
|     |    battery_1_bms_software_version, grid_ct_l3_power, serial, grid_connected,     |
|     |     grid_phase_warning, battery_1_bms_alarm_flag, pv2_power, load_l3_power,      |
|     |    pv_power, load_l1_power, total_active_power_from_side_to_side_of_the_grid,    |
|     |           grid_power, device_type, grid_ct_l2_power, grid_ct_l1_power,           |
|     |    parallel_communication_quality_warning, battery_1_bms_type, grid_ct_power,    |
|     |          fan_warning, inverter_l1_power, grid_l2_power, overall_state,           |
|     |                   c_phase_power_on_the_inner_side_of_the_grid,                   |
|     |               grid_side___inside_total_apparent_power, use_timer,                |
|     |    lithium_battery_loss_warning, b_phase_power_on_the_inner_side_of_the_grid,    |
|     |      grid_l1_power, inverter_l2_power, load_power, rated_power, pv3_power,       |
[... 33 lines cut ...]
|     |    inverter_l1_power, load_l2_voltage, battery_1_bms_charging_current_limit,     |
|     |    load_l3_voltage, use_timer, inverter_frequency, pv3_power, priority_load,     |
|     |       grid_l3_power, battery_1_bms_discharging_voltage, grid_ct_l3_power,        |
|     |      inverter_l2_voltage, total_active_power_from_side_to_side_of_the_grid,      |
|     |         battery_1_bms_max_discharge_current_limit, inverter_l3_current,          |
|     |        grid_ct_l1_power, inverter_l2_current, pv3_current, grid_l2_power,        |
|     |     grid_l2_voltage, pv2_voltage, pv4_voltage, lithium_battery_loss_warning,     |
|     |     load_power, inverter_l2_power, pv4_power, battery_1_power, pv3_voltage,      |
|     |     a_phase_power_on_the_inner_side_of_the_grid, pv1_power, grid_frequency,      |
|     |      battery_1_voltage, radiator_temperature, grid_l1_voltage, grid_power,       |
|     |          device_type, grid_ct_l2_power, battery_1_soc, load_l1_voltage,          |
|     |                  battery_1_bms_soh, battery_1_bms_temperature,                   |
|     |        c_phase_power_on_the_inner_side_of_the_grid, inverter_l1_current,         |
|     |     battery_1_bms_charging_voltage, grid_side___inside_total_apparent_power,     |
|     | battery_1_current, grid_l3_voltage, inverter_l1_voltage, battery_1_bms_current,  |
|     |                                  load_l2_power                                   |
| 300 |  day_active_energy, battery_1_bms_software_version, day_battery_charge, serial,  |
|     |  grid_connected, total_grid_import, battery_1_bms_alarm_flag, day_grid_import,   |
|     |       day_pv_energy, day_grid_export, battery_1_bms_type, day_load_energy,       |
|     |      overall_state, total_grid_export, total_pv_energy, total_load_energy,       |
|     | total_battery_discharge, battery_1_bms_fault_flag, day_battery_discharge, fault, |
|     |                               total_battery_charge                               |
+-----+----------------------------------------------------------------------------------+

```

kellerza, 2025-01-15:
If the slug of the names match, the last definitions will overwrite the first. Slug = all lowercase and spaces replaced by underscores

This is the same mechanism mysensors.py uses to overwrite some of the standard sensors

kellerza, 2025-01-15:
So line 202, the SwitchRWSensor, should overwrite line 35, the Sensor

proggaras, 2025-01-15:
> So line 202, the SwitchRWSensor, should overwrite line 35, the Sensor

Exactly, I can confirm that. That's what you see in my previous Home Assistant Screenshot.

maslyankov, 2025-01-15:
> > So line 202, the SwitchRWSensor, should overwrite line 35, the Sensor
> 
> 
> 
> Exactly, I can confirm that. That's what you see in my previous Home Assistant Screenshot. 

I see. Maybe we can rename the switch to "Inverter Powered" or "Inverter Enable" (Or Enabled)?

kellerza, 2025-01-16:
I'd go for "Inverter enabled"

proggaras, 2025-01-16:
That's good for me too. 
Thx!

proggaras, 2025-01-16:
PR created https://github.com/kellerza/sunsynk/pull/393

proggaras, 2025-01-16:
Everything works like a charm again :) 
[image]

Thank you guys!

closed 2025-01-16 by proggaras