kellerza/sunsynk #394: AddOn stoped working (DEYE 30kW Hybrid)

xelares, 2025-01-16:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

The AddOn stopped working some days ago an stops with the error in the logs. I have checked all wiring an the RS485 Adapter an they work fine, also rebooted the inverter and checked the settings.

**Expected behavior**
-/-

### Your environment

- Home Assistant version: 2025.1.2
- Addon:
  - Name: sunsynk-multi and edge/dev
  - Version: 0.8.2 and f23619a
- Inverter:
  - Make: Deye
  - Model: SUN-30K-SG01HP3-EU-BM3 (30kW Three Phase Hybrid Inverter)
  - Firmware: Main: 3102-1058-1C08 / HMI: 2001-C019 / Protocol: 0104
- Adaptor details:
  - RS485 Waveshare USB-to-RS485

### You configuration

```yaml

DRIVER: umodbus
INVERTERS:
  - SERIAL_NR: "******4188"
    HA_PREFIX: DEYE
    MODBUS_ID: 1
    PORT: serial:///dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AQ043UZR-if00-port0
SENSOR_DEFINITIONS: three-phase-hv
SENSORS:
  - all
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: DEYE
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 15
    REPORT_EVERY: 60
    CHANGE_BY: 10
  - KEY: RW
    READ_EVERY: 15
    REPORT_EVERY: 60
    CHANGE_ANY: true
  - KEY: any_unit
    READ_EVERY: 30
    REPORT_EVERY: 60
    CHANGE_BY: 80
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: "5"
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqtt
MQTT_PASSWORD: *password*
DEBUG: 2
DEBUG_DEVICE: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AQ043UZR-if00-port0

```

### Logs

```logs

2025-01-16 09:56:22,560 DEBUG   : Rated power changed: Enqueue discovery info updates for prog3_power, prog6_power, export_limit_power, prog4_power, prog5_power, prog1_power, prog2_power
2025-01-16 09:56:22,560 DEBUG   : Prog1 Time changed: Enqueue discovery info updates for prog2_time, prog6_time
2025-01-16 09:56:22,560 DEBUG   : Prog6 Time changed: Enqueue discovery info updates for prog1_time, prog5_time
2025-01-16 09:56:22,560 DEBUG   : Prog2 Time changed: Enqueue discovery info updates for prog1_time, prog3_time
2025-01-16 09:56:22,560 DEBUG   : Prog3 Time changed: Enqueue discovery info updates for prog2_time, prog4_time
2025-01-16 09:56:22,561 DEBUG   : Prog4 Time changed: Enqueue discovery info updates for prog3_time, prog5_time
2025-01-16 09:56:22,561 DEBUG   : Prog5 Time changed: Enqueue discovery info updates for prog6_time, prog4_time
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 55, in read_sensors
    await self.inv.read_sensors(sensors)
  File "/usr/local/lib/python3.12/site-packages/sunsynk/sunsynk.py", line 134, in read_sensors
    raise IOError("; ".join(errs))
OSError: AttributeError reading 10 registers from 108: 'Serial' object has no attribute 'host'
2025-01-16 09:56:22,564 DEBUG   asyncio Close <_UnixSelectorEventLoop running=False closed=False debug=True>
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/__main__.py", line 76, in <module>
    main()
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/__main__.py", line 72, in main
    asyncio.run(main_loop())
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 686, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/__main__.py", line 38, in main_loop
    await ist.connect()
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 146, in connect
    await self.read_sensors(sensors=sensors)
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 55, in read_sensors
    await self.inv.read_sensors(sensors)
  File "/usr/local/lib/python3.12/site-packages/sunsynk/sunsynk.py", line 134, in read_sensors
    raise IOError("; ".join(errs))
OSError: AttributeError reading 10 registers from 108: 'Serial' object has no attribute 'host'
[09:56:22] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner successfully stopped

```

kellerza, 2025-01-16:
The error seems to come directly from the driver (umodbus). I doubt there was a change on umodbus (it hasn't been update in a long time) 

Please use pymodbus as the driver

Ideally you should not connect directly to a serial port, but rather through mbusd, in the long run that should be a lot more stable. You can use the mbusb addon in the same repository

lukeansell, 2025-01-16:
I also seem to be having the same issue. I'm using pymodbus and a serial gateway. It's been a problem the last few days. I think since I've updated. Seems to sometimes work for a bit after a restart. Let me know if I should add my config and logs. 

Thanks for all the great work you do for us.

xelares, 2025-01-16:
> The error seems to come directly from the driver (umodbus). I doubt there was a change on umodbus (it hasn't been update in a long time)
>
> Please use pymodbus as the driver
>
> Ideally you should not [...]

I've changed the driver to pymodbus and port to "" so the addon used the debug device. After a restart the addon is working fine now 👍

kellerza, 2025-01-16:
> I've changed the driver to pymodbus and port to "" so the addon used the debug device. After a restart the addon is working fine now 👍

If you still have issues the next step is going via mbusd

kellerza, 2025-01-16:
> I also seem to be having the same issue. I'm using pymodbus and a serial gateway. It's been a problem the last few days. I think since I've updated. Seems to sometimes work for a bit after a restart. Let me know if I should add my config and logs.

If you use pymodbus it unlikely to be the same. But pymodbus was updated, so best to post your own logs&cfg

kellerza, 2025-01-20:
@xelares once you move to pymodbus, you might see #395

closed 2025-01-20 by kellerza