kellerza/sunsynk #396: Only the information doesn't belong in battery_charge and for me it's a useless value for the user

JanKraslice, 2025-01-17:
Hello this does not belong to the battery
These parameters:
grid_standard
configured_grid_frequency
configured_grid_phases
track_grid_phase

[image]

Set by the installer when the inverter is first connected to the grid ( distribution grid) 
For example, grid_standard is set the first time it is set and does not change during use.

[image]

[image]

Just information for the technician who installs the inverters, changing these parameters is really only done once. Thank you

kellerza, 2025-01-20:
You mean these ones?
https://github.com/kellerza/sunsynk/blob/main/src/ha_addon_sunsynk_multi/sensor_options.py#L245-L259

date_time should not be in here either, as its very spammy

closed 2025-01-20 by kellerza

JanKraslice, 2025-01-20:
> Rozdělte některé z nich do pokročilé skupiny [a2de710](https://github.com/kellerza/sunsynk/commit/a2de71059a3afaf2de0f144895fc3a344c5c5eee)

These values are set the first time 

[image]