kellerza/sunsynk #398: Power values jump to 65535 kW and -65509 kW

Doesi, 2025-01-18:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

Since the last commit, power sensors randomly jump to 65535 kW and -65509 kW. 

**Expected behavior**
Real values should be shown

### Your environment

- Home Assistant version: 2025.1.2
- Addon:
  - Name: sunsynk-multi
  - Version: f23619a
- Inverter:
  - Make: Deye
  - Model: SUN 20K-SG01 HP3-EU-AM2
  - Firmware: -
- Adaptor details:
  - solarman

See also [https://github.com/kellerza/sunsynk/pull/391#issuecomment-2599665488](url)

maslyankov, 2025-01-18:
Related comment (I see it above, but url does not link there): [https://github.com/kellerza/sunsynk/pull/391#issuecomment-2599665488](https://github.com/kellerza/sunsynk/pull/391#issuecomment-2599665488)

JanKraslice, 2025-01-18:
Hello yes that's right version:f23619a is wrong
I'm attaching pictures, please kellerza

[image]

[image]

[image]

Marow, 2025-01-22:
I have exactly the same problem with DEYE SG05LP3 20K (the new one).

I am using Sunsynk/Deye Inverter Add-on (edge/dev) with Version: e6c5d80

kellerza, 2025-01-22:
Can you test with the new edge version that includes #397 ?

proggaras, 2025-01-22:
Unfortunately I can't update: 

[image]

maslyankov, 2025-01-22:
> Unfortunately I can't update:
> 
> <img alt="Image" width="1011" src="https://private-user-images.githubusercontent.com/9094312/405606109-34cc10e1-4aa8-4cba-81e1-49b682d45cab.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzc1NDc1NzYsIm5iZiI6MTczNzU0NzI3NiwicGF0aCI6Ii85MDk0MzEyLzQwNTYwNjEwOS0zNGNjMTBlMS00YWE4LTRjYmEtODFlMS00OWI2ODJkNDVjYWIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMjJUMTIwMTE2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjNjY2VkZGU3MjFjZTBlYTM3Y2M2YTMwMjczNzg1MTM3Yjc4OTUxMDk1NmQzOGYyMDQ3YWFkZDc1M2Q0YmNjNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.2WMKs0u3MDqV9R0r-4iYQNT7CzpxeV93NJTdNzI0QIE">

There seem to be some build cicd issues occasionally which I've noticed. They have affected the new version build. I have made a PR that aims to fix this.

Marow, 2025-01-22:
Looks like there no peaks anymore with `02f085f`

maslyankov, 2025-01-22:
> Looks like there no peaks anymore with `02f085f`
> 
> I will check it again after few hours and will send information here. But first, looks good!

Let's keep this issue open for now. Pls confirm in a day or two if fix is consistent.

Marow, 2025-01-22:
I found that:
https://github.com/StephanJoubert/home_assistant_solarman/issues/329

Looks nearly the same problem on Solarman. Probably you can check how they fixed it.

Its not fixed. Thats last 8 mintes

[image]

maslyankov, 2025-01-22:
Yes, I just checked over at https://github.com/davidrapan/ha-solarman/blob/c1e893c1c79e2b969104bf478deca8f2e286f504/custom_components/solarman/inverter_definitions/deye_p3.yaml#L2461-L2485 that they have the same registers are defined here: https://github.com/kellerza/sunsynk/blob/bbb7a600960076fd032baf858edf767e7422d470/src/sunsynk/definitions/three_phase_common.py#L78-L80

Marow, 2025-01-22:
> Yes, I just checked over at https://github.com/davidrapan/ha-solarman/blob/c1e893c1c79e2b969104bf478deca8f2e286f504/custom_components/solarman/inverter_definitions/deye_p3.yaml#L2461-L2485 that they have the same registers are defined here:
> 
> [sunsynk/src/sunsynk/definitions/three_phase_common.py](https://github.com/kellerza/sunsynk/blob/bbb7a600960076fd032baf858edf767e7422d470/src/sunsynk/definitions/three_phase_common.py#L78-L80)
> 
> Lines 78 to 80 in [bbb7a60](/kellerza/sunsynk/commit/bbb7a600960076fd032baf858edf767e7422d470)
> 
>  Sensor((650, 656), "Load L1 power", WATT, -1), 
>  Sensor((651, 657), "Load L2 power", WATT, -1), 
>  Sensor((652, 658), "Load L3 power", WATT, -1), 
> This appears to be the fix they implemented: [StephanJoubert/home_assistant_solarman#270](https://github.com/StephanJoubert/home_assistant_solarman/pull/270)

Yeah, i think thats it.
Hope for update soon, than i can try again.

Marow, 2025-01-22:
For your information:

 I've found out something else that's very strange! Since the solar has stopped coming in because the sun has gone, there have been no more peaks. From about 4 p.m. onwards, you can't see any more peaks.

 Is it somehow related to this? 
Maybe too much data being sent together, which is messing things up or something?

[image]

maslyankov, 2025-01-22:
Probably is related, yes. Maybe data gets a bit messed up when many modbus frames come in. I also implemented CRC validation as well, as other projects have.

maslyankov, 2025-01-22:
Version e37dd… is up 🙌 Hopefully this does it. 🤞

Doesi, 2025-01-22:
Now it's completely broken. In the logs I get continously:

`Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number'`

Marow, 2025-01-23:
> Now it's completely broken. In the logs I get continously:
> 
> `Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number'`

Not for me. I didnt have that error.

> Version [e37dd6d](https://github.com/kellerza/sunsynk/commit/e37dd6d0c33e5fdeae0dda28183d7345a3712f01) is up 🙌 Hopefully this does it. 🤞

Will check it later, when Solar Power comes in. Without solar, i didnt have that peaks

[image]

maslyankov, 2025-01-23:
> Not for me. I didnt have that error.
>
>
>
>
> Will check it later, when Solar Power comes in. Without solar, i didnt have that peaks
>
> [image]

Are you using solarman connection? I am guessing not (tcp do modbus?)

Doesi, 2025-01-23:
Yes, solarman with the original wifi dongle.

Marow, 2025-01-23:
> > > Now it's completely broken. In the logs I get continously:
> > > `Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number'`
> > 
> > 
> > Not for me. I didnt have that error.
> > > Version [e37dd6d](https://github.com/kellerza/sunsynk/commit/e37dd6d0c33e5fdeae0dda28183d7345a3712f01) is up 🙌 Hopefully this does it. 🤞
> > 
> > 
> > Will check it later, when Solar Power comes in. Without solar, i didnt have that peaks
> > [image]
> 
> Are you using solarman connection? I am guessing not (tcp to modbus?)

No, i am using cable from modbus with Waveshare RS485 to USB (USB into my Home Assistant mini PC)

martinarva, 2025-01-23:
I'm running the latest and still have issue with power jumps

[image]

maslyankov, 2025-01-23:
Just found that there is a related issue from a long time ago: https://github.com/kellerza/sunsynk/issues/184

Marow, 2025-01-23:
My Solar Power starts now, lets see... right now, on low solar power there are no peaks with newest version. Hope that also doesnt comes with more power soon

maslyankov, 2025-01-23:
> My Solar Power starts now, lets see... right now, on low solar power there are no peaks with newest version. Hope that also doesnt comes with more power soon

I am skeptical. https://github.com/kellerza/sunsynk/pull/408 I think we need to revert this PR, as it only affects solarman (pysolarmanv5) driver usage (and overlaps with validations inside it).

proggaras, 2025-01-23:
Mine is working just some spikes are there from time to time...

[image]

### Your environment

- Home Assistant version: 2025.1.3
- Addon:
  - Name: edge/dev
  - Version: e37dd6d
- Inverter:
  - Make: Deye
  - Model: SUN-25K-SG01HP3-EU-AM2
  - Firmware: HMI: Ver 2001-C030 MAIN: Ver 3002-1066-1E08
- Adaptor details:
  - RS485
  - Waveshare RS485_TO_POE_ETH

maslyankov, 2025-01-23:
I will do another PR with revert of solarman breaking changes, also will add more validation to reading of data and more debug logs, so it it continues to happen we can hopefully see why.

Marow, 2025-01-23:
Peaks starting again :(

[image]

Saurman, 2025-01-23:
yes, I get 65k spikes for CT entities

spinza, 2025-01-23:
Running add on version e37dd6d (edge) with recent bugfix.
Sunsynk 3 Phase Inverter
Connected via pymodbus to mbusd instance running on a separate Raspberry Pi.

Still get glitchy values after the latest fix.

[image]

Marow, 2025-01-23:
For your info: I started 16000 Watt charging my car with a wallbox.

There are no peaks!
Peaks are really only coming, when solar power is coming

When peaks are stopping, solar was at the end of power

[image]

JanKraslice, 2025-01-23:
I've got them in and out of solar.

See picture this morning I have 65000W even when it's still dark outside 

[image]

Marow, 2025-01-27:
For me i can see, that is only when Solar is running.
But i didnt think that solar is the problem, i think its a problem, that i have a lot of data in that time than.

TheBazeman, 2025-02-03:
Hi, I also have the same problem when updating to the last version c838b5d. Rolled back to version 92dd837 and I dont have the spikes at all.

[image]

TheBazeman, 2025-02-03:
Hi, Just using the backups page in Home Assistant. You can restore per addon if you want:
***@***.***
KR, Bas

afbeelding.png (view on web)<https://github.com/user-attachments/assets/ea7d4f3d-08b7-4f62-a795-042029cec4c7>

Can you tell me, how to rollback? I will try also

Marow, 2025-02-03:
Looks like the [92dd837](https://github.com/kellerza/sunsynk/commit/92dd837cb779b04b351363a00e354e64714c1de3) works well!

So something wrong at the other version after that than?

spinza, 2025-02-03:
I'm trying to figure out what change led to the the power values jumping.  The new version with single registers cause issues, but for Grid Power (not load power) on Sunsynk 3-phase 12kw.  But there are other changes (including how the the reg_to_value functions work).

kellerza, 2025-02-03:
Grid power and Load power now have 2x registers each
https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions/three_phase_common.py#L50

I doubt it's reg_to_value (as that should be covered by unit tests) and that is why I suggest trying the old (single register) definitions

JanKraslice, 2025-02-03:
I observe it on:
sensor.ss_grid_ct_power
sensor.ss_grid_ct_l1_power
sensor.ss_grid_ct_l2_power
sensor.ss_grid_ct_l3_power
3 phase DEYE 30Kw high voltage inverter exact designation is SUN-30K-SG01HP3-EU-BM3

During today there was no power draw from the grid all was covered by generation or batteries.

[image]

spinza, 2025-02-04:
> I doubt it's reg_to_value (as that should be covered by unit tests) and that is why I suggest trying the old (single register) definitions

Sorry of course.  I compared the two commits but got them mixed up in the comparison.  I compared from new to old which made me think the single value is the newer ones.  

I will run single values as soon as I figure out how to run custom code.

The error happens more if there is little or now power draw from the grid.

Marow, 2025-02-04:
I have commit f044dd2 running on my home assistant over 24 hours now and no peaks. There must be a commit after that, which brings the problems

JanKraslice, 2025-02-09:
Hello and please.
Is there really no other solution but to go back to 92dd837 ? 
In the future we will never use updates again ? 
I have version 174e080 since its release on 3.2.2025 and the 65XXX spikes are still every day.
There used to be patching and new releases quite often now nothing :-( 
I'm just a mere user apart from the test and report I can't help with the code so sorry.
Thank you

kellerza, 2025-02-10:
Did you try the custom sensors mentioned above?

JanKraslice, 2025-02-10:
I have tried 2 times before to create a custom sensor and never succeeded.
I'm seeing the aforementioned 65xxx on the default 
sensor.ss_grid_l1_power
sensor.ss_grid_l2_power
sensor.ss_grid_l3_power
sensor.ss_grid_power
a
sensor.ss_grid_ct_l1_power
sensor.ss_grid_ct_l2_power
sensor.ss_grid_ct_l3_power
sensor.ss_grid_ct_power

proggaras, 2025-02-10:
For now I added the following definitions to: /share/hass-addon-sunsync/mysensors.py
The changes from: https://github.com/kellerza/sunsynk/commit/e3af4300b33296457f5fbddee99f1509cd031365 seem to be reverted. 

```
SENSORS += (
    Sensor(616, "Grid CT L1 power", WATT, -1),  # aPower
    Sensor(617, "Grid CT L2 power", WATT, -1),  # bPower
    Sensor(618, "Grid CT L3 power", WATT, -1),  # cPower
    Sensor(619, "Grid CT power", WATT, -1),  # totalPower

    Sensor(625, "Grid power", WATT, -1),  # gridTotalPac
    Sensor(622, "Grid L1 power", WATT, -1),  # aPower
    Sensor(623, "Grid L2 power", WATT, -1),  # bPower
    Sensor(624, "Grid L3 power", WATT, -1),  # cPower

    Sensor(636, "Inverter power", WATT, -1),
    Sensor(633, "Inverter L1 power", WATT, -1),
    Sensor(634, "Inverter L2 power", WATT, -1),
    Sensor(635, "Inverter L3 power", WATT, -1),

    Sensor(653, "Load power", WATT, -1),
    Sensor(650, "Load L1 power", WATT, -1),
    Sensor(651, "Load L2 power", WATT, -1),
    Sensor(652, "Load L3 power", WATT, -1),

    Sensor(604, "A Phase Power on the Inner Side of the Grid", "W"),
    Sensor(605, "B Phase Power on the Inner Side of the Grid", "W"),
    Sensor(606, "C Phase Power on the Inner Side of the Grid", "W"),
    Sensor(607, "Total Active Power from Side to Side of the Grid", "W", -1),
    Sensor(608, "Grid Side - Inside Total Apparent Power", "W"),

    Sensor(636, "Inverter power", WATT, -1),
    Sensor(633, "Inverter L1 power", WATT, -1),
    Sensor(634, "Inverter L2 power", WATT, -1),
    Sensor(635, "Inverter L3 power", WATT, -1),
)
```

JanKraslice, 2025-02-10:
> For now I added the following definitions to: /share/hass-addon-sunsync/mysensors.py The changes from: [e3af430](https://github.com/kellerza/sunsynk/commit/e3af4300b33296457f5fbddee99f1509cd031365) seem to be reverted.
> 
> ```
> SENSORS += (
>     Sensor(616, "Grid CT L1 power", WATT, -1),  # aPower
>     Sensor(617, "Grid CT L2 power", WATT, -1),  # bPower
>     Sensor(618, "Grid CT L3 power", WATT, -1),  # cPower
>     Sensor(619, "Grid CT power", WATT, -1),  # totalPower
> 
>     Sensor(625, "Grid power", WATT, -1),  # gridTotalPac
>     Sensor(622, "Grid L1 power", WATT, -1),  # aPower
>     Sensor(623, "Grid L2 power", WATT, -1),  # bPower
>     Sensor(624, "Grid L3 power", WATT, -1),  # cPower
> 
>     Sensor(636, "Inverter power", WATT, -1),
>     Sensor(633, "Inverter L1 power", WATT, -1),
>     Sensor(634, "Inverter L2 power", WATT, -1),
>     Sensor(635, "Inverter L3 power", WATT, -1),
> 
>     Sensor(653, "Load power", WATT, -1),
>     Sensor(650, "Load L1 power", WATT, -1),
>     Sensor(651, "Load L2 power", WATT, -1),
>     Sensor(652, "Load L3 power", WATT, -1),
> 
>     Sensor(636, "Inverter power", WATT, -1),
>     Sensor(633, "Inverter L1 power", WATT, -1),
>     Sensor(634, "Inverter L2 power", WATT, -1),
>     Sensor(635, "Inverter L3 power", WATT, -1),
> )
> ```

If you use 6xx sensors like that, don't you already have 65xxx,-kW peaks?

JanKraslice, 2025-02-11:
Sorry, I couldn't add my own sensors.
Can I now use the sensors a,b,c you mentioned?

proggaras, 2025-02-11:
> Sorry, I couldn't add my own sensors. Can I now use the sensors a,b,c you mentioned?

Did you add mysensors in the addon configuration? 

```
...
SENSORS:
   ...
   - mysensors
   ...
```

schein-heilig, 2025-02-13:
I have the same issues with the Battery Charge (sensor.deye_day_battery_charge) .

JanKraslice, 2025-02-15:
> Zkoušel jsi výše zmíněné vlastní senzory?

I confirm both the 92dd837 version and the custom sensors solve the 65xxx peak problem

Will there be a permanent solution in the next release? Can we try it please.

kellerza, 2025-02-16:
> ```
> 11:53:55 WARNING High value rated_power=120000 (#20=0xd4c0, #21=0x1)
> ```
> 
> EDIT: would it be possible to turn off this logging, as this probably chrashed my sunsynk this night.

If this crashed your addon, you probably need to update your HA OS system :wink:

Anyway, I've removed the logging and implemented a possible fix based on a new Sensor16 class 
<https://github.com/kellerza/sunsynk/commit/5ae3b7665641a2e5a7713fea23d61145e0beea1f#diff-975965ed99ea476ca9badb6a59ccdc1867f1570d8d136ae22458627a812bc456R64>

Please try it without the custom sensors and let me know.

nicopret1, 2025-02-16:
Tried https://github.com/kellerza/sunsynk/commit/5ae3b7665641a2e5a7713fea23d61145e0beea1f but unfortunately the spikes are back when I remove custom sensors.

kellerza, 2025-02-16:
Did you see anything in your log?
What inverter do you have?

nicopret1, 2025-02-16:
I see regular entries like these:
18:26:30 INFO    Sensor Grid L2 power reported a 32-bit value: ['0xfffe', '0xffff']
18:26:31 INFO    Sensor Grid L2 power reported a 32-bit value: ['0xfff6', '0xffff']
18:26:35 INFO    Sensor Grid L2 power reported a 32-bit value: ['0xfff9', '0xffff']
18:26:35 INFO    Sensor Grid L2 power reported a 32-bit value: ['0xfff9', '0xffff']
18:26:40 INFO    Sensor Grid L2 power reported a 32-bit value: ['0xfff6', '0xffff']
18:26:40 INFO    Sensor Grid power reported a 32-bit value: ['0xffff', '0xffff']
18:26:40 INFO    Sensor Grid L2 power reported a 32-bit value: ['0xfffa', '0xffff']
18:26:45 INFO    Sensor Grid L2 power reported a 32-bit value: ['0xfffc', '0xffff']
18:26:45 INFO    Sensor Grid L2 power reported a 32-bit value: ['0xfffa', '0xffff']
18:26:55 INFO    Sensor Grid L2 power reported a 32-bit value: ['0xfffb', '0xffff']
18:26:55 INFO    Sensor Grid L2 power reported a 32-bit value: ['0xfffe', '0xffff']
18:27:00 INFO    Sensor Grid L2 power reported a 32-bit value: ['0xffff', '0xffff']

I experience the problem on the following inverters:
- Three phase HV Hybrid 50kW
- Three phase LV Hybrid 12kW

kellerza, 2025-02-17:
Hi @nicopret1 - the latest version should log the first 50 readings with more detail (including rated_power). Please post the log for both inverters.

With a 1-register sensor (old behavior) - any value between 32768-65535 would have been reported as a (small) negative value - so it might be that this happened on your 50kW inverter, but you never noticed it.

For an inverter<32k, a 1-register sensor should be enough - maybe the solution is just to use 1 register when rated_power is low.

With a 2-register sensor you might get:
- a big positive spike if the 2nd register is zero
- a big negative spike if the first register is 0

nicopret1, 2025-02-17:
Hi @kellerza thank you for the detailed information. What I've experienced, now makes more sense. I've attached the logs files for both installations: 

- The 2x12kW installation uses very little grid power and that was the site where I first noticed the spikes.
- The 4x50kW inverter installation was not that obvious initially. I've also included some log history for this install. 

[4x50kW-4bb530c7_hass-addon-sunsynk-edge_2025-02-17T06-02-39.364Z.log](https://github.com/user-attachments/files/18820732/4x50kW-4bb530c7_hass-addon-sunsynk-edge_2025-02-17T06-02-39.364Z.log)
[2x12kW-4bb530c7_hass-addon-sunsynk-edge_2025-02-17T05-52-31.308Z.log](https://github.com/user-attachments/files/18820731/2x12kW-4bb530c7_hass-addon-sunsynk-edge_2025-02-17T05-52-31.308Z.log)

martinarva, 2025-02-17:
T

> > ```
> > 11:53:55 WARNING High value rated_power=120000 (#20=0xd4c0, #21=0x1)
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > EDIT: would it be possible to turn off this logging, as this probably chrashed my sunsynk this night.
> 
> If this crashed your addon, you probably need to update your HA OS system 😉
> 
> Anyway, I've removed the logging and implemented a possible fix based on a new Sensor16 class [5ae3b76#diff-975965ed99ea476ca9badb6a59ccdc1867f1570d8d136ae22458627a812bc456R64](https://github.com/kellerza/sunsynk/commit/5ae3b7665641a2e5a7713fea23d61145e0beea1f#diff-975965ed99ea476ca9badb6a59ccdc1867f1570d8d136ae22458627a812bc456R64)
> 
> Please try it without the custom sensors and let me know.

The latest version now works and my grid entity is clean. Only CT Grid has errorrs, but currently my CT clamps are not on thle wires as I moved my load to inverter backup.

Here is a log>

[sunsynk-multi.log](https://github.com/user-attachments/files/18821534/sunsynk-multi.log)

JanKraslice, 2025-03-09:
Does that really bother anyone?
It happens over and over and over again.
I don't know what happened here, before when there was a bug in the addon it was fixed in the next version, now everyone doesn't care. That's a shame.
[image]

schein-heilig, 2025-03-10:
same here with Grid spikes

[image]

JanKraslice, 2025-03-10:
> Ahoj[@nicopret1](https://github.com/nicopret1)- nejnovější verze by měla zaznamenávat prvních 50 odečtů s podrobnějšími údaji (včetně jmenovitého výkonu). Zašlete prosím protokol pro oba střídače.
> 
> U senzoru s 1 registrem (staré chování) - jakákoli hodnota mezi 32768-65535 by byla hlášena jako (malá) záporná hodnota - takže se to mohlo stát na vašem 50kW měniči, ale nikdy jste si toho nevšimli.
> 
> Pro invertor<32k by mělo stačit čidlo s 1 registrem - možná je řešením jen použít 1 registr, když je rating_power nízký.
> 
> S 2registrovým senzorem můžete získat:
> 
> * velký kladný hrot, pokud je 2. registr nulový
> * velký záporný vrchol, pokud je první registr 0

Please is there really no solution to use only one register as before? 
Currently there are still spikes on GRID CT L1, CT L2, CT L3

kellerza, 2025-03-10:
Upgrade your firmware or use a custom definition with the single register for grid and battery

JanKraslice, 2025-03-10:
> Pochybuji, že se jedná o chybu.
> 
> Zdá se, že různé měniče se chovají odlišně
> 
> Používáme nejnovější dostupné definice modbus. Přečtěte si mé vysvětlení výše. Nejsem si jistý, zda to dokážeme zlepšit a stále podporovat 2 registry.
> 
> Nejlepší by bylo použít vlastní definici a pokud to někdo s těmito měniči chce zdokumentovat, ještě lépe

I totally agree I would stick with the original definition of version 92dd837 where the default is read only one register.
Simply as it was before.

If some versions of inverters need to read both registers, let the option be via mysensor. 
Personally I think most don't need to read the value of both registers I don't see any advantage in that.

Do you agree?

[9 more replies]

closed 2025-04-05 by kellerza