kellerza/sunsynk #409: Doesn't read any data anymore (PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument)

Saurman, 2025-01-23:
### Describe the issue/bug

Something has happened overnight (about 8-12h ago) and add-on is no longer working, it cannot read any data from dongle. Tried rebooting everything, same issue.

### Your environment

- Home Assistant version: 2025.1.3
- Addon:
  - Name: sunsynk-multi
  - Version: 2025-01-11
- Inverter:
  - Make: Deye
  - Model: SUN20K
  - Firmware: LSW3_32U_5406_1.06
- Adaptor details:
  - solarman

### You configuration

Unchanged, worked for at least 9 months

### Logs

```
[06:26:18] INFO: Service restart after closing
06:26:20 INFO    Loading config: /data/options.json
2025-01-23 06:26:20,374 INFO    __main__ sunsynk library version: 0.8.2
2025-01-23 06:26:20,438 DEBUG   : solarman driver options: kwargs:{'port': 'tcp://192.168.131.3:8899', 'server_id': 1, 'timeout': 10, 'read_sensors_batch_size': 20, 'allow_gap': 2, 'dongle_serial_number': 2789767211} - inv:InverterOptions(port='tcp://192.168.131.3:8899', modbus_id=1, ha_prefix='Deye', serial_nr='XXXXXXXX', dongle_serial_number=XXXXXXXXX)
2025-01-23 06:26:20,439 DEBUG   sunsynk.solarmansunsynk DBG: check_serial: Attribute(name='dongle_serial_number', default=NOTHING, validator=<function SolarmanSunsynk.check_serial at 0x7f8e641bc0>, repr=True, eq=True, eq_key=None, order=True, order_key=None, hash=None, init=True, metadata=mappingproxy({}), type=<class 'int'>, converter=None, kw_only=True, inherited=False, on_setattr=None, alias='dongle_serial_number') XXXXXXXX
2025-01-23 06:26:20,465 INFO    ha_addon_sunsynk_multi.timer_schedule Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|    enum   |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  12  |   60   |    80.0   |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
2025-01-23 06:26:20,466 INFO    ha_addon_sunsynk_multi.sensor_options Using three phase HV sensor definitions.
2025-01-23 06:26:20,467 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule enum used for Device type (reason: EnumSensor)
2025-01-23 06:26:20,468 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule no_unit used for Protocol (reason: No unit)
2025-01-23 06:26:20,468 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule no_unit used for Serial (reason: No unit)
2025-01-23 06:26:20,468 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule w used for PV2 power (reason: Unit)
2025-01-23 06:26:20,469 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule any_unit used for PV2 voltage (reason: Any unit)
2025-01-23 06:26:20,469 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule any_unit used for PV2 current (reason: Any unit)
2025-01-23 06:26:20,469 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule any_unit used for Battery 1 current (reason: Any unit)
2025-01-23 06:26:20,469 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule w used for Battery 1 power (reason: Unit)
2025-01-23 06:26:20,469 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule any_unit used for Battery 1 SOC (reason: Any unit)
2025-01-23 06:26:20,470 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule any_unit used for Battery 1 voltage (reason: Any unit)
2025-01-23 06:26:20,470 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule rw used for Priority Load (reason: RWSensor)
2025-01-23 06:26:20,470 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Total Battery Charge (reason: Unit)
2025-01-23 06:26:20,471 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Total Battery Discharge (reason: Unit)
2025-01-23 06:26:20,471 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Total Grid Export (reason: Unit)
2025-01-23 06:26:20,471 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Total Grid Import (reason: Unit)
2025-01-23 06:26:20,471 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Total PV Energy (reason: Unit)
2025-01-23 06:26:20,472 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Day Battery Charge (reason: Unit)
2025-01-23 06:26:20,472 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Day Battery discharge (reason: Unit)
2025-01-23 06:26:20,472 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Day Grid Export (reason: Unit)
2025-01-23 06:26:20,472 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Day Grid Import (reason: Unit)
2025-01-23 06:26:20,472 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Day Load Energy (reason: Unit)
2025-01-23 06:26:20,472 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule kwh used for Day PV Energy (reason: Unit)
2025-01-23 06:26:20,473 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule no_unit used for Grid Connected (reason: No unit)
2025-01-23 06:26:20,473 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule w used for Grid CT power (reason: Unit)
2025-01-23 06:26:20,473 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule any_unit used for Grid frequency (reason: Any unit)
2025-01-23 06:26:20,473 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule w used for Grid L1 power (reason: Unit)
2025-01-23 06:26:20,473 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule w used for Grid L2 power (reason: Unit)
2025-01-23 06:26:20,474 DEBUG   ha_addon_sunsynk_multi.timer_schedule Schedule w used for Grid L3 power (reason: Unit)
[... 73 lines cut ...]
2025-01-23 06:26:55,815 DEBUG   asyncio <asyncio.TransportSocket fd=8, family=2, type=1, proto=6, laddr=('172.30.33.2', 47454), raddr=('192.168.131.3', 8899)> connected to 192.168.131.3:8899: (<_SelectorSocketTransport fd=8 read=polling write=<idle, bufsize=0>>, <asyncio.streams.StreamReaderProtocol object at 0x7f8e453ad0>)
2025-01-23 06:26:55,817 ERROR   sunsynk.solarmansunsynk Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number' (retry 2)
2025-01-23 06:26:57,860 DEBUG   asyncio <asyncio.TransportSocket fd=8, family=2, type=1, proto=6, laddr=('172.30.33.2', 47466), raddr=('192.168.131.3', 8899)> connected to 192.168.131.3:8899: (<_SelectorSocketTransport fd=8 read=polling write=<idle, bufsize=0>>, <asyncio.streams.StreamReaderProtocol object at 0x7f8ff9faa0>)
2025-01-23 06:26:57,863 ERROR   sunsynk.solarmansunsynk Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number' (retry 3)
2025-01-23 06:26:59,888 DEBUG   asyncio <asyncio.TransportSocket fd=8, family=2, type=1, proto=6, laddr=('172.30.33.2', 47474), raddr=('192.168.131.3', 8899)> connected to 192.168.131.3:8899: (<_SelectorSocketTransport fd=8 read=polling write=<idle, bufsize=0>>, <asyncio.streams.StreamReaderProtocol object at 0x7f8e453560>)
2025-01-23 06:26:59,891 ERROR   sunsynk.solarmansunsynk Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number' (retry 4)
2025-01-23 06:27:01,910 DEBUG   asyncio <asyncio.TransportSocket fd=8, family=2, type=1, proto=6, laddr=('172.30.33.2', 47478), raddr=('192.168.131.3', 8899)> connected to 192.168.131.3:8899: (<_SelectorSocketTransport fd=8 read=polling write=<idle, bufsize=0>>, <asyncio.streams.StreamReaderProtocol object at 0x7f8e452180>)
2025-01-23 06:27:01,912 ERROR   sunsynk.solarmansunsynk Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number' (retry 5)
2025-01-23 06:27:01,913 ERROR   ha_addon_sunsynk_multi.a_inverter OSError reading 1 registers from 2: Failed to read register 2 Protocol
2025-01-23 06:27:02,057 DEBUG   asyncio <asyncio.TransportSocket fd=8, family=2, type=1, proto=6, laddr=('172.30.33.2', 47480), raddr=('192.168.131.3', 8899)> connected to 192.168.131.3:8899: (<_SelectorSocketTransport fd=8 read=polling write=<idle, bufsize=0>>, <asyncio.streams.StreamReaderProtocol object at 0x7f8e452ae0>)
[... cut ...]
2025-01-23 06:27:10,278 CRITICAL ha_addon_sunsynk_multi.a_inverter Could not read sensors: ['Device type', 'Protocol', 'Serial']
2025-01-23 06:27:10,283 INFO    ha_addon_sunsynk_multi.a_inverter ############################################################
2025-01-23 06:27:10,347 INFO    ha_addon_sunsynk_multi.a_inverter No response on the Modbus interface tcp://192.168.131.3:8899, see https://kellerza.github.io/sunsynk/guide/fault-finding
2025-01-23 06:27:10,348 INFO    ha_addon_sunsynk_multi.a_inverter ############################################################
```

Doesi, 2025-01-23:
Same here
[https://github.com/kellerza/sunsynk/issues/398#issuecomment-2608352362](https://github.com/kellerza/sunsynk/issues/398#issuecomment-2608352362)

Saurman, 2025-01-23:
fixed by updating to `e37dd6d `

closed 2025-01-23 by Saurman

Doesi, 2025-01-23:
It is not fixed with e37dd6d. Instead, the error came with that version.

Saurman, 2025-01-23:
yes, I still get 65k spikes for CT entities

JanKraslice, 2025-01-25:
Hello please I have version f23619a I got a bit lost and I see various error messages in the problems.
Please is the version offered to update f25e68e functional in case of solarman communication via dongle? Or should I stay with the current f23619.
Thank you, I don't mean that in a bad way