kellerza/sunsynk #411: Error

Dvun, 2025-01-23:
[image]

This error get ad not working after today update 

14:45:40 INFO    Connecting to tcp://192.168.31.6:8899
14:45:40 INFO    Reading startup sensors Device type, Protocol, Serial
14:45:40 ERROR   Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number' (retry 1)
14:45:42 ERROR   Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number' (retry 2)
14:45:44 ERROR   Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number' (retry 3)
14:45:46 ERROR   Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number' (retry 4)
14:45:48 ERROR   Error reading: PySolarmanV5Async.read_holding_registers() got an unexpected keyword argument 'sequence_number' (retry 5)
14:45:48 ERROR   OSError reading 8 registers from 0: Failed to read register 0

maslyankov, 2025-01-23:
Pls check for issues, before posting. This is duplicate issue.

Dvun, 2025-01-23:
Give a link please

maslyankov, 2025-01-23:
https://github.com/kellerza/sunsynk/issues/409

closed 2025-02-04 by kellerza