kellerza/sunsynk #432: ha supervisor removes sunsync addon

thunderchild-git, 2025-02-14:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

A clear and concise description of what the bug is.

**Expected behavior**
A clear and concise description of what you expected to happen.
I expected  the addon to work like a charm like it did about 6 months before until  last week.
But after updating homeassistant from 2025-2-1 to 2 the supervisor told me
that the addon has been removed from the repo, which is not true.
When i try to add the repo to the addon-repos again  i get the answer 👍 repo not known and a trashcan ...

### Your environment

- Home Assistant version: 2025.1.x
- Addon:
  - Name: sunsynk-edge/dev
  - Version: 2025-01-11
- Inverter:
  - Make: Deye
  - Model: SUN-3.6K single phase
  - Firmware: 
- Adaptor details:
  - RS485/RS232/solarman
    /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0026-if00-port0

### You configuration

```DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "22xxx"
    HA_PREFIX: SS
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "27xxx"
    PORT: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0MCGET-if00-port0
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Deye
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 30
SCHEDULES:
  - KEY: W
    READ_EVERY: 30
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 60
    CHANGE_PERCENT: 1
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: "15"
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqtt_jb
MQTT_PASSWORD: xxx
DEBUG_DEVICE: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0MCGET-if00-port0
```

### Logs

```
Cmd('git') failed due to: exit code(128)
  cmdline: git clone -v --recursive --depth=1 --shallow-submodules -- https://github.com/kellerza/sunsynk /data/addons/git/4bb530c7
  stderr: 'Cloning into '/data/addons/git/4bb530c7'...
POST git-upload-pack (175 bytes)
POST git-upload-pack (244 bytes)
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 30 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
```

kellerza, 2025-02-15:
I've not seen this before, but the addon does not use git, so this log is probably from the supervisor?

You might try the homeassistant community for more help

closed 2025-02-23 by kellerza

JanKraslice, 2025-03-01:
> ### Problém související s
> Doplněk Sunsynk / mbusd Home Assistant
> 
> ### Popište problém/chybu
> Jasný a stručný popis toho, co je chyba.
> 
> **Očekávané chování** Jasný a stručný popis toho, co jste očekávali, že se stane. Očekával jsem, že doplněk bude fungovat jako kouzlo před 6 měsíci až do minulého týdne. Ale po aktualizaci homeassistanta z 2025-2-1 na 2 mi supervizor řekl, že addon byl odstraněn z repo, což není pravda. Když se znovu pokusím přidat repo do addon-repo, dostanu odpověď 👍 repo neznámé a odpadkový koš ...
> 
> ### Vaše prostředí
> * Verze Home Assistant: 2025.1.x
> * Addon:
>   
>   * Název: sunsynk-edge/dev
>   * Verze: 2025-01-11
> * Invertor:
>   
>   * Značka: Deye
>   * Model: SUN-3.6K jednofázový
>   * Firmware:
> * Podrobnosti o adaptéru:
>   
>   * RS485/RS232/solarman 
>     /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0026-if00-port0
> 
> ### Vaše konfigurace
> ```
> INVERTERS:
>   - SERIAL_NR: "22xxx"
>     HA_PREFIX: SS
>     MODBUS_ID: 1
>     DONGLE_SERIAL_NUMBER: "27xxx"
>     PORT: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0MCGET-if00-port0
> SENSOR_DEFINITIONS: single-phase
> SENSORS:
>   - energy_management
>   - power_flow_card
>   - pv2_power
> SENSORS_FIRST_INVERTER:
>   - settings
> MANUFACTURER: Deye
> READ_ALLOW_GAP: 2
> READ_SENSORS_BATCH_SIZE: 30
> SCHEDULES:
>   - KEY: W
>     READ_EVERY: 30
>     REPORT_EVERY: 60
>     CHANGE_ANY: false
>     CHANGE_BY: 60
>     CHANGE_PERCENT: 1
> NUMBER_ENTITY_MODE: auto
> PROG_TIME_INTERVAL: "15"
> MQTT_HOST: core-mosquitto
> MQTT_PORT: 1883
> MQTT_USERNAME: mqtt_jb
> MQTT_PASSWORD: xxx
> DEBUG_DEVICE: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0MCGET-if00-port0
> ```
> 
> ### Protokoly
> ```
> Cmd('git') failed due to: exit code(128)
>   cmdline: git clone -v --recursive --depth=1 --shallow-submodules -- https://github.com/kellerza/sunsynk /data/addons/git/4bb530c7
>   stderr: 'Cloning into '/data/addons/git/4bb530c7'...
> POST git-upload-pack (175 bytes)
> POST git-upload-pack (244 bytes)
> error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
> error: 30 bytes of body are still expected
> fetch-pack: unexpected disconnect while reading sideband packet
> fatal: early EOF
> fatal: fetch-pack: invalid index-pack output
> ```

Did you manage to resolve it?