kellerza/sunsynk #435: 3x DEYE SUN-12k-SG04-LP3 no configuration entities at 2. and 3. inverter

Triple5soul89ny, 2025-02-26:
### Issue related to

 Home Assistant Add-On

### Describe the issue/bug

Hi guys, first things first, thank you for this great addon!
I have a little problem. I have 3x DEYE SUN 12k connected via rs485 to tcp adaptor.
This first inverter I get alle entities for write informations to the inverter.
But the second and thrid inverter I don't have these entities.
What do I do wrong? In the log I see some timeout errors

DEYE01

[image]

DEYE02 nad DEYE03

[image]
 

### Your environment

- Home Assistant version: 2025.2.5
- Addon:
  - Name: sunsynk-multi
  - Version: 0.8.2 
- Inverter:
  - Make: Deye
  - Model: DEYE SUN-12k-SG04-LP3
  - Firmware: \_\_
- Adaptor details:
  - RS485 to tcp ebyte 
  

### You configuration

```DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "2310182302"
    HA_PREFIX: DEYE01
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: tcp://192.168.178.4:502
  - SERIAL_NR: "2310182343"
    HA_PREFIX: DEYE02
    MODBUS_ID: 2
    DONGLE_SERIAL_NUMBER: "0"
    PORT: tcp://192.168.178.4:502
  - SERIAL_NR: "2310182298"
    HA_PREFIX: DEYE03
    MODBUS_ID: 3
    DONGLE_SERIAL_NUMBER: "0"
    PORT: tcp://192.168.178.4:502
SENSOR_DEFINITIONS: three-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: DEYE
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 60
    CHANGE_ANY: false

[... 302 lines cut ...]
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/sensor_callback.py", line 110, in callback_sensor
    await ist.read_sensors(
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 55, in read_sensors
    await self.inv.read_sensors(sensors)
  File "/usr/local/lib/python3.12/site-packages/sunsynk/sunsynk.py", line 134, in read_sensors
    raise IOError("; ".join(errs))
OSError: timeout reading 1 registers from 146 poll_need_to_read
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: 15
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: xxxxx
MQTT_PASSWORD: xxxxx

```

### Logs

```logs

paste logs in 3x backticks...

```

kellerza, 2025-02-27:
Maybe change the settings? :wink:

```
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
SENSORS_FIRST_INVERTER:
  - settings
```

Waz-Cpt, 2025-02-27:
@Triple5soul89ny, I think part of your problem is that you need to have each inverter physically connected.

With a three inverter setup you would need to have a individual Modbus RTU to TCP adaptor or you need to get a 4 channel unit. Typically these gateway devices would need to you assign a different TCP port for each channel on the RTU to TCP gateway device.

jeybee74, 2025-02-27:
Hi, I have one Sunsynk 12k-SG04-LP3, currently I am using an intergration that reads/writes settings via the sunsynk cloud, which has a rather slow sampling rate at 7min..

Is this integration compatible to also write settings to the 12k deye/sunsynk inverters ?
I am planning to use a waveshare rs485 to ethernet converter.

(I was told by solar assistant that the 12k inverters is not compatible with write settings through RS485, only RS232..? 
I would really like to have local control and logging directly to HA, and at the same time use sunsynk dongle and cloud) 

Thanks.

Triple5soul89ny, 2025-02-27:
Hi,Yes it is possible to Write and read over RS485. I do this tooVon meinem iPhone gesendet

jeybee74, 2025-02-27:
>   Hi, I have one Sunsynk 12k-SG04-LP3, currently I am using an intergration that reads/writes settings via the sunsynk cloud, which has a rather slow sampling rate at 7min..
> Is this integration compatible to also write settings to the 12k deye/sunsynk inverters ?
> I am planning to use a waveshare rs485 to ethernet converter.
> (I was told by solar assistant that the 12k inverters is not compatible with write settings through RS485, only RS232..?
> I would really like to have local control and logging directly to HA, and at the same time use sunsynk dongle and cloud)
> 
> 
> 
> jeybee74 left a comment ([kellerza/sunsynk#435](https://github.com/kellerza/sunsynk/issues/435))
> Hi, I have one Sunsynk 12k-SG04-LP3, currently I am using an intergration that reads/writes settings via the sunsynk cloud, which has a rather slow sampling rate at 7min..
> Is this integration compatible to also write settings to the 12k deye/sunsynk inverters ?
> I am planning to use a waveshare rs485 to ethernet converter.
> (I was told by solar assistant that the 12k inverters is not compatible with write settings through RS485, only RS232..?
> I would really like to have local control and logging directly to HA, and at the same time use sunsynk dongle and cloud)
> Thanks.

Thanks, I will give it a try! :)

Triple5soul89ny, 2025-02-27:
> Maybe change the settings? 😉
> 
> ```
> SENSORS:
>   - energy_management
>   - power_flow_card
>   - pv2_power
> SENSORS_FIRST_INVERTER:
>   - settings
> ```

Hi kellerza,

What can I do there? The inverters are connected to Modbus but the inverters are stand alone, so they have their own Modbus Slaave addresses, but not in parallel mode.

Triple5soul89ny, 2025-02-27:
> Thanks, I will give it a try! :)

I use this RS485 to TCP converter:
https://www.amazon.de/Waveshare-RS485-Ethernet-Converter-Auto-Negotiation/dp/B09QMNWYLQ/ref=sr_1_3?crid=X40A10J3OOKS&dib=eyJ2IjoiMSJ9.GLUBp0CHYEsIiM1i7nzMU2fzgsZuX7DWq8levS6yCXpU0-xsALvg8_Q75jUgdXU405uEFZYpj7ufZ5apnNTjg1IZxOOlCDwJarPaNpd0QQ6tiysYvu05fxmRdMSI4_ghC0cwLSpV2lbQc4NxxSVQC-DlxlSO0PJQn4xXqjCe8SLh4ccuK4dutZUwsI_Zudmiz65HcihWTrLkmeUd_HN9hU1tAlcLA36TmB_QCjpSv2k.TaHDVqoUAgqgYoyaXTorOLGdfrNVzIQyyKm2OqCBGew&dib_tag=se&keywords=rs485%2Bto%2Bethernet&qid=1740686291&sprefix=rs485%2Bto%2B%2Caps%2C104&sr=8-3&th=1

Waz-Cpt, 2025-02-27:
> > Maybe change the settings? 😉
> > ```
> > SENSORS:
> >   - energy_management
> >   - power_flow_card
> >   - pv2_power
> > SENSORS_FIRST_INVERTER:
> >   - settings
> > ```
> 
> Hi kellerza,
> 
> What can I do there? The inverters are connected to Modbus but the inverters are stand alone, so they have their own Modbus Slaave addresses, but not in parallel mode.
> 
> EDIT
> 
> Oh man, ... it was a had day at work today... now I see the problem.... :-D

What was the problem. Were you able to get the data from each inverter over a single Modbus master device?

Triple5soul89ny, 2025-02-28:
Reading three inverter over one ModBus to TCP adaptor was Not the problem. You have to change ModBus adress to each inverter. You can see the config in my first post. The problem was to write something separate to each inverter.Now I figuered out, that you have to add „settings“ to inverter. Now I can write each ModBus register of each inverter.Von meinem iPhone gesendet

kellerza, 2025-03-01:
SENSORS_FIRST_INVERTER Is only useful if the three inverters works in a master-slave config.

(Probably not the most inclusive language)

closed 2025-03-01 by kellerza

jeybee74, 2025-03-01:
> I use this RS485 to TCP converter: [...]

Hi, I have configured this integration (multi) today with Waveshare RS485 to POE (B) device.
It worked fine for a bit, but the communication failes randomly.
I have configured the Waveshare device like the picture in the example settings.

Is it working fine for you?