kellerza/sunsynk #438: Communication issue SUNSYNK-12K-SG04LP3 using Waveshare RS485 to PoE (B)

jeybee74, 2025-03-02:
Hi, Thanks for a great integration 💯 !
I installed it yesterday and made it work using the same settings as shown in the Getting Started - Adapers & Wiring section for the SS 25k HV inverter.

When I moved the Waveshare POE device over to a permanent position I noticed that the data was not read every 5sek as I was expecting and now I feel I have tried everything. I now think that I probably had communication issues all along.

I have tried both serial-tcp with waveshare protocol set to none, and tcp with waveshare device protocol set to Modbus TCP-RTU. With excact same results.
I have also tried 3 cables for RS485 and 2 LAN cables + tried without PoE and local power to waveshare device.

Does anyone else had same issues with this Waveshare RS485 to PoE (B) device, and found a fix?

Current config: 
DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: *******
    HA_PREFIX: SS12k
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "0"
    PORT: serial-tcp://192.168.0.33:4196
SENSOR_DEFINITIONS: three-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
SENSORS_FIRST_INVERTER:
  - settings
  - load_limit
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 10
    REPORT_EVERY: 60
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: "60"
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: *
MQTT_PASSWORD: *

Log file:
[12:23:57] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sunsynk (no readiness notification)
s6-rc: info: service legacy-services successfully started
12:24:01 INFO    Loading config: /data/options.json
12:24:01 INFO    sunsynk library version: 0.8.2
12:24:01 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|    enum   |     |  5   |  300   |           |          |    True    |
[... 139 lines cut ...]
OSError: timeout reading 1 registers from 0; timeout reading 1 registers from 102; timeout reading 10 registers from 108; timeout reading 4 registers from 127; timeout reading 19 registers from 141; timeout reading 12 registers from 166; timeout reading 3 registers from 182 poll_need_to_read
12:36:31 ERROR   OSError in callback read_ss12k: timeout reading 10 registers from 108; timeout reading 4 registers from 127; timeout reading 19 registers from 141; timeout reading 12 registers from 166; timeout reading 3 registers from 182; timeout reading 1 registers from 209 poll_need_to_read
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/timer_callback.py", line 81, in wrap_callback
    await cb_call
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/sensor_callback.py", line 110, in callback_sensor
    await ist.read_sensors(
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 55, in read_sensors
    await self.inv.read_sensors(sensors)
  File "/usr/local/lib/python3.12/site-packages/sunsynk/sunsynk.py", line 134, in read_sensors
    raise IOError("; ".join(errs))
OSError: timeout reading 10 registers from 108; timeout reading 4 registers from 127; timeout reading 19 registers from 141; timeout reading 12 registers from 166; timeout reading 3 registers from 182; timeout reading 1 registers from 209 poll_need_to_read
12:40:01 ERROR   OSError in callback read_ss12k: timeout reading 8 registers from 0; timeout reading 2 registers from 20; timeout reading 3 registers from 62; timeout reading 1 registers from 102; timeout reading 10 registers from 108; timeout reading 4 registers from 127; timeout reading 19 registers from 141; timeout reading 12 registers from 166; timeout reading 3 registers from 182; timeout reading 1 registers from 209; timeout reading 1 registers from 500; timeout reading 1 registers from 552; timeout reading 5 registers from 587; timeout reading 1 registers from 609; timeout reading 1 registers from 619; timeout reading 4 registers from 622; timeout reading 1 registers from 636; timeout reading 6 registers from 650; timeout reading 8 registers from 672 poll_need_to_read
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/timer_callback.py", line 81, in wrap_callback
    await cb_call
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/sensor_callback.py", line 110, in callback_sensor
    await ist.read_sensors(
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 55, in read_sensors
    await self.inv.read_sensors(sensors)
  File "/usr/local/lib/python3.12/site-packages/sunsynk/sunsynk.py", line 134, in read_sensors
    raise IOError("; ".join(errs))
OSError: timeout reading 8 registers from 0; timeout reading 2 registers from 20; timeout reading 3 registers from 62; timeout reading 1 registers from 102; timeout reading 10 registers from 108; timeout reading 4 registers from 127; timeout reading 19 registers from 141; timeout reading 12 registers from 166; timeout reading 3 registers from 182; timeout reading 1 registers from 209; timeout reading 1 registers from 500; timeout reading 1 registers from 552; timeout reading 5 registers from 587; timeout reading 1 registers from 609; timeout reading 1 registers from 619; timeout reading 4 registers from 622; timeout reading 1 registers from 636; timeout reading 6 registers from 650; timeout reading 8 registers from 672 poll_need_to_read
12:40:52 WARNING Did not complete read, only read 1/10
12:44:58 ERROR   OSError in callback read_ss12k: timeout reading 19 registers from 141; timeout reading 12 registers from 166; timeout reading 3 registers from 182; timeout reading 1 registers from 209; timeout reading 1 registers from 500; timeout reading 1 registers from 552; timeout reading 5 registers from 587; timeout reading 1 registers from 609; timeout reading 1 registers from 619; timeout reading 4 registers from 622; timeout reading 1 registers from 636; timeout reading 6 registers from 650; timeout reading 8 registers from 672 poll_need_to_read

jeybee74, 2025-03-02:
RS485 is connected using the red cat6 cable

[image]

proggaras, 2025-03-10:
More infos about this splitter can be found here: https://solar-assistant.io/help/deye/2_in_1_bms_port

jeybee74, 2025-03-10:
Thank you for the suggestion :) to connect to the BMS port,

However, I get the same issue, with the connection to the BMS port, pinout is the same, pin 1-3 for RS485

proggaras, 2025-03-10:
Not 100% sure, but as I understand this splitter is just used for those models that have this CAN and RS485 port combined.

kellerza, 2025-03-18:
From your logs you successfully connect & read the initial sensors (including serial) from the inverter

 ```
    12:24:01 INFO    Connecting to serial-tcp://192.168.0.33:4196
    12:24:01 INFO    PyModbus 3.8.3 serial-tcp: 192.168.0.33:4196
    12:24:01 INFO    Reading startup sensors Device type, Protocol, Serial
    12:24:01 INFO    ############################################################
    12:24:01 INFO                 Inverter serial number '****7495'
    12:24:01 INFO    ############################################################
    12:24:01 INFO    Device type: Low voltage three phase hybrid, using the three-phase sensor definitions
    12:24:01 INFO    Protocol version: 1.4
```

The next line successfully reads all your sensors. +mqtt connect and you should get initial values in HA

```
    12:24:01 INFO    Reading all sensors Device type, Protocol, Serial, PV2 power, Battery current, Battery power, Battery SOC, Battery voltage, Day Battery Charge, Day Battery discharge, Day Grid Export, Day Grid Import, Day Load Energy, Day PV Energy, Grid Connected, Grid CT power, Grid frequency, Grid L1 power, Grid L2 power, Grid L3 power, Grid power, Inverter power, Load frequency, Load power, Load L1 power, Load L2 power, Load L3 power, Overall state, Priority Load, PV power, PV1 current, PV1 power, PV1 voltage, PV2 current, PV2 voltage, Use Timer, Total Battery Charge, Total Battery Discharge, Total Grid Export, Total Grid Import, Total PV Energy, Load Limit, Prog1 Capacity, Prog1 charge, Prog1 power, Prog1 Time, Prog2 Capacity, Prog2 charge, Prog2 power, Prog2 Time, Prog3 Capacity, Prog3 charge, Prog3 power, Prog3 Time, Prog4 Capacity, Prog4 charge, Prog4 power, Prog4 Time, Prog5 Capacity, Prog5 charge, Prog5 power, Prog5 Time, Prog6 Capacity, Prog6 charge, Prog6 power, Prog6 Time, Date Time, Grid Charge Battery current, Grid Charge Start Battery SOC, Grid Charge enabled, Solar Export, Export Limit power, Battery Max Charge current, Battery Max Discharge current, Battery Capacity current, Battery Shutdown Capacity, Battery Restart Capacity, Battery Low Capacity, Battery Type, Battery Wake Up, Battery Resistance, Battery Charge Efficiency, Grid Standard, Configured Grid Frequency, Configured Grid Phases, UPS delay time, Rated power
 ```

but then things start timing out.

So either you unplugged the RS485 cable after the addon started up, or the waveshare acts up in some way.

You could try:
- adding only 1/2 sensors (no groups!!)
- making the schedules muuuuuch less aggressive. See the solarman example in the docs as guidance.

Once either of the above is working, gradually increase things (more sensors, more frequent reads) until it is working as you expect.

If it's a device issue, you can also try playing with these settings: (but almost doubt it will make a difference, since that first read of all sensors was ok)

```yaml
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
```

jeybee74, 2025-03-27:
> From your logs you successfully connect & read the initial sensors (including serial) from the inverter
> 
> ```
>    12:24:01 INFO    Connecting to serial-tcp://192.168.0.33:4196
>    12:24:01 INFO    PyModbus 3.8.3 serial-tcp: 192.168.0.33:4196
>    12:24:01 INFO    Reading startup sensors Device type, Protocol, Serial
>    12:24:01 INFO    ############################################################
>    12:24:01 INFO                 Inverter serial number '****7495'
>    12:24:01 INFO    ############################################################
>    12:24:01 INFO    Device type: Low voltage three phase hybrid, using the three-phase sensor definitions
>    12:24:01 INFO    Protocol version: 1.4
> ```
> 
> The next line successfully reads all your sensors. +mqtt connect and you should get initial values in HA
> 
> ```
>     12:24:01 INFO    Reading all sensors Device type, Protocol, Serial, PV2 power, Battery current, Battery power, Battery SOC, Battery voltage, Day Battery Charge, Day Battery discharge, Day Grid Export, Day Grid Import, Day Load Energy, Day PV Energy, Grid Connected, Grid CT power, Grid frequency, Grid L1 power, Grid L2 power, Grid L3 power, Grid power, Inverter power, Load frequency, Load power, Load L1 power, Load L2 power, Load L3 power, Overall state, Priority Load, PV power, PV1 current, PV1 power, PV1 voltage, PV2 current, PV2 voltage, Use Timer, Total Battery Charge, Total Battery Discharge, Total Grid Export, Total Grid Import, Total PV Energy, Load Limit, Prog1 Capacity, Prog1 charge, Prog1 power, Prog1 Time, Prog2 Capacity, Prog2 charge, Prog2 power, Prog2 Time, Prog3 Capacity, Prog3 charge, Prog3 power, Prog3 Time, Prog4 Capacity, Prog4 charge, Prog4 power, Prog4 Time, Prog5 Capacity, Prog5 charge, Prog5 power, Prog5 Time, Prog6 Capacity, Prog6 charge, Prog6 power, Prog6 Time, Date Time, Grid Charge Battery current, Grid Charge Start Battery SOC, Grid Charge enabled, Solar Export, Export Limit power, Battery Max Charge current, Battery Max Discharge current, Battery Capacity current, Battery Shutdown Capacity, Battery Restart Capacity, Battery Low Capacity, Battery Type, Battery Wake Up, Battery Resistance, Battery Charge Efficiency, Grid Standard, Configured Grid Frequency, Configured Grid Phases, UPS delay time, Rated power
> ```
> 
> but then things start timing out.
> 
> So either you unplugged the RS485 cable after the addon started up, or the waveshare acts up in some way.
> 
> You could try:
> 
> * adding only 1/2 sensors (no groups!!)
> * making the schedules muuuuuch less aggressive. See the solarman example in the docs as guidance.
> 
> Once either of the above is working, gradually increase things (more sensors, more frequent reads) until it is working as you expect.
> 
> If it's a device issue, you can also try playing with these settings: (but almost doubt it will make a difference, since that first read of all sensors was ok)
> 
> READ_ALLOW_GAP: 2
> READ_SENSORS_BATCH_SIZE: 20

Thanks Johann, 

I have tried updating the Waveshare rs485 to poe device to the latest firmware, still no improvements, I mean the integration is ok, I get data but still a lot of timeouts, seems to be about 3600 in a 24h period.

I have tried quite a few settings both for READ_ALLOW_GAP and READ_SENSORS_BATCH_SIZE, but it is the same issue with timeouts.
I have also tried READ every 30seconds, still the same.

I have ordered a new waveshare device, version (C) from another supplier along with a rs485 to USB, so I will test some more in a week or two.

I will post my results here when I hopefully get this fixed :)

Regards Jon

Waz-Cpt, 2025-04-01:
I made my own Wifi/LAN RS485 gateway device from an ESP32 with LAB and an RS485 breakout module.

[https://github.com/kellerza/sunsynk/discussions/324#discussion-7043693](url)

It has been running pretty reliably since August last year.

jeybee74, 2025-04-02:
> I made my own Wifi/LAN RS485 gateway device from an ESP32 with LAB and an RS485 breakout module.
> 
> [https://github.com/kellerza/sunsynk/discussions/324#discussion-7043693](url)
> 
> It has been running pretty reliably since August last year.

How often do you collect data, and is your entities updated consistently?

I have now run the waveshare USB-rs485 adapter since last night with READ_EVERY: 5.
sometimes the entity is updated every 5sek-ish and sometimes longer, it is not consistent, often it can be over 20, and sometimes 38sek+ for a entity to update.
I think the issue with the waveshare ethernet device is just that this type of delay in data is shown as timeout and that it also happens on the USB device but then it is not reported as a timeout..

I am not sure, but maybe a firmware update on the inverter could fix this, or maybe all users experience the same issue but doesn't notice it..?

I also see that the data from the CT is often reported all over the place, like you see in this picture

[image]

BR

Waz-Cpt, 2025-04-02:
[image]

`DRIVER: pymodbus
INVERTERS:
  - SERIAL_NR: "********"
    HA_PREFIX: sunsynk
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "*********"
    PORT: tcp://192.168.10.***:502
SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - power_flow_card
  - diagnostics
  - battery
SENSORS_FIRST_INVERTER:
  - settings
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 5
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 10
    REPORT_EVERY: 30
    CHANGE_BY: 80
  - KEY: RW
    READ_EVERY: 10
    REPORT_EVERY: 30
    CHANGE_ANY: true
  - KEY: any_unit
    READ_EVERY: 15
    REPORT_EVERY: 30
    CHANGE_BY: 80
NUMBER_ENTITY_MODE: auto
PROG_TIME_INTERVAL: 15
MQTT_HOST: 192.168.10.***
MQTT_PORT: 1883
MQTT_USERNAME: ***********
MQTT_PASSWORD: ***********
DEBUG: 2
DEBUG_DEVICE: /dev/ttyS0`

jeybee74, 2025-04-02:
heres how my debug log is:

[Debug_log_ss12k.txt](https://github.com/user-attachments/files/19566787/Debug_log_ss12k.txt)

mamontuka, 2025-04-06:
put cable in **METER485** port. it will work by **ORANGE pair** - white-orange is B(D-) and orange is A(D+)

jeybee74, 2025-04-08:
> put cable in **METER485** port. it will work by **ORANGE pair** - white-orange is B(D-) (rj45 pin 1) and orange is A(D+) (rj45 pin 2)

thanks for your comment.

I have now tried 7, yes SEVEN different USB-RS485 devices. ALL works fine with no timeout issues :)
So I guess the cabling from the inverter is fine ;)

I have also tried 3 different Waveshare RS485-ethernet devices; (B), Poe (B) and (C), with different settings, same issue with timeouts on all.
But they all work fine and data is coming in to the entities just as good as with the USB-RS485 devices 👍 

BUT also with the USB device the data is not updated consistently every 5sek as it should 👎 , sometimes the entities is updated every 5 sek, but sometimes the data is 28sek, or more before the entity is updated (sometimes 1m+), there is no timeouts while this happends via the USB-RS485 devices.

I am beginning to think that the "timeout" issues or whatever is always there, but that it is visible in the logs only when I use the Ethernet devices due to the different communication methods, if this makes sense...?

jeybee74, 2025-05-06:
Here is an update of my trials with the Waveshare ETH devices.

I have tried a lot of settings and I have found out that this is the settings that works best for me:

[image]
Marked out the RS485-ETH adapter spesific settings.

remember to set the IP and port to the Waveshare ETH device.

This gives me still a lot of timeouts, but all data is correct in the entities.

jeybee74, 2025-06-02:
Issues with timeouts was fixed in the latest update.
The integration now shows 0 timeouts with my Waveshare ETH RS-485 device since I updated to edge/dev version 3d60254.

👍👍