kellerza/sunsynk #440: How come the three phase definitions don't include essential power?

Nurgus, 2025-03-09:
How come the three phase definitions don't include essential power?

It's in the single phase definition. Am I missing something really obvious? I can calculate it for a sensor in ha.

proggaras, 2025-03-10:
There is:
```
essential_l1_power
essential_l2_power
essential_l3_power
```

Nurgus, 2025-03-10:
Where are you getting that from? Those are not in the definitions and I've even checked "essential.*l1" against all the code with no matches. I must be being really dumb?

proggaras, 2025-03-10:
You are right. I couldn't find essential.*l1 in the Deye documentation as well. My sensors exist because I created 3 helpers in Home Assistant. 

[image]

Nurgus, 2025-03-10:
Hi, I appreciate your thoughts but I'm asking specifically why the sensor definition doesn't exist for the three phase?  It seems odd that it exists for single phase and not for three phase.

As I said in the original comment, I can create them in HA.

kellerza, 2025-03-18:
It's not a register, but a calculation using several registers. Single phase has several of these, depending on how people's inverter are wired & configured.

If you have a proposal for the 3-phase, you are welcome to create a PR