kellerza/sunsynk #453: Approved work with Deye 6K-SG03LP-1-EU

mamontuka, 2025-04-06:
All works fine over rs485 to ethernet gateway, with turned on modbus2tcp function : https://vkmodule.com.ua/Converter/ConverterENET485_ua.html with **Deye 6K-SG03LP-1-EU** inverter, tested last DEV EDGE version **d0c6cd5**

I think will work with all SG03LP family

Two batteries **Ritar BAT-5KWH-51.2V** LiFePo4 : https://github.com/mamontuka/ritar-bms-ha (rs485 mode)

Not critical - inverter have **two** PV, but your addon anonce **three** - but ok, its not critical at all

Bad - alot wrong sensors registers and BMS protocols, repaired by me with mysensors.py (here will be wrong intendation,how im understand, so im append mysensors.py(txt) below

Definitions : 

SENSOR_DEFINITIONS: single-phase
SENSORS:
  - energy_management
  - diagnostics
  - power_flow_card
  - load_limit
  - advanced
  - mysensors
SENSORS_FIRST_INVERTER:
  - settings

code - [mysensors.py.txt](https://github.com/user-attachments/files/19619619/mysensors.py.txt)

#############################################################
#############   Deye SG03LP-1 Series ajustments   ####################
#########    for Sunsynk addon https://github.com/kellerza/sunsynk     ######
####################   by Oleh Mamont    #########################
#############################################################

from sunsynk import AMPS, CELSIUS, KWH, VOLT, WATT
from sunsynk.rwsensors import NumberRWSensor, SelectRWSensor, TimeRWSensor
from sunsynk.sensors import (
    MathSensor,
    Sensor,
    SensorDefinitions,
    TempSensor,
)

SENSORS = SensorDefinitions()

SENSORS += (
    Sensor(182, "Battery Temperature", CELSIUS, 0.1),
    Sensor(166, "Generator Power", WATT, -1),
    Sensor(181, "Generator Voltage", VOLT, 0.1),
    Sensor(196, "Generator Frequency", "Hz", 0.1),
)

BMS_protocol = {
    0: "00_PYLON_CAN",
    1: "01_SACRED_SUN_RS485",
    2: "02",
    3: "03_DYNESS_CAN",
    4: "04",
    5: "05",
    6: "06_GenixGreen_RS485",
    7: "07",
    8: "08",
    9: "09",
    10: "10",
    11: "11",
    12: "12_PYLON_RS485",
    13: "13_VISION_CAN",
    14: "14_WATTSONIC_RS485",
    15: "15_UNIPOWER_RS485",
    16: "16",
    17: "17_LD_RS485",
    18: "18",
    19: "19_UNKNOWN_RS485",
}

SENSORS += (
    SelectRWSensor(325, "BMS Protocol", options=BMS_protocol),
)

################################################################

Thank you for work !

jeybee74, 2025-04-08:
Hi, what kind of RS485-ethernet device do you use?

I have tried 3 different types from Waveshare and I get a lot of timeouts in my logs.
The intergration works fine, no issues, but the logfile is messy with all these timeouts.

I have also tested about 7 different USB-RS485 converters from 3 different manufacturers and they ALL worked as fine, but no timeouts in the logs.
I want to have my HASS-device in another location and therefore I would prefer to use RS485-ethernet rather than the USB-RS485 device.

SS-12K-3phase-EU model LV hybrid

mamontuka, 2025-04-08:
> Hi, what kind of RS485-ethernet device do you use?
>
> I have tried 3 different types from Waveshare and I get a lot of timeouts in my logs. The intergration works fine, no issues, but the logfile is [...]

Hello, im live in Ukraine and have our local brand devices - https://vkmodule.com.ua/Converter/ConverterENET485_ua.html
Very nice and not expensive at all : 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Second device wired by brown pair to my master battery, and read batteries.

on inverter board rs485 orange and brown pairs mirrored to each other by vendor schematics, i mean white-orange=brown, orange=white-brown.

mamontuka, 2025-04-08:
also try to do timeouts in integration config: 
- KEY: W
  READ_EVERY: 15
  REPORT_EVERY: 60
- KEY: RW
  READ_EVERY: 15
  REPORT_EVERY: 60
  CHANGE_ANY: true
- KEY: any_unit
  READ_EVERY: 30
  REPORT_EVERY: 60
  CHANGE_BY: 80
- KEY: no_unit
  READ_EVERY: 15
  REPORT_EVERY: 60
  CHANGE_ANY: true

that values work fine and responsive for me

mamontuka, 2025-04-08:
Also, all low voltage devices - relays, rs485 gates, mikrotik router and ethernet switch - powered by low voltage UPS boards : https://proline.biz.ua/ef9012 with own 26ah lead batteries , connected to deye load for better autonomy. that boards have not stabilized output 10.5-13.8, so if you need stabilized low voltage output - you need add to scheme dc-dc raise-down converters after UPS board, like i do here : 

[image]

here is time-by-time thug life without electricity (