kellerza/sunsynk #458: read_ss error since last update

Wayne-nulitics, 2025-04-09:
Since the update 4 days ago I'm getting poll errors, everything worked fine before

Error:

16:09:19 ERROR   OSError in callback read_ss: timeout reading 1 registers from 0 poll_need_to_read
Traceback (most recent call last):
  File "/.venv/lib/python3.12/site-packages/ha_addon_sunsynk_multi/timer_callback.py", line 81, in wrap_callback
    await cb_call
  File "/.venv/lib/python3.12/site-packages/ha_addon_sunsynk_multi/sensor_callback.py", line 112, in callback_sensor
    await ist.read_sensors(
  File "/.venv/lib/python3.12/site-packages/ha_addon_sunsynk_multi/a_inverter.py", line 55, in read_sensors
    await self.inv.read_sensors(sensors)
  File "/.venv/lib/python3.12/site-packages/sunsynk/sunsynk.py", line 135, in read_sensors
    raise OSError("; ".join(errs))
OSError: timeout reading 1 registers from 0 poll_need_to_read

Upgraded core to 2025.4.1 and supervisor to 2025.03.4
Upgraded plugin to version d0c6cd5

I rebooted the Rasberry Pi, and then that error, I left it for a few minutes and now the plugin / dashboard updates but I lost data on Daily Charge, Daily Discharge and Daily Load. Will monitor for a day or two and updated issue

Wayne-nulitics, 2025-04-09:
Additionally, getting these errors now after a few hours:

16:25:00 ERROR   (1 in 5 min) OSError in callback read_ss: timeout reading 1 registers from 0 poll_need_to_read
16:50:10 WARNING Did not complete read, only read 12/1
16:50:10 WARNING device_type: Unknown register value 80. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
17:14:13 WARNING Did not complete read, only read 12/1
17:14:13 WARNING device_type: Unknown register value 80. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
17:16:58 WARNING Did not complete read, only read 12/1
17:16:58 WARNING device_type: Unknown register value 80. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions
17:40:00 ERROR   (1 in 5 min) OSError in callback read_ss: timeout reading 1 registers from 0 poll_need_to_read
18:14:43 WARNING Did not complete read, only read 12/1
18:14:43 WARNING device_type: Unknown register value 80. Consider extending the definition with a PR. https://github.com/kellerza/sunsynk/tree/main/src/sunsynk/definitions

mamontuka, 2025-04-10:
> Since the update 4 days ago I'm getting poll errors, everything worked fine before
>
> Error:
>
> 16:09:19 ERROR OSError in callback read_ss: timeout reading 1 registers from 0 poll_need_to_read Traceback [...]

what kind inverter model you have ?

Wayne-nulitics, 2025-04-10:
I have the Single Phase SunSynk 16KW 3x MTTP inverter

mamontuka, 2025-04-10:
> I have the Single Phase Sunsynk 16KW 3x MTTP inverter

on single phase Daily Charge, Daily Discharge and Daily Load - is 72,73,84

try add ovverride custom sensors construction : 

#############################################################################
#############   Deye SG03LP1 Series ajustments   ###########################
#########    for Sunsynk addon https://github.com/kellerza/sunsynk     ######
########################    by Oleh Mamont    ###############################
#############################################################################

from sunsynk import AMPS, CELSIUS, KWH, VOLT, WATT
from sunsynk.rwsensors import NumberRWSensor, SelectRWSensor, TimeRWSensor
from sunsynk.sensors import (
    MathSensor,
    Sensor,
    SensorDefinitions,
    TempSensor,
)

SENSORS = SensorDefinitions()

SENSORS += (
    Sensor(182, "Battery Temperature", CELSIUS, 0.1),
    Sensor(166, "Generator Power", WATT, -1),
    Sensor(181, "Generator Voltage", VOLT, 0.1),
    Sensor(196, "Generator Frequency", "Hz", 0.1),
    Sensor(72, "Daily Charge", WATT, -1),
    Sensor(73, "Daily Discharge", WATT, -1),
    Sensor(84, "Daily Load", WATT, -1),
)

BMS_protocol = {
    0: "00_PYLON_CAN",
    1: "01_SACRED_SUN_RS485",
    2: "02",
    3: "03_DYNESS_CAN",
    4: "04",
    5: "05",
    6: "06_GenixGreen_RS485",
    7: "07",
    8: "08",
    9: "09",
    10: "10",
    11: "11",
    12: "12_PYLON_RS485",
    13: "13_VISION_CAN",
    14: "14_WATTSONIC_RS485",
    15: "15_UNIPOWER_RS485",
    16: "16",
    17: "17_LD_RS485",
    18: "18",
    19: "19_UNKNOWN_RS485",
}

SENSORS += (
    SelectRWSensor(325, "BMS Protocol", options=BMS_protocol),
)

properly indented file binded, add like telled in addon manual mysensors.py

total daily load can be 85,86 registers too

[mysensors.py.txt](https://github.com/user-attachments/files/19687975/mysensors.py.txt)

mamontuka, 2025-04-10:
oh i did mistake... in strings : 

Sensor(72, "Daily Charge", WATT, -1),
Sensor(73, "Daily Discharge", WATT, -1),
Sensor(84, "Daily Load", WATT, -1),

WATT - must be KWH i think

Wayne-nulitics, 2025-04-10:
I've dropped the READ_SENSOR_BATCH_SIZE to 20, will see if that helps

mamontuka, 2025-04-10:
> I'm getting all the Daily values, that's fine, the issue is the constant error and warning logs:
> 
Unknown register value 80.
> Other that that it's working fine

ok, im understand now. your inverter just dont know what is register 80

normaly register 80 is Total Energy Import so you need override that register by right 78 i think

so construction of custom sensor will be some like this:

Sensor(78, "Total Grid Import", KWH, 0.1),