kellerza/sunsynk #469: Possible reading data from every battery cell?

europer, 2025-05-02:
First of all, fantastic program—I've been using it for 15 months with no issues (Deye 12K 3-Phase Inverter).
I have a question: Is it possible to monitor temperature, voltage, and current per rack?
I have 5 racks, each with 48V 100Ah batteries. In total, that’s 5 racks with 16 cells per rack (each cell 3.2V), and I’d like to get a deeper view into the system at the rack or even cell level.

Thanks,
Per

mamontuka, 2025-05-25:
from back side, like im done here : https://github.com/mamontuka/ritar-bms-ha/

basti122303, 2025-06-03:
It depend on what batteries / BMS you are using. For Deye batteries you can use https://github.com/Psynosaur/esphome-deye-bms for example.