kellerza/sunsynk #492: Custom Sensors Not Working For Me

alistairts, 2025-06-29:
I created custom sensors with a 50kW 3ph inverter before without issue, but when I try do the same for a 5kW 1ph inverter it is just not working, I hope you can assist

Here is my mysensor.py:
from sunsynk import AMPS, CELSIUS, KWH, VOLT, WATT
from sunsynk.rwsensors import NumberRWSensor, SelectRWSensor, SwitchRWSensor, SystemTimeRWSensor, TimeRWSensor
from sunsynk.sensors import (
    BinarySensor,
    FaultSensor,
    InverterStateSensor,
    MathSensor,
    SDStatusSensor,
    Sensor,
    TempSensor,
)

SENSORS = SensorDefinitions()

SENSORS += (
	SystemTimeRWSensor((22, 23, 24), "Date Time", unit=""),
	InverterStateSensor(59, "Inverter Status"),
	Sensor(70, "Day Battery Charge", KWH, 0.1),
	Sensor(71, "Day Battery Discharge", KWH, 0.1),
	Sensor((72, 73), "Total Battery Charge", KWH, 0.1),
    	Sensor((74, 75), "Total Battery Discharge", KWH, 0.1),
    	Sensor(76, "Day Grid Import", KWH, 0.1),
    	Sensor(77, "Day Grid Export", KWH, 0.1),
	Sensor((78, 80), "Total Grid Import", KWH, 0.1),
	Sensor((81, 82), "Total Grid Export", KWH, 0.1),
	TempSensor(90, "DC Transformer Temperature", CELSIUS, 0.1),
	TempSensor(91, "IGBT Temperature", CELSIUS, 0.1),
	Sensor(93, "Inverter Power Factor", 0.1),
	Sensor((96, 97), "Total PV Energy", KWH, 0.1),
	FaultSensor((103, 104, 105, 106), "Fault Information"),
	Sensor(108, "Day PV Energy", KWH, 0.1),
	Sensor(109, "PV Voltage 1", VOLT, 0.1),
   	Sensor(110, "PV Current 1", AMPS, 0.1),
	Sensor(111, "PV Voltage 2", VOLT, 0.1),
   	Sensor(112, "PV Current 2", AMPS, 0.1),
	Sensor(150, "Grid Port Voltage", VOLT, 0.1),
	Sensor(154, "Inverter Voltage", VOLT, 0.1),
	Sensor(160, "Grid Port Current", AMPS, 0.01),
	Sensor(164, "Inverter Current", AMPS, 0.01),
	Sensor(166, "Smart Loads Power", WATT, -1),
	Sensor(167, "Grid Port Power", WATT, -1),
	Sensor(172, "Grid CT Power", WATT, -1),
	MathSensor((172, 167), "Non Essential Power", WATT, factors=(1, -1), no_negative=True),
	Sensor(173, "Inverter Power", WATT, -1),
	Sensor(176, "Essential Loads Power", WATT, -1),
	Sensor(181, "Smart Loads Voltage", VOLT, 0.1),
	TempSensor(182, "Battery Temperature", CELSIUS, 0.1),
	Sensor(183, "Battery Voltage", VOLT, 0.01),
    	Sensor(184, "Battery SOC", "%"),
	Sensor(186, "PV Power 1", WATT, 1),
	Sensor(187, "PV Power 2", WATT, 1),
	Sensor(190, "Battery Power", WATT, -1),
    	Sensor(191, "Battery Current", AMPS, -0.01),
	Sensor(192, "Essential Loads Frequency", HZ, 0.01),
	Sensor(193, "Inverter Frequency", HZ, 0.01),
	BinarySensor(194, "Grid Relay Status"),
	BinarySensor(195, "Smart Loads Relay Status"),
	Sensor(196, "Smart Loads Frequency", HZ, 0.1),
	NumberRWSensor(210, "Battery Max Charge Current", AMPS, min=0, max=185),
	NumberRWSensor(211, "Battery Max Discharge Current", AMPS, min=0, max=185),
	Sensor(216, "Battery Charging Efficiency", "%"),
	NumberRWSensor(229, "Grid Charge Start Battery SOC", "%", min=1, max=95),
	NumberRWSensor(230, "Grid Charge Battery Current", AMPS, min=0, max=185),
	SwitchRWSensor(232, "Grid Charge", "", bitmask=1),
	NumberRWSensor(237, "Smart Loads OFF Battery SOC", "%", min=1, max=95),
	NumberRWSensor(239, "Smart Loads ON Battery SOC", "%", min=1, max=95),
	SwitchRWSensor(243, "Priority Load", bitmask=1),
	SwitchRWSensor(248, "Use Timer", bitmask=1),
	NumberRWSensor(282, "Grid Reconnection Time", "s", min=10, max=300),
	NumberRWSensor(294, "Smart Loads Reconnection Time", "min", min=1, max=120),
)

Here is my configuration (minus the sensitive information):
DRIVER: solarman
INVERTERS:
  - SERIAL_NR: ""
    HA_PREFIX: SS
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: ""
    PORT: tcp://192.168.101.79:8899
SENSOR_DEFINITIONS: single-phase
SENSORS:
- mysensors
SENSORS_FIRST_INVERTER:
  - prog1_capacity
  - prog1_charge
  - prog1_power
  - prog1_time
  - prog2_capacity
  - prog2_charge
  - prog2_power
  - prog2_time
  - prog3_capacity
  - prog3_charge
  - prog3_power
  - prog3_time
  - prog4_capacity
  - prog4_charge
  - prog4_power
  - prog4_time
  - prog5_capacity
  - prog5_charge
  - prog5_power
  - prog5_time
  - prog6_capacity
  - prog6_charge
  - prog6_power
  - prog6_time
MANUFACTURER: Sunsynk
READ_ALLOW_GAP: 1
READ_SENSORS_BATCH_SIZE: 30
SCHEDULES:
  - KEY: date_time
    READ_EVERY: 60
    REPORT_EVERY: 60
  - KEY: enum
    READ_EVERY: 10
    REPORT_EVERY: 10
  - KEY: RW
    READ_EVERY: 10
    REPORT_EVERY: 10
  - KEY: W
    READ_EVERY: 10
    REPORT_EVERY: 10
  - KEY: kWh
    READ_EVERY: 10
    REPORT_EVERY: 10
  - KEY: V
    READ_EVERY: 10
    REPORT_EVERY: 10
  - KEY: A
    READ_EVERY: 10
    REPORT_EVERY: 10
  - KEY: "%"
    READ_EVERY: 10
    REPORT_EVERY: 10
  - KEY: any_unit
    READ_EVERY: 10
    REPORT_EVERY: 10
  - KEY: no_unit
    READ_EVERY: 10
    REPORT_EVERY: 10
NUMBER_ENTITY_MODE: box
PROG_TIME_INTERVAL: 15
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: 
MQTT_PASSWORD: 
DEBUG_DEVICE: /dev/ttyS0

Here is the log from the part which shows something is wrong:
18:55:39 INFO    Importing /share/hass-addon-sunsynk/mysensors.py...
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/helpers.py", line 47, in import_mysensors
    mod = import_module("mysensors", str(root))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ha_addon_sunsynk_multi/helpers.py", line 20, in import_module
    mod_obj = _import_module(mod_name)
              ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 999, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/share/hass-addon-sunsynk/mysensors.py", line 13, in <module>
    SENSORS = SensorDefinitions()
              ^^^^^^^^^^^^^^^^^
NameError: name 'SensorDefinitions' is not defined
18:55:39 ERROR   Error importing /share/hass-addon-sunsynk/mysensors.py: name 'SensorDefinitions' is not defined
18:55:39 ERROR   Unknown sensor specified: mysensors
18:55:39 INFO    Added hidden sensors as other sensors depend on it: Device type, Protocol, Serial, Battery Low Capacity, Rated power, Battery Shutdown Capacity
18:55:39 INFO    Connecting to tcp://192.168.101.79:8899
18:55:39 INFO    Reading startup sensors Device type, Protocol, Serial
18:55:39 INFO    ############################################################
18:55:39 INFO                 Inverter serial number '****2979'
18:55:39 INFO    ############################################################
18:55:39 INFO    Device type: Single phase hybrid, using the single-phase sensor definitions
18:55:39 INFO    Protocol version: 2.1
18:55:39 INFO    Reading all sensors Device type, Protocol, Serial, Prog1 Capacity, Prog1 charge, Prog1 power, Prog1 Time, Prog2 Capacity, Prog2 charge, Prog2 power, Prog2 Time, Prog3 Capacity, Prog3 charge, Prog3 power, Prog3 Time, Prog4 Capacity, Prog4 charge, Prog4 power, Prog4 Time, Prog5 Capacity, Prog5 charge, Prog5 power, Prog5 Time, Prog6 Capacity, Prog6 charge, Prog6 power, Prog6 Time, Battery Low Capacity, Rated power, Battery Shutdown Capacity
18:55:40 INFO    MQTT: Connecting to Mosquitto@core-mosquitto:1883
18:55:40 INFO    MQTT: Connection successful
18:55:41 INFO    Read every (inverter 1)

kellerza, 2025-06-29:
Missing imports?

```python
File "/share/hass-addon-sunsynk/mysensors.py", line 13, in
SENSORS = SensorDefinitions()
^^^^^^^^^^^^^^^^^
NameError: name 'SensorDefinitions' is not defined
18:55:39 ERROR Error importing /share/hass-addon-sunsynk/mysensors.py: name 'SensorDefinitions' is not defined
```

closed 2025-07-05 by alistairts

kellerza, 2025-07-05:
> Just an update, I started again and now it works, I must've had some dodgy code in the mix...

Nothing dodgy, you simply missed the import of SensorDefinitions