kellerza/sunsynk #494: MQTT: Home Assistant not online - problem

helnora, 2025-07-09:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

The addon terminates with following messages:
WARNING Payload >20000: 26765
ERROR   MQTT: Home Assistant not online. Topic homeassistant/status is empty

### Your environment

- Home Assistant version: 2025.07.01
- Addon:
  - Name: sunsynk-edge
  - Version: 7509f46
- Inverter:
  - Make: Deye 
  - Model: SUN-12K-SG04LP3-EU
  - Firmware: \_\_
- Adaptor details:
  - RS485/RS232/solarman

### You configuration

DRIVER: solarman
INVERTERS:
  - SERIAL_NR: "zzz"
    HA_PREFIX: SS
    MODBUS_ID: 1
    DONGLE_SERIAL_NUMBER: "xxx"
    PORT: tcp://192.168.1.219:8899
SENSOR_DEFINITIONS: three-phase
SENSORS:
  - energy_management
  - power_flow_card
  - pv2_power
  - settings
  - load_limit
  - battery
  - battery_temperature
  - rated_power
  - grid_peak-shaving
SENSORS_FIRST_INVERTER: []
MANUFACTURER: Deye
READ_ALLOW_GAP: 2
READ_SENSORS_BATCH_SIZE: 20
SCHEDULES:
  - KEY: W
    READ_EVERY: 5
    REPORT_EVERY: 30
    CHANGE_ANY: false
    CHANGE_BY: 80
    CHANGE_PERCENT: 0
NUMBER_ENTITY_MODE: box
PROG_TIME_INTERVAL: 15
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqtt_user
MQTT_PASSWORD: 
DEBUG_DEVICE: /dev/ttyS0

### Logs

06:15:21 INFO    Report every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  30 |       load_l3_power, grid_ct_power, grid_l2_power, rated_power, pv_power,        |
|     |     inverter_power, load_power, grid_l1_power, grid_l3_power, battery_power,     |
|     |    grid_power, load_l1_power, load_l2_power, pv2_power, pv1_power, pv3_power,    |
|     |                                    pv4_power                                     |
| 300 |      grid_peak_shaving, prog2_power, day_battery_charge, total_grid_export,      |
|     |     prog2_capacity, battery_equalization_hours, day_grid_import, load_limit,     |
|     |     battery_wake_up, pv1_current, prog6_power, grid_connected, prog4_power,      |
|     |      total_pv_energy, day_battery_discharge, battery_max_discharge_current,      |
|     |     overall_state, battery_resistance, prog4_capacity, grid_charge_enabled,      |
|     |              total_battery_discharge, battery_voltage, pv4_voltage,              |
|     |      battery_equalization_days, battery_equalization_voltage, prog2_charge,      |
|     |  prog6_charge, prog3_capacity, battery_low_voltage, prog6_capacity, prog5_time,  |
|     |              battery_type, load_l1_voltage, use_timer, prog3_power,              |
|     | battery_shutdown_capacity, battery_shutdown_voltage, grid_frequency, prog4_time, |
|     |     battery_restart_capacity, battery_restart_voltage, battery_temperature,      |
|     |       day_load_energy, battery_capacity_current, battery_soc, prog2_time,        |
|     |      battery_absorption_voltage, day_pv_energy, prog5_charge, pv3_voltage,       |
|     |     load_l3_voltage, priority_load, day_grid_export, battery_float_voltage,      |
|     | total_battery_charge, pv1_voltage, prog5_capacity, load_l2_voltage, pv2_voltage, |
|     |     total_grid_import, grid_current, pv3_current, solar_export, pv2_current,     |
|     |  prog1_charge, prog3_time, prog1_capacity, prog1_time, prog6_time, prog1_power,  |
|     |     battery_charge_efficiency, battery_current, prog4_charge, prog3_charge,      |
|     |    prog5_power, pv4_current, export_limit_power, battery_max_charge_current,     |
|     |                               battery_low_capacity                               |
+-----+----------------------------------------------------------------------------------+
06:15:23 WARNING Payload >20000: 26765
06:15:31 ERROR   MQTT: Home Assistant not online. Topic homeassistant/status is empty
[06:15:31] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

kellerza, 2025-07-09:
From the HA MQTT docs. Did you change this?

By default, Home Assistant sends online and offline to homeassistant/status.

MQTT Birth and Last Will messages can be customized or disabled from the UI. To do this, click on “Configure” in the integration page in the UI, then “Re-configure MQTT” and then “Next”.

Waz-Cpt, 2025-07-09:
You need to set retain birth message, this is not set by default. That solved it for me.

FYI, does anyone else have a problem with using the backup feature with this add-on? I cannot use the automated backup before update feature. There are some inconsistencies with the file name generation.

helnora, 2025-07-09:
Ok, changing the "will message" under MQTT option, and enabling "Birth message retain" and "Will message payload" to "online" did fix the problem. Now I get some warning messages in the log.

WARNING Payload >20000: 26765
WARNING MQTT: Topic SS/ss/load_limit_set already subscribed
WARNING MQTT: Topic SS/ss/grid_peak_shaving_set already subscribed
etc

kellerza, 2025-07-09:
Thanks, will have to document the need for LWT messages

Everything should be ok even with the warning messages. 

Mqttexplorer has a 20kb limit, but HA accepts them (256Mb is the protocol limit)

Need to track re-subscriptions a bit better, but not expecting any serious issues