kellerza/sunsynk #516: Change in scaling of Energy Sensors after Firmware Update

Johnny1206, 2025-09-05:
### Your environment

- Home Assistant version: Core: 2025.2.5 | OS: 14.2
- Addon:
  - Name: multi
  - Version: 0.8.2
- Inverter:
  - Make: Deye
  - Model: 25k-SG01HP3
  - Firmware: MCU1:C048-0804 | MCU2:1098-0709
- Adaptor details:
  - RS485

### Describe the issue/bug and what you expect

Today I installed a Firmware Update on my inverter: 
Old Firmware Version:
- MCU1-3PHV-Ver2001-C040-T3-CK
- MCU2_SG15KHP3-V300x-1095-0407

New Firmware Version:
- MCU1-3PHV-Ver2001-C048-0804-CK
- MCU2_SG15KHP3-V300x-1098-0709

After installation of the new Firmware, all Energy Sensors I use show a value which is too small by a factor of 10.

As a quick fix I redefined the sensors in the mysensors.py:
    # Redefinition of existing Sensors because value are too small by a factor of 10
    Sensor(514, "Day Battery Charge", KWH, 1),
    Sensor(515, "Day Battery discharge", KWH, 1),
    Sensor((516, 517), "Total Battery Charge", KWH, 1),
    Sensor((518, 519), "Total Battery Discharge", KWH, 1),
    Sensor(529, "Day PV Energy", KWH, 1),
    Sensor((534, 535), "Total PV Energy", KWH, 1),
    Sensor(536, "Day Gen Energy", KWH, 1),
    Sensor(520, "Day Grid Import", KWH, 1),
    Sensor(521, "Day Grid Export", KWH, 1),
    Sensor((522, 523), "Total Grid Import", KWH, 1),
    Sensor((524, 525), "Total Grid Export", KWH, 1),
    Sensor(526, "Day Load Energy", KWH, 1),
    Sensor((527, 528), "Total Load Energy", KWH, 1),

With the new scaling all sensors return the right values again. 
Compared to the Energy Sensors in the definitions the scaling of the energy registers seems to have been changed from 0.1 kWh to 1 kWh:
SENSORS += (
    Sensor(502, "Day Active Energy", KWH, -0.1),
    Sensor(514, "Day Battery Charge", KWH, 0.1),
    Sensor(515, "Day Battery discharge", KWH, 0.1),
    Sensor(521, "Day Grid Export", KWH, 0.1),
    Sensor(520, "Day Grid Import", KWH, 0.1),
    Sensor(536, "Day Gen Energy", KWH, 0.1),
    Sensor(526, "Day Load Energy", KWH, 0.1),  # I guess "used" = load?
    Sensor(529, "Day PV Energy", KWH, 0.1),
    Sensor((506, 507), "Total Active Energy", KWH, 0.1),  # signed?
    Sensor((516, 517), "Total Battery Charge", KWH, 0.1),
    Sensor((518, 519), "Total Battery Discharge", KWH, 0.1),
    Sensor((524, 525), "Total Grid Export", KWH, 0.1),
    Sensor((522, 523), "Total Grid Import", KWH, 0.1),
    Sensor((527, 528), "Total Load Energy", KWH, 0.1),
    Sensor((534, 535), "Total PV Energy", KWH, 0.1),
)

This comment is just to ask if you are aware of this / to document this change.

maslyankov, 2025-09-05:
Having the same "issue". Where/how do you see this full version number?

Johnny1206, 2025-09-06:
I receive Firmware Updates from Deye as Bin-Files via Email because my inverter is not connected to the network via the WiFi Dongle. So the "full version numbers" are simply the File Names of the Bin Files.

maslyankov, 2025-09-06:
> @maslyankov 
> I receive Firmware Updates from Deye as Bin-Files via Email because my inverter is not connected to the network via the WiFi Dongle. So the "full version numbers" are simply the File Names of the Bin Files. 
> 
> As I really like to have 0.1 kWh precision of the energy values on my modbus readouts and the new Firmware doesn't seem to improve the small issue I observe, I have downgraded again today to 1095 and C040. 

Oh okay. How do you upload the bin firmware to the inverter? Updates are the only reason I keep my solarman dongle online... it'd be great to know. Thanks!

Johnny1206, 2025-09-06:
> Oh okay. How do you upload the bin firmware to the inverter? Updates are the only reason I keep my solarman dongle online... it'd be great to know. Thanks!

For Firmware Updates i plug in the Solarman Dongle, connect my Laptop to its Wifi and Upload the Binaries one after the other via the Web Interface of the Dongle. I just don't have any internet connection at the installation site of the inverter. 
They usually send an Upgrade Guide with Binaries which describes the steps with screenshots.